introlab/rtabmap_ros #4: How to build and run against OpenNI 1.x under Ubuntu 12.04 LTS

hustcalm, 2015-01-09:
Hi there,

Tried to install the binary which version is 0.7, all the workarounds described in issue 3 suits well, however got 'illegal instruction(core dumped)" error when try to roslaunch. 

The box runs Ubuntu 12.04 LTS, using Ros-Hydro, however installed OpenNI 1.x and SensorKinect driver. Judging from that rtabmap links against OpenNI 2.x by default, is it possible to build from source using the existed OpenNI 1.x and how? Now I find the CMakelists.txt tries to search OpenNI2.

Also, is it possible to use OpenNI 2.x under Ubuntu 12.04, rumors in the internet say it just works for Windows not Linux, is that true?

Thanks in advance in @matlabbe

hustcalm, 2015-01-09:
Hi, looks like I figured out the solution, just comments out  **#FIND_PACKAGE(OpenNI2)** in CMakeLists.txt of rtabmap and build, you are done!

Hope this heads up does some help who runs into the same problem like me：-）

closed 2015-01-09 by hustcalm

matlabbe, 2015-01-09:
The "illegal instruction(core dumped)" error on binaries on Hydro is recent, there is somewhat broken in these binaries... I'm working on it. The binaries on Indigo don't have this error.

By default, if it finds OpenNI2 or Freenect, it will link on them... otherwise RTAB-Map will still build without them. OpenNI1 is always linked through the PCL library.