introlab/rtabmap_ros #6: Rtabmap with stereo camera not found

fariner, 2015-03-02:
Hello,

i am trying to run Rtabmap (concretly the demo launch file of stereo outdoor adding some modifications to it) with 2 cameras calibrated and not synchronized. The node stereo_image _proc work good and i can view the PointCloud2 output. When i run the launch file, rviz open but i cant see point clouds output of Rtabmap in it. In the launch file I only add the aprox sync parameter (in True value) in the nodes that have this parameter and I add too the parameter "frame_id" with the value "camera" because in this moment I dont have a URDF file of the final version of mi robot, in this moment, I only need to try the Rtabmap with the camera. (I suppose that the "frame_id" "camera" is a default parameter in Ros)
I believe that the problem is between tf and "frame_id". Can you help me, what could be happening?
How can i to run Rtabmap with a correct "frame_id" or a good tf configuration?

Thanks in advance.

Best regards.

matlabbe, 2015-03-02:
The default "frame_id" for rtabmap nodes is "base_link". You can setup a [static transform publisher](http://wiki.ros.org/tf#static_transform_publisher) to publish the "base_link" -> "camera" frame (assuming that the frame_id in the stereo images/camera_infos is "camera" too):

```
<node pkg="tf" type="static_transform_publisher" name="base_to_camera_tf" 
    args="0.0 0.0 0.0 0.0 0.0 0.0 /base_link /camera 100" />
```

fariner, 2015-03-04:
Hello,

thank you for you early response.
I tried it that you told me in the last comment but I recibe a warning mensaje and the Rtabmap dont work.
The warning message is: "base_link" passed to lookupTransform argument target_frame does not exist.

Do you know what is the problem?

Best Regards.

I dont have URDF file.
The stereo image proc is running and working good when i execute the launch file.
I pass you the launch file that i use:

<launch>

<arg name="rviz" default="true" />
<arg name="rtabmapviz" default="false" />
<param name="use_sim_time" type="bool" value="True"/>

<node pkg="tf" type="static_transform_publisher" name="base_to_camera_tf" args="0.0 0.0 0.0 0.0 0.0 0.0 /base_link /camera 100" /> 

<node pkg="rtabmap_ros" type="stereo_odometry" name="stereo_odometry" output="screen">
<remap from="left/image_rect" to="/stereo/left/image_rect"/>
<remap from="right/image_rect" to="/stereo/right/image_rect"/>
<remap from="left/camera_info" to="/stereo/left/camera_info"/>
<remap from="right/camera_info" to="/stereo/right/camera_info"/>
<remap from="odom" to="/odometry"/>
<param name="frame_id" type="string" value="/base_link"/>
<param name="odom_frame_id" type="string" value="odom"/>
<param name="Odom/InlierDistance" type="string" value="0.1"/>
<param name="Odom/MinInliers" type="string" value="10"/>
<param name="Odom/RoiRatios" type="string" value="0.03 0.03 0.04 0.04"/>
<param name="Odom/MaxDepth" type="string" value="10"/>
<param name="OdomBow/NNDR" type="string" value="0.8"/>
<param name="GFTT/MaxCorners" type="string" value="500"/>
<param name="GFTT/MinDistance" type="string" value="5"/>
<param name="Odom/FillInfoData" type="string" value="$(arg rtabmapviz)"/>
<param name="approx_sync" type="bool" value="True"/>
</node>
<group ns="rtabmap">

<node name="rtabmap" pkg="rtabmap_ros" type="rtabmap" output="screen" args="--delete_db_on_start --udebug">
<param name="frame_id" type="string" value="/base_link"/>
<param name="subscribe_stereo" type="bool" value="true"/>
<param name="subscribe_depth" type="bool" value="false"/>
<remap from="left/image_rect" to="/stereo/left/image_rect_color"/>
<remap from="right/image_rect" to="/stereo/right/image_rect"/>
<remap from="left/camera_info" to="/stereo/left/camera_info"/>
<remap from="right/camera_info" to="/stereo/right/camera_info"/>
<remap from="odom" to="/odometry"/>
<param name="queue_size" type="int" value="5"/>
<param name="stereo_approx_sync" type="bool" value="True"/>

<param name="Rtabmap/TimeThr" type="string" value="700"/>
<param name="Rtabmap/DetectionRate" type="string" value="1"/>
<param name="Kp/WordsPerImage" type="string" value="200"/>
<param name="Kp/RoiRatios" type="string" value="0.03 0.03 0.04 0.04"/>
<param name="Kp/DetectorStrategy" type="string" value="0"/> 
<param name="Kp/NNStrategy" type="string" value="1"/> 
<param name="SURF/HessianThreshold" type="string" value="1000"/>
<param name="LccBow/MaxDepth" type="string" value="5"/>
<param name="LccBow/MinInliers" type="string" value="10"/>
<param name="LccBow/InlierDistance" type="string" value="0.02"/>
<param name="LccReextract/Activated" type="string" value="true"/>
<param name="LccReextract/MaxWords" type="string" value="500"/>

<param name="RGBD/ToroIterations" type="string" value="0"/>
</node>

<node pkg="rtabmap_ros" type="map_optimizer" name="map_optimizer"/>
<node if="$(arg rviz)" pkg="rtabmap_ros" type="map_assembler" name="map_assembler">
<param name="occupancy_grid" type="bool" value="true"/>
<remap from="mapData" to="mapData_optimized"/>
<remap from="grid_projection_map" to="/map"/>
</node>

<node if="$(arg rtabmapviz)" pkg="rtabmap_ros" type="rtabmapviz" name="rtabmapviz" args="-d $(find rtabmap_ros)/launch/config/rgbd_gui.ini" output="screen">
<param name="subscribe_stereo" type="bool" value="true"/>
<param name="subscribe_odom_info" type="bool" value="true"/>
<param name="queue_size" type="int" value="5"/>
<param name="frame_id" type="string" value="/base_link"/>
<remap from="left/image_rect" to="/stereo/left/image_rect_color"/>
<remap from="right/image_rect" to="/stereo/right/image_rect"/>
<remap from="left/camera_info" to="/stereo/left/camera_info"/>
<remap from="right/camera_info" to="/stereo/right/camera_info"/>
<remap from="odom_info" to="/odom_info"/>
<remap from="odom" to="/odometry"/>
<remap from="mapData" to="mapData_optimized"/>
</node>
</group>

<node if="$(arg rviz)" pkg="rviz" type="rviz" name="rviz" args="-d $(find rtabmap_ros)/launch/config/demo_stereo_outdoor.rviz"/>
</launch>

matlabbe, 2015-03-04:
You should not use `<param name="use_sim_time" type="bool" value="True"/>` when you are not using a rosbag. Can you show the Tf tree? :
```
$ rosrun tf view_frames
$ evince frames.pdf
```

fariner, 2015-03-05:
Hello,

After your last comment,

I changed the "use_sim_time" value to "False".

i executed the "rosrun tf view_frames" and "evince frames.pdf" command and:

If i execute this commands before to run the launch file (with the cameras and stereo image proc running and working good) the result is: no tf data received

If i execute this commands after  to run the launch file (with the cameras in a static position without to move the cameras) the result is:
map->odom->base_link->camera
In this case i can see in rviz some points (output of Rtabmap)

BUT, if i move the camera i believe that the odometry is loss, 
I execute now the tf command above and the result is:
map->odom
base_link->camera
i receive these messages too:
[Error]                  Rtabmap.cpp:765::process() RGB-D SLAM node is enabled and no odometry is provided. Image XXXXX is ignored!
[Warn]                  RTAB-Map could not process the data recieved! (ROS id = YYYYY)

Do you know what is the problem?

Best Regards.

matlabbe, 2015-03-05:
Effectively, I've meant *after* running the launch file. So, the odometry seems lost. The tf between /odom and /base_link is not published when the odometry cannot be computed. You can set `<arg name="rtabmapviz" default="true" />` to see better when the odometry cannot be computed (there will be a red screen). Refer to "Odometry Lost" section on [this page](https://github.com/introlab/rtabmap/wiki/Kinect-mapping) for more info. 

**Important**: [stereo_odometry](http://wiki.ros.org/rtabmap_ros#stereo_odometry) node is not really good **inside**, unless there is a lot of texture. You could try another stereo odometry packages like [viso2_ros or fovis_ros](http://wiki.ros.org/rtabmap_ros/Tutorials/StereoOutdoorMapping#Community_Stereo_Odometry) if they fit more your need. [viso2_ros](http://wiki.ros.org/viso2_ros) or [fovis_ros](http://wiki.ros.org/fovis_ros) can then provide odometry for [rtabmap](http://wiki.ros.org/rtabmap_ros#rtabmap) node. Note also that a [good stereo calibration](http://wiki.ros.org/camera_calibration/Tutorials/StereoCalibration) is required for any stereo odometry approaches.

fariner, 2015-03-05:
Thank you very much for your support.

I am working in a university proyect about SLAM in robotic and i am investigating about Rtabmap for integrate it to a custom robot.

Best Regards.

fariner, 2015-03-17:
Hello,

i have problems to configure the launch file to my custom robot. In this case, Rtabmap works more or less good in an illuminated room but does not work good outdoor because the odometry is lost since Rtabmap began.

Please, Can we to talk by email to help me to configure Rtabmap to my specific situation?

My email is: enriquesaizm@gmail.com

Thank you in advance.

Best Regards.

closed 2016-02-24 by matlabbe