introlab/rtabmap_ros #7: Can't locate node [rgbd_odometry] in package [rtabmap_ros]

IlyaPetrovM, 2015-03-26:
I've installed rtabmap_ros using *apt-get*.
After that I tried to launch the mapping, but I got this error:
```bash
ERROR: cannot launch node of type [rtabmap_ros/rgbd_odometry]: can't locate node [rgbd_odometry] in package [rtabmap_ros]
ERROR: cannot launch node of type [rtabmap_ros/rtabmap]: can't locate node [rtabmap] in package [rtabmap_ros]
ERROR: cannot launch node of type [rtabmap_ros/rtabmapviz]: can't locate node [rtabmapviz] in package [rtabmap_ros]
```
Here is the full story:
```bash
ilya@workstation:~/catkin_ws$ sudo apt-get install ros-hydro-rtabmap-ros 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  python-aptdaemon python-defer aptdaemon xul-ext-ubufox
Use 'apt-get autoremove' to remove them.
The following extra packages will be installed:
  ros-hydro-rtabmap
The following NEW packages will be installed:
  ros-hydro-rtabmap ros-hydro-rtabmap-ros
0 upgraded, 2 newly installed, 0 to remove and 11 not upgraded.
Need to get 0 B/5,573 kB of archives.
After this operation, 15.2 MB of additional disk space will be used.
Do you want to continue [Y/n]? y
Selecting previously unselected package ros-hydro-rtabmap.
(Reading database ... 213152 files and directories currently installed.)
Unpacking ros-hydro-rtabmap (from .../ros-hydro-rtabmap_0.8.3-0precise-20150209-0138-+0000_i386.deb) ...
Selecting previously unselected package ros-hydro-rtabmap-ros.
Unpacking ros-hydro-rtabmap-ros (from .../ros-hydro-rtabmap-ros_0.8.3-0precise-20150214-0413-+0000_i386.deb) ...
Setting up ros-hydro-rtabmap (0.8.3-0precise-20150209-0138-+0000) ...
Setting up ros-hydro-rtabmap-ros (0.8.3-0precise-20150214-0413-+0000) ...
Processing triggers for libc-bin ...
ldconfig deferred processing now taking place
ilya@workstation:~/catkin_ws$ roslaunch rtabmap_ros rgbd_mapping.launch 
... logging to /home/ilya/.ros/log/8ec9a150-d38d-11e4-a2a9-00248c31f50c/roslaunch-workstation-5090.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://workstation:40384/

SUMMARY
========

PARAMETERS
 * /rosdistro
 * /rosversion
 * /rtabmap/rtabmap/LccBow/InlierDistance
 * /rtabmap/rtabmap/LccBow/MinInliers
 * /rtabmap/rtabmap/subscribe_depth
 * /rtabmap/rtabmapviz/subscribe_depth
 * /rtabmap/rtabmapviz/subscribe_odom_info
 * /rtabmap/visual_odometry/Odom/FeatureType
 * /rtabmap/visual_odometry/Odom/FillInfoData
 * /rtabmap/visual_odometry/Odom/InlierDistance
 * /rtabmap/visual_odometry/Odom/MaxDepth
 * /rtabmap/visual_odometry/Odom/MinInliers
 * /rtabmap/visual_odometry/Odom/Strategy
 * /rtabmap/visual_odometry/OdomBow/LocalHistorySize
 * /rtabmap/visual_odometry/OdomBow/NNType

NODES
  /rtabmap/
    rtabmap (rtabmap_ros/rtabmap)
    rtabmapviz (rtabmap_ros/rtabmapviz)
    visual_odometry (rtabmap_ros/rgbd_odometry)
  /
    base_to_camera_tf (tf/static_transform_publisher)

auto-starting new master
process[master]: started with pid [5104]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 8ec9a150-d38d-11e4-a2a9-00248c31f50c
process[rosout-1]: started with pid [5117]
started core service [/rosout]
process[base_to_camera_tf-2]: started with pid [5129]
ERROR: cannot launch node of type [rtabmap_ros/rgbd_odometry]: can't locate node [rgbd_odometry] in package [rtabmap_ros]
ERROR: cannot launch node of type [rtabmap_ros/rtabmap]: can't locate node [rtabmap] in package [rtabmap_ros]
ERROR: cannot launch node of type [rtabmap_ros/rtabmapviz]: can't locate node [rtabmapviz] in package [rtabmap_ros]
```

matlabbe, 2015-03-26:
Try to relaunch the terminal or execute this after installing rtabmap:
```bash
$ source /opt/ros/hydro/setup.bash
```

IlyaPetrovM, 2015-03-26:
Thank you! It helped!

closed 2015-03-26 by IlyaPetrovM