introlab/rtabmap_ros #8: Process has died. exit code -4

IlyaPetrovM, 2015-03-26:
I've tried to launch mapping:
```
$ roslaunch  rtabmap_ros rgbd_mapping.launch 
... logging to /home/ilya/.ros/log/7397259a-d3a3-11e4-afe3-00248c31f50c/roslaunch-workstation-5728.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://workstation:53964/

SUMMARY
========

PARAMETERS
 * /rosdistro
 * /rosversion
 * /rtabmap/rtabmap/LccBow/InlierDistance
 * /rtabmap/rtabmap/LccBow/MinInliers
 * /rtabmap/rtabmap/subscribe_depth
 * /rtabmap/rtabmapviz/subscribe_depth
 * /rtabmap/rtabmapviz/subscribe_odom_info
 * /rtabmap/visual_odometry/Odom/FeatureType
 * /rtabmap/visual_odometry/Odom/FillInfoData
 * /rtabmap/visual_odometry/Odom/InlierDistance
 * /rtabmap/visual_odometry/Odom/MaxDepth
 * /rtabmap/visual_odometry/Odom/MinInliers
 * /rtabmap/visual_odometry/Odom/Strategy
 * /rtabmap/visual_odometry/OdomBow/LocalHistorySize
 * /rtabmap/visual_odometry/OdomBow/NNType

NODES
  /rtabmap/
    rtabmap (rtabmap_ros/rtabmap)
    rtabmapviz (rtabmap_ros/rtabmapviz)
    visual_odometry (rtabmap_ros/rgbd_odometry)
  /
    base_to_camera_tf (tf/static_transform_publisher)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[base_to_camera_tf-1]: started with pid [5746]
process[rtabmap/visual_odometry-2]: started with pid [5757]
process[rtabmap/rtabmap-3]: started with pid [5758]
process[rtabmap/rtabmapviz-4]: started with pid [5759]
[ INFO] [1427366017.888529423]: Starting node...
[rtabmap/visual_odometry-2] process has died [pid 5757, exit code -4, cmd /opt/ros/hydro/lib/rtabmap_ros/rgbd_odometry rgb/image:=/camera/rgb/image_rect_color depth/image:=/camera/depth_registered/image_raw rgb/camera_info:=/camera/depth_registered/camera_info __name:=visual_odometry __log:=/home/ilya/.ros/log/7397259a-d3a3-11e4-afe3-00248c31f50c/rtabmap-visual_odometry-2.log].
log file: /home/ilya/.ros/log/7397259a-d3a3-11e4-afe3-00248c31f50c/rtabmap-visual_odometry-2*.log
[ INFO] [1427366018.000439385]: Starting node...
[rtabmap/rtabmap-3] process has died [pid 5758, exit code -4, cmd /opt/ros/hydro/lib/rtabmap_ros/rtabmap --delete_db_on_start rgb/image:=/camera/rgb/image_rect_color depth/image:=/camera/depth_registered/image_raw rgb/camera_info:=/camera/depth_registered/camera_info __name:=rtabmap __log:=/home/ilya/.ros/log/7397259a-d3a3-11e4-afe3-00248c31f50c/rtabmap-rtabmap-3.log].
log file: /home/ilya/.ros/log/7397259a-d3a3-11e4-afe3-00248c31f50c/rtabmap-rtabmap-3*.log
[ INFO] [1427366018.090905738]: rtabmapviz: Using configuration from "/opt/ros/hydro/share/rtabmap_ros/launch/config/rgbd_gui.ini"
[rtabmap/rtabmapviz-4] process has died [pid 5759, exit code -4, cmd /opt/ros/hydro/lib/rtabmap_ros/rtabmapviz -d /opt/ros/hydro/share/rtabmap_ros/launch/config/rgbd_gui.ini rgb/image:=/camera/rgb/image_rect_color depth/image:=/camera/depth_registered/image_raw rgb/camera_info:=/camera/depth_registered/camera_info __name:=rtabmapviz __log:=/home/ilya/.ros/log/7397259a-d3a3-11e4-afe3-00248c31f50c/rtabmap-rtabmapviz-4.log].
log file: /home/ilya/.ros/log/7397259a-d3a3-11e4-afe3-00248c31f50c/rtabmap-rtabmapviz-4*.log
```

matlabbe, 2015-03-26:
All nodes crashed. Can you try if the standalone works and doesn't crash at start? Simply try:
```bash
$ rtabmap
```

IlyaPetrovM, 2015-03-26:
I've tried:

ilya@workstation:~$ rtabmap 
Illegal instruction (core dumped)

matlabbe, 2015-03-26:
I know that on some computers (even my laptop), the binaries built by ROS have this issue. I recommend to build rtabmap from source. It seems to be a similar problem than https://github.com/introlab/rtabmap_ros/issues/5.

1) Uninstall ros-hydro-rtabmap:
```
sudo apt-get remove ros-hydro-rtabmap ros-hydro-rtabmap-ros
```

2) [Install rtabmap_ros from source](https://github.com/introlab/rtabmap_ros#build-from-source)

IlyaPetrovM, 2015-03-27:
Thank you! It works!

closed 2015-03-27 by IlyaPetrovM

jiminotron, 2015-08-04:
I HAVE Same problem and i cant fix it ,I try to install rtabmap by binary i get the segmentation fault error ,and i try to install rtab by source i get "process died error"

please help to fix it

chiragmajithia, 2016-10-16:
rtabmap standalone crashes with Segmetation Fault as soon as I start the Mapping. Using Ubuntu 16.04 and ros-kinetic.  
The rtabmap and rtabmap_ros both compile without errors.
Did anyone face the same issue?

matlabbe, 2016-10-16:
Can rtabmap standalone crash too on start?
```bash
$ rtabmap
```

chiragmajithia, 2016-10-16:
yes, standalone crashes - as soon as it receives first data from r200 with Segmentation Fault

matlabbe, 2020-04-08:
cross-ref, see debugging instructions here: https://github.com/introlab/rtabmap_ros/issues/28