introlab/rtabmap_ros #10: compilation does not end for CoreWrapper.cpp for Ubuntu utopic and vivid

vrabaud, 2015-05-12:
One the buildfarms, it stays for an hour or two on the compilation of that file:
http://jenkins.ros.org:8080/job/ros-jade-rtabmap-ros_binarydeb_vivid_amd64/2/console

matlabbe, 2015-05-12:
I've noticed too this morning the long compilation of CoreWrapper.cpp. On Trusty, the amd64 binary of rtabmap_ros took around 50 min:
 * Indigo 49 min (http://jenkins.ros.org/job/ros-indigo-rtabmap-ros_binarydeb_trusty_amd64/33/)
 * Jade 55 min (http://jenkins.ros.org/job/ros-jade-rtabmap-ros_binarydeb_trusty_amd64/2/)

I've already seen a long compilation time of rtabmap_ros when I used "catkin -j4" on a machine with RAM available below 6 Gig. Some files require over 1 Gig to compile, so with 4 threads, the swap was required. *Using the swap memory increases a lot the compilation time*.

Well, I tried with a fresh virtual machine with Ubuntu 15.04 + Jade and built rtabmap + rtabmap_ros. It builds flawlessly. The maximum RAM used was *1.9 Gig for CoreWrapper.cpp*. This may be the problem.

vrabaud, 2015-05-12:
ok, well, you're on top of it :) Probably highly templated code.Maybe you can split it in smaller files.

matlabbe, 2015-05-12:
I'll check if I can split some of the PCL headers to reduce the memory load used for compilation. What is the RAM available on the buildfarm machines for a target memory usage limit?

vrabaud, 2015-05-12:
some code could not use PCL. E.g. https://github.com/introlab/rtabmap_ros/blob/master/src/CoreWrapper.cpp#L765 could use tf2_sensor_msgs by keeping a PointCloud2 (got added recently). You don't even need the whole tf2 infrastructure, you can just use: https://github.com/ros/geometry_experimental/blob/indigo-devel/tf2_sensor_msgs/include/tf2_sensor_msgs/tf2_sensor_msgs.h#L59

matlabbe, 2015-05-12:
Thank you, I'll look at this!

matlabbe, 2015-05-20:
Some refactoring were done in upstream rtabmap library (particularly the big rtabmap/core/util3d.h file, which included many pcl headers) as well as splitting CoreWrapper.cpp in two files.

Looking on Jenkins, the long build time seems not here anymore. It took 18 min on the latest version 0.9.0 on vivid amd64 (instead of [1 hr 34 min](http://jenkins.ros.org/job/ros-jade-rtabmap-ros_binarydeb_vivid_amd64/4/) for previous 0.8.13 version):
http://jenkins.ros.org/job/ros-jade-rtabmap-ros_binarydeb_vivid_amd64/5/

closed 2015-05-20 by matlabbe