introlab/rtabmap_ros #11: compilation error while using catkin_make

CarlosEnriqueMontes, 2015-05-13:
Version of product:
Ubuntu 14.04 
ROS Indigo

Hi! Yesterday I just came across rtabmap and installed it using sudo apt-get install. But since I am adapting this wonderful project into my research, I really need to look into the code, and so I decided to install from source instead. That's why I did the following:

$ sudo apt-get remove ros-indigo-rtabmap ros-indigo-rtabmap-ros
$ sudo purge rtabmap  // just to make sure there is no confusion in the directories when compiling again from source
$ sudo apt-get update

And I followed the rest of the installation as indicated in the tutorial. But here is what I received:

--------------------------------------------------------------------------------------------------------------------------------------

Base path: /home/charly/indigo_ws
Source space: /home/charly/indigo_ws/src
Build space: /home/charly/indigo_ws/build
Devel space: /home/charly/indigo_ws/devel
Install space: /home/charly/indigo_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/charly/indigo_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/charly/indigo_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/charly/indigo_ws/devel;/opt/ros/indigo
-- This workspace overlays: /home/charly/indigo_ws/devel;/opt/ros/indigo
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/charly/indigo_ws/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.11
-- BUILD_SHARED_LIBS is on
/opt/ros/indigo/share/catkin/cmake/em/order_packages.cmake.em:23: error: <type 'exceptions.RuntimeError'>: Multiple packages found with the same name "rtabmap":
- rtabmap
- share/rtabmap
Traceback (most recent call last):
  File "/usr/bin/empy", line 3276, in <module>
    if __name__ == '__main__': main()
  File "/usr/bin/empy", line 3274, in main
    invoke(sys.argv[1:])
  File "/usr/bin/empy", line 3257, in invoke
    interpreter.wrap(interpreter.file, (file, name))
  File "/usr/bin/empy", line 2262, in wrap
    self.fail(e)
  File "/usr/bin/empy", line 2253, in wrap
    callable(*args)
  File "/usr/bin/empy", line 2326, in file
    self.safe(scanner, done, locals)
  File "/usr/bin/empy", line 2368, in safe
    self.parse(scanner, locals)
  File "/usr/bin/empy", line 2388, in parse
    token.run(self, locals)
  File "/usr/bin/empy", line 1403, in run
    interpreter.execute(self.code, locals)
  File "/usr/bin/empy", line 2565, in execute
    exec(statements, self.globals)
  File "<string>", line 17, in <module>
  File "/usr/lib/python2.7/dist-packages/catkin_pkg/topological_order.py", line 111, in topological_order
    for path, package in find_packages(space).items():
  File "/usr/lib/python2.7/dist-packages/catkin_pkg/packages.py", line 93, in find_packages
    raise RuntimeError('\n'.join(duplicates))
RuntimeError: Multiple packages found with the same name "rtabmap":
- rtabmap
- share/rtabmap
CMake Error at /opt/ros/indigo/share/catkin/cmake/safe_execute_process.cmake:11 (message):
  
  execute_process(/home/charly/indigo_ws/build/catkin_generated/env_cached.sh
  "/usr/bin/python" "/usr/bin/empy" "--raw-errors" "-F"
  "/home/charly/indigo_ws/build/catkin_generated/order_packages.py" "-o"
  "/home/charly/indigo_ws/build/catkin_generated/order_packages.cmake"
  "/opt/ros/indigo/share/catkin/cmake/em/order_packages.cmake.em") returned
  error code 1
Call Stack (most recent call first):
  /opt/ros/indigo/share/catkin/cmake/em_expand.cmake:25 (safe_execute_process)
  /opt/ros/indigo/share/catkin/cmake/catkin_workspace.cmake:29 (em_expand)
  CMakeLists.txt:63 (catkin_workspace)

-- Configuring incomplete, errors occurred!
See also "/home/charly/indigo_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/charly/indigo_ws/build/CMakeFiles/CMakeError.log".
make: *** [cmake_check_build_system] Error 1

---------------------------------------------------------------------------------------------------------------------------------------

I am still a newbie in the debugging and found nowhere on explaining this kind of error. Though there is still a naming cliche on "rtabmap", it seems that the real problem lies in CMake Error, which I cannot drag any lues out of it. Would you kindly give me a hand on the problem? Thank you !

matlabbe, 2015-05-13:
Be aware that "rtabmap" **should not** be built in the catkin workspace, but it should be installed in the Devel space ("cmake -DCMAKE_INSTALL_PREFIX=~/indigo_ws/devel .."). See [install rtabmap_ros from source](https://github.com/introlab/rtabmap_ros#build-from-source).

Make sure that file "/opt/ros/indigo/share/rtabmap/package.xml" is not here. Same for "rtabmap_ros".

Megacephalo, 2015-05-14:
Hi, Mathieu. Thank you for reminding me to check my installation procedure, that is when I found out that before installing rtabmap in my /indigo_ws/devel, I ignored cloning the package outside of the workspace first, and instead I clone the package inside of the devel folder, that is why the error occurred. Once I ran through the installation steps again, rtabmap_ros did compile almost completely, except the part of CoreWrapper, which with some searching I found out that in "Issue 8" (https://code.google.com/p/rtabmap/issues/detail?id=8) you actually recommended removing the entire opencv packages first and then installed the vision_opencv package from source. 
So I did remove every opencv package and install the cloned one, here I get:

---------------------------------------------------------------------------------------------------------------------------------------

CMake Error at vision_opencv/image_geometry/CMakeLists.txt:5 (find_package):
  By not providing "FindOpenCV.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "OpenCV", but
  CMake did not find one.

  Could not find a package configuration file provided by "OpenCV" with any
  of the following names:

    OpenCVConfig.cmake
    opencv-config.cmake

  Add the installation prefix of "OpenCV" to CMAKE_PREFIX_PATH or set
  "OpenCV_DIR" to a directory containing one of the above files.  If "OpenCV"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "/home/charly/indigo_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/charly/indigo_ws/build/CMakeFiles/CMakeError.log".
make: *** [cmake_check_build_system] Error 1

-------------------------------------------------------------------------------------------------------------------------------------------

What can I do to successfully install the OpenCV from source? Once again, merci beacoup !

matlabbe, 2015-05-14:
OpenCV built from source is not mandatory anymore, RTAB-Map should be able to build with the binaries of OpenCV. It is even recommended to keep the OpenCV binaries installed because some ROS packages depend on it.

To make RTAB-Map build with OpenCV from source (I assume it is because you want the SURF/SIFT nonfree module), set the environment variable `OpenCV_DIR` to where the `OpenCVConfig.cmake` file is installed when you have built/installed OpenCV from source. Mine is installed (by default) in `/usr/local/share/OpenCV` 
```bash
# build rtabmap 
$ export OpenCV_DIR=/usr/local/share/OpenCV
$ cd rtabmap/build
$ cmake -DCMAKE_INSTALL_PREFIX=~/indigo_ws/devel ..
$ make
$ make install

# build rtabmap_ros
$ cd ~/indigo_ws
$ catkin_make
```

What is exactly the error on CoreWrapper.cpp?

Megacephalo, 2015-05-14:
Well, I lost the messages, but I remember it looks something like 

>undefined reference to `cv::ocl::integral(cv::ocl::oclMat const&, cv::ocl::oclMat&)' 

And these error messages appeared when I still have my OpenCV packages installed. 
Oh! just a quick question before making any modification. Can I be installing OpenCV from source as independent libraries from its official repository and installing vision_opencv as bridge? Will I get the binaries of OpenCV with nonfree module while still be able to use it with ROS ? Thanks !

matlabbe, 2015-05-14:
Thanks, I've seen this error before. The solution may be in this [post](http://official-rtab-map-forum.67519.x6.nabble.com/catkin-make-ros-pkg-failed-tp22p24.html).

You can keep installed all related OpenCV binaries from ROS Indigo (including cv_bridge...). The SURF stuff are used only by the rtabmap library, they are not required for the rtabmap_ros package as it can build with the ROS OpenCV version.

Megacephalo, 2015-05-16:
Thank you first for giving so much help. I did finally compiled the package from source and the package did run regardless of having the nonfree module installed or not. I did run into another bug (maybe by my own system settings), but that is not related to this post, so I think I will close here and starts another post if necessary. Thank you so much !

closed 2016-02-24 by matlabbe