introlab/rtabmap_ros #12: Missing dependencies on fresh system

DonatasKozlovskis, 2015-05-20:
Hello, 
When installing of fresh system, rtabmap_ros complains about missing libraries:

libcostmap_2d.so: cannot open shared object file: No such file or directory
libopencv_nonfree.so.2.4: cannot open shared object file: No such file or directory

it can be easily fix installing required packages, but 
it would be nice to have notices and updatex package.xml dependency list.

matlabbe, 2015-05-20:
On which ROS version you are? The package.xml of rtabmap_ros contains costmap_2d dependency... though not required as it can builds/runs without it. 

For opencv_nonfree, it is optional too... rtabmap package should build without it too.

libg2o, libfreenect are dependencies set in package.xml of the rtabmap package, they are optional too. 

Well, normally, you should have a CMake error when a required dependency is not met. Is the error happens when launching a node? If so, very that you sourced the /opt/ros/ROSDISTRO/setup.bash to make sure that the linker can fin the ROS libraries.

On which system / ros version are you using?

DonatasKozlovskis, 2015-05-29:
Hello,

It was just freshly installed ROS indigo. Environment variables were already set and correct,
I just downloaded ros-indigo-rtabmap and ros-indigo-rtabmap-ros packages, but mentioned dependencies I had to install manually.

matlabbe, 2015-05-29:
Thank you for testing. However, I'm still not able to reproduce the missing dependencies error on a Ubuntu 14.04/Indigo virtual machine. What is your computer? and Ubuntu version?

Here are some outputs that I have. I removed every dependencies of rtabmap:
```bash
$ sudo apt-get remove ros-indigo* libopencv* libpcl*
```

Then installed directly ros-indigo-rtabmap-ros (note the ros-indigo-costmap-2d):
```bash
$ sudo apt-get update
$ sudo apt-get install ros-indigo-rtabmap-ros
The following extra packages will be installed:
  libcv-dev libcvaux-dev libhighgui-dev libopencv-calib3d-dev
  libopencv-calib3d2.4 libopencv-contrib-dev libopencv-contrib2.4
  libopencv-core-dev libopencv-core2.4 libopencv-dev libopencv-features2d-dev
  libopencv-features2d2.4 libopencv-flann-dev libopencv-flann2.4
  libopencv-gpu-dev libopencv-gpu2.4 libopencv-highgui-dev
  libopencv-highgui2.4 libopencv-imgproc-dev libopencv-imgproc2.4
  libopencv-legacy-dev libopencv-legacy2.4 libopencv-ml-dev libopencv-ml2.4
  libopencv-objdetect-dev libopencv-objdetect2.4 libopencv-ocl-dev
  libopencv-ocl2.4 libopencv-photo-dev libopencv-photo2.4
  libopencv-stitching-dev libopencv-stitching2.4 libopencv-superres-dev
  libopencv-superres2.4 libopencv-ts-dev libopencv-ts2.4 libopencv-video-dev
  libopencv-video2.4 libopencv-videostab-dev libopencv-videostab2.4
  libopencv2.4-java libopencv2.4-jni libopenni-sensor-pointclouds0
  libpcl-1.7-all libpcl-1.7-all-dev libpcl-1.7-bin libpcl-1.7-doc
  libpcl-apps-1.7 libpcl-apps-1.7-dev libpcl-common-1.7 libpcl-common-1.7-dev
  libpcl-features-1.7 libpcl-features-1.7-dev libpcl-filters-1.7
  libpcl-filters-1.7-dev libpcl-geometry-1.7-dev libpcl-io-1.7
  libpcl-io-1.7-dev libpcl-kdtree-1.7 libpcl-kdtree-1.7-dev
  libpcl-keypoints-1.7 libpcl-keypoints-1.7-dev libpcl-octree-1.7
  libpcl-octree-1.7-dev libpcl-outofcore-1.7 libpcl-outofcore-1.7-dev
  libpcl-people-1.7 libpcl-people-1.7-dev libpcl-recognition-1.7
  libpcl-recognition-1.7-dev libpcl-registration-1.7
  libpcl-registration-1.7-dev libpcl-sample-consensus-1.7
  libpcl-sample-consensus-1.7-dev libpcl-search-1.7 libpcl-search-1.7-dev
  libpcl-segmentation-1.7 libpcl-segmentation-1.7-dev libpcl-surface-1.7
  libpcl-surface-1.7-dev libpcl-tracking-1.7 libpcl-tracking-1.7-dev
  libpcl-visualization-1.7 libpcl-visualization-1.7-dev python-opencv
  ros-indigo-actionlib ros-indigo-actionlib-msgs ros-indigo-angles
  ros-indigo-bond ros-indigo-bondcpp ros-indigo-catkin ros-indigo-class-loader
  ros-indigo-compressed-depth-image-transport
  ros-indigo-compressed-image-transport ros-indigo-costmap-2d
  ros-indigo-cpp-common ros-indigo-cv-bridge ros-indigo-dynamic-reconfigure
  ros-indigo-gencpp ros-indigo-genlisp ros-indigo-genmsg ros-indigo-genpy
  ros-indigo-geometry-msgs ros-indigo-image-geometry
  ros-indigo-image-transport ros-indigo-image-transport-plugins
  ros-indigo-interactive-markers ros-indigo-kdl-conversions
  ros-indigo-laser-geometry ros-indigo-libfreenect ros-indigo-map-msgs
  ros-indigo-media-export ros-indigo-message-filters
  ros-indigo-message-generation ros-indigo-message-runtime
  ros-indigo-move-base-msgs ros-indigo-nav-msgs ros-indigo-nodelet
  ros-indigo-nodelet-topic-tools ros-indigo-octomap ros-indigo-octomap-msgs
  ros-indigo-octomap-ros ros-indigo-orocos-kdl ros-indigo-pcl-conversions
  ros-indigo-pcl-msgs ros-indigo-pcl-ros ros-indigo-pluginlib
  ros-indigo-python-orocos-kdl ros-indigo-python-qt-binding
  ros-indigo-resource-retriever ros-indigo-rosbag ros-indigo-rosbag-storage
  ros-indigo-rosbuild ros-indigo-rosclean ros-indigo-rosconsole
  ros-indigo-rosconsole-bridge ros-indigo-roscpp
  ros-indigo-roscpp-serialization ros-indigo-roscpp-traits ros-indigo-rosgraph
  ros-indigo-rosgraph-msgs ros-indigo-roslaunch ros-indigo-roslib
  ros-indigo-roslz4 ros-indigo-rosmaster ros-indigo-rosmsg ros-indigo-rosnode
  ros-indigo-rosout ros-indigo-rospack ros-indigo-rosparam ros-indigo-rospy
  ros-indigo-rosservice ros-indigo-rostest ros-indigo-rostime
  ros-indigo-rostopic ros-indigo-rosunit ros-indigo-roswtf ros-indigo-rtabmap
  ros-indigo-rviz ros-indigo-sensor-msgs ros-indigo-smclib ros-indigo-std-msgs
  ros-indigo-std-srvs ros-indigo-stereo-msgs ros-indigo-tf
  ros-indigo-tf-conversions ros-indigo-tf2 ros-indigo-tf2-msgs
  ros-indigo-tf2-py ros-indigo-tf2-ros ros-indigo-theora-image-transport
  ros-indigo-topic-tools ros-indigo-urdf ros-indigo-urdf-parser-plugin
[... 82 lines cut ...]
[ INFO] [1432913209.739163438]: 
/rtabmap/rtabmapviz subscribed to:
   /camera/rgb/image_rect_color,
   /camera/depth_registered/image_raw,
   /camera/rgb/camera_info,
   /rtabmap/odom,
   /rtabmap/odom_info
[ INFO] [1432913209.782019588]: rtabmapviz started.
[ INFO] [1432913210.149706425]: RTAB-Map rate detection = 1.000000 Hz
[ INFO] [1432913210.149967270]: rtabmap: Deleted database "/home/mathieu/.ros/rtabmap.db" (--delete_db_on_start is set).
[ INFO] [1432913210.150024713]: rtabmap: Using database from "/home/mathieu/.ros/rtabmap.db".
[ INFO] [1432913210.639748035]: Registering Depth callback...
[ INFO] [1432913210.640651470]: 
/rtabmap/rtabmap subscribed to:
   /camera/rgb/image_rect_color,
   /camera/depth_registered/image_raw,
   /camera/rgb/camera_info,
   /rtabmap/odom
[ INFO] [1432913210.641024698]: rtabmap started...
```

The standalone works too:
```bash
$ rtabmap
```

DonatasKozlovskis, 2015-05-30:
Thank you for comments. 
The machine:
Intel® Core™ i7 4700HQ Processor
DDR3L 1600 MHz 16 G 
NVIDIA® GeForce® GTX850M

I got mentioned error after fresh install of Ubuntu 14.04.2 LTS
and ros-indigo-ros-base

It could be that it was only issues on my machine.

closed 2016-02-24 by matlabbe