introlab/rtabmap_ros #14: obstacle_detection could detect holes

mryellow, 2015-05-30:
Given a situation like this:  
[image]

Current implementation extracts something like this (green=ground, red=obstacle + global/local costmaps from move_base):  
[image]

Given the missing green "ground" features and further on below-normal  "obstacle" features, surely there is some math magic which could project forward the obstacle to the furtherest known good ground feature.

Does `segmentFlatObstacles` in [util3d_mapping](https://github.com/introlab/rtabmap/blob/9fde57843f397d396ecbc071613212f5a714ff39/corelib/include/rtabmap/core/impl/util3d_mapping.hpp#L43) have something which can help here?

matlabbe, 2015-06-03:
`segmentFlatObstacles` is used to segment planar objects like a table to be labeled as an obstacle instead of ground. All planes not coplanar to the biggest plane are considered as obstacles. To work, the camera must at least see always the ground and that the ground takes a big part of the camera image.

In the current code, holes are not segmented as obstacles. In your example, obstacles are detected on the other side of the hole, and the "hole" itself doesn't have any ground or obstacle points. Here is an idea:
[image]
We could project the points hitting under the ground plane (on the hole's wall) on the ground plane and label them as obstacles. This could be done by finding the intersection of the rays with the ground plane (doing it **only** with points under the ground plane). If you are in a 2D environment, ray tracing for all points under z = 0 could be done.

On the move_base part:
* laser : mark, don't clear
* kinect_obstacles (obstacles over ground plane including the projected points above): mark, clear
* kinect_ground: don't mark, clear

cheers

mryellow, 2015-06-03:
Good stuff, sounds achievable in a usable way.

Lasers not clearing could throw up a few extra false-positives where move_base gets stuck trying to recover with a nearby non-existent collision, which then can't be cleared via the point_cloud/kinect/stereo input. Would have to rely on timeout clearing costmaps.

I'm by no means a mathematician but can brute-force my way through some Pythagoras when required. Got a lot on for the next month or 3, but will come back to this eventually if no one else does.

closed 2016-02-24 by matlabbe