introlab/rtabmap_ros #15: Compilation error with ROS Hydro: undefined reference to costmap...

tanmayshankar, 2015-06-04:
Hello, 

I had followed the read me instructions on installing rtabmap_ros from source, (I am running Ubuntu 12.04 and ROS Hydro). Cmake is able to find costmap_2d - it says built target costmap_2d. 
These are the errors I get - I think it is related to linking costmap when compiling librtabmap. 

/home/tanmay/catkin_ws/devel/lib/librtabmap_ros.so: error: undefined reference to 'costmap_2d::CostmapLayer::useExtraBounds(double*, double*, double*, double*)'
collect2: ld returned 1 exit status
make[2]: *** [/home/tanmay/catkin_ws/devel/lib/rtabmap_ros/data_player] Error 1
make[1]: *** [rtabmap_ros/CMakeFiles/data_player.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/home/tanmay/catkin_ws/devel/lib/librtabmap_ros.so: error: undefined reference to 'costmap_2d::CostmapLayer::useExtraBounds(double*, double*, double*, double*)'
collect2: ld returned 1 exit status
make[2]: *** [/home/tanmay/catkin_ws/devel/lib/rtabmap_ros/map_assembler] Error 1
make[1]: *** [rtabmap_ros/CMakeFiles/map_assembler.dir/all] Error 2
/home/tanmay/catkin_ws/devel/lib/librtabmap_ros.so: error: undefined reference to 'costmap_2d::CostmapLayer::useExtraBounds(double*, double*, double*, double*)'
collect2: ld returned 1 exit status
make[2]: *** [/home/tanmay/catkin_ws/devel/lib/rtabmap_ros/grid_map_assembler] Error 1
make[1]: *** [rtabmap_ros/CMakeFiles/grid_map_assembler.dir/all] Error 2
/home/tanmay/catkin_ws/devel/lib/librtabmap_ros.so: error: undefined reference to 'costmap_2d::CostmapLayer::useExtraBounds(double*, double*, double*, double*)'
collect2: ld returned 1 exit status
make[2]: *** [/home/tanmay/catkin_ws/devel/lib/rtabmap_ros/map_optimizer] Error 1
make[1]: *** [rtabmap_ros/CMakeFiles/map_optimizer.dir/all] Error 2
make: *** [all] Error 2
Invoking "make" failed

I'm not sure if modifying my Cmake to explicitly include this is a good idea - any suggestions as to what I can do? 

Thanks in advance!

tanmayshankar, 2015-06-04:
I tried manually linking costmap, but I was still getting the errors. Since costmap is optional, I commented out find_package(costmap_2d) in the Cmake file to compile it, and it compiled. 

I'd still like to sort out the issue though. Any ideas?

matlabbe, 2015-06-08:
looks like a duplicate of [issue 12](https://github.com/introlab/rtabmap_ros/issues/12)

matlabbe, 2015-06-08:
EDIT here it is from source. It is like that something is broken with the costmap_2d dependencies. Normally if costmap_2d is detected, the libraries should be added [here](https://github.com/introlab/rtabmap_ros/blob/927c80d3564fce9fa9097592cdc9d0a2adceccb8/CMakeLists.txt#L145-L157) in the CMakeLists.txt.

tanmayshankar, 2015-06-19:
This was magically solved by removing costmap and putting it back. Thanks for the help! :)

closed 2015-06-19 by tanmayshankar