introlab/rtabmap_ros #18: PCL / Flann undefined reference - compilation error.

tanmayshankar, 2015-06-20:
Hi Mathieu, 

After compiling RTABMap on my Odroid, #26 on RTABMAP issues, rtabmap_ros is refusing to compile by the skin of its teeth. 

After it progresses to 96%, I got a similar error to http://official-rtab-map-forum.67519.x6.nabble.com/Compiling-in-ROS-Indigo-with-PCL-1-7-2-Undefined-reference-to-td386.html , with the undefined references to pcl::search::kdtree and flann, etc. 

Now I have pcl-1.7.2 installed from source, as well as the libpcl-1.7-all binary. I am not sure what rtabmap_ros is linking to.  

I tried to follow what you suggested, installing the (hydro) pcl_conversions and perception_pcl by source. The C++ compiler reports an internal error compiling perception_pcl, after consuming all 2Gb of RAM and 1Gb SWAP. 

This forces me to abandon compiling perception_pcl  from source. Do you know if there is another workaround to this?

tanmayshankar, 2015-06-22:
Here is where things are interesting. 
I just compiled perception_pcl from source (I had to up the SWAP to 3 Gb, and it took 2 hours). 
RTABMAP still cited the same error upon compiling, but I need to delete the cache and check again.

tanmayshankar, 2015-06-22:
Hi Mathieu, 

Your solution was spot on - installing perception_pcl and pcl_conversions from source, and preventing binary installations did the trick, since they linked to PCL 1.7.2 source.  I needed to build rviz from source too. It required me to build my entire catkin workspace again, do a rebuild of rtabmap, and then rtabmap_ros. Collectively it took 6-8 hours, and needed every bit of the 3Gb Swap I put on it, plus the 2Gb RAM. 

In any case, thank you very much for all the help - here's evidence rtabmap can compile on an odroid.
I plan to use a remote client for rviz, shall let you know how that goes.

closed 2015-06-22 by tanmayshankar

matlabbe, 2015-06-23:
A lot of work, but good news! thx for the info!