introlab/rtabmap_ros #19: using bluefox

ksgaur12, 2015-06-29:
how to use this driver to run bluefox camera.

matlabbe, 2015-06-30:
You should change the `frame_id` parameter of each node from `base_footprint` to the fixed link to which your camera is attached (example `camera_link` or `base_link`). What is the TF that your camera driver "bluefox2" is providing? You can run this to see the frames after launching the camera driver:
```bash
$ rosrun tf view_frames
```

The [stereo outdoor demo](http://wiki.ros.org/rtabmap_ros/Tutorials/StereoOutdoorMapping) launch file is configured to use inputs from a rosbag. I have updated the [Stereo B configuration](http://wiki.ros.org/rtabmap_ros/Tutorials/SetupOnYourRobot#Stereo_B_.28recommended.29) with an example of launch file with a real camera.

cheers

ksgaur12, 2015-07-20:
I am not much familiar with the tf 
the output I am getting for $rosrun tf view_frames is this
Listening to /tf for 5.000000 seconds
Done Listening
dot - graphviz version 2.36.0 (20140111.2315)

Detected dot version 2.36
frames.pdf generated

please help me further

matlabbe, 2015-07-20:
I think you can drop the frames.pdf in the Github comment box so we can see your frames. To learn more about TF, see http://wiki.ros.org/tf/Tutorials/Introduction%20to%20tf

Well, I've recently added a new launch file for testing stereo cameras for convenience. See [stereo_mapping.launch](https://github.com/introlab/rtabmap_ros/blob/master/launch/stereo_mapping.launch). I can't test it here for bluefox2 camera, but based on [bluefox2 launch file](https://github.com/KumarRobotics/bluefox2/blob/master/launch/stereo_node.launch), it maybe something like that:
```bash
$ roslaunch bluefox2 stereo_node.launch camera:=stereo_camera
$ roslaunch rtabmap_ros stereo_mapping.launch frame_id:=stereo_camera
```

ksgaur12, 2015-07-23:
I did 
$ rosrun rqt_tf_tree rqt_tf_tree 
it shows no tf data received.

matlabbe, 2015-07-23:
Can you try launching the two launch files above and see what are the errors?

If a TF is still required, you can generate one. Example:
```
$ roslaunch bluefox2 stereo_node.launch camera:=stereo_camera
$ rosrun tf static_transform_publisher 0 0 0 -1.57079 0 -1.57079  base_link stereo_camera 100
$ roslaunch rtabmap_ros stereo_mapping.launch frame_id:=base_link
```
(the static transform publisher rotates at the same time the camera to put it in the world frame)

closed 2016-02-24 by matlabbe