introlab/rtabmap_ros #20: [rtabmap/rgbd_odometry-1] process has died [pid 6538, exit code -11

umair4848, 2015-07-08:
Hi, I have compiled rtabmap and rtabmap_ros fropm source according to the instructions on rtabmap github link.

I'm on RPi 2 and Lubuntu 14.04 LTS.

First I ran this command:

$ roslaunch openni2_launch openni2.launch depth_registration:=true

It seems to run fine. Then in a new terminal tab, I ran this command:

$ roslaunch openni2_launch openni2.launch depth_registration:=true

and it gives the following output along with the error mentioned in the title:

... logging to /home/ubuntu/.ros/log/e5302dea-24fd-11e5-bdf0-801f02d322c9/roslaunch-ubuntu-6520.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:57461/

SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.13
 * /rtabmap/rgbd_odometry/Odom/FeatureType: 6
 * /rtabmap/rgbd_odometry/Odom/FillInfoData: true
 * /rtabmap/rgbd_odometry/Odom/InlierDistance: 0.02
 * /rtabmap/rgbd_odometry/Odom/MaxDepth: 4.0
 * /rtabmap/rgbd_odometry/Odom/MinInliers: 20
 * /rtabmap/rgbd_odometry/Odom/Strategy: 0
 * /rtabmap/rgbd_odometry/OdomBow/LocalHistorySize: 1000
 * /rtabmap/rgbd_odometry/OdomBow/NNType: 3
 * /rtabmap/rgbd_odometry/frame_id: camera_link
 * /rtabmap/rtabmap/LccBow/InlierDistance: 0.02
 * /rtabmap/rtabmap/LccBow/MinInliers: 10
 * /rtabmap/rtabmap/frame_id: camera_link
 * /rtabmap/rtabmap/subscribe_depth: True

NODES
  /rtabmap/
    rgbd_odometry (rtabmap_ros/rgbd_odometry)
    rtabmap (rtabmap_ros/rtabmap)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[rtabmap/rgbd_odometry-1]: started with pid [6538]
process[rtabmap/rtabmap-2]: started with pid [6539]
[rtabmap/rgbd_odometry-1] process has died [pid 6538, exit code -11, cmd /home/ubuntu/catkin_ws/devel/lib/rtabmap_ros/rgbd_odometry rgb/image:=/camera/rgb/image_rect_color depth/image:=/camera/depth_registered/image_raw rgb/camera_info:=/camera/rgb/camera_info __name:=rgbd_odometry __log:=/home/ubuntu/.ros/log/e5302dea-24fd-11e5-bdf0-801f02d322c9/rtabmap-rgbd_odometry-1.log].
log file: /home/ubuntu/.ros/log/e5302dea-24fd-11e5-bdf0-801f02d322c9/rtabmap-rgbd_odometry-1*.log
[rtabmap/rtabmap-2] process has died [pid 6539, exit code -11, cmd /home/ubuntu/catkin_ws/devel/lib/rtabmap_ros/rtabmap --delete_db_on_start rgb/image:=/camera/rgb/image_rect_color depth/image:=/camera/depth_registered/image_raw rgb/camera_info:=/camera/rgb/camera_info __name:=rtabmap __log:=/home/ubuntu/.ros/log/e5302dea-24fd-11e5-bdf0-801f02d322c9/rtabmap-rtabmap-2.log].
log file: /home/ubuntu/.ros/log/e5302dea-24fd-11e5-bdf0-801f02d322c9/rtabmap-rtabmap-2*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

matlabbe, 2015-11-28:
Similar to: 
 * http://official-rtab-map-forum.67519.x6.nabble.com/Rtabmap-error-in-process-rtabmap-rgbd-odometry-1-ubuntu-on-raspberry-pi-td444.html
 * http://answers.ros.org/question/220482/rtapmap-nodes-crash-on-raspi-2/

closed 2016-02-24 by matlabbe

matlabbe, 2020-04-08:
cross-ref, see debugging instructions here: https://github.com/introlab/rtabmap_ros/issues/28

lygods, 2023-03-30:
I have encountered the same problem. How did you solve it?