introlab/rtabmap_ros #21: Not able to see all options on GUI

rs1990, 2015-07-09:
Hello, I am not able to see all the options on the top of the GUI. All i see is a pause button. I used sudo apt-get install to install it on my system. 

[image]

and solution?

matlabbe, 2015-07-09:
On ROS, the core is running in [rtabmap](http://wiki.ros.org/rtabmap_ros#rtabmap) node. So Start/Stop actions are not used in the GUI (as well as New/Open database). Only pause action is still here, which calls `/rtabmap/pause` service under the hood.

Well, just saw that on [rtabmapviz](http://wiki.ros.org/rtabmap_ros#rtabmapviz) documentation, the screenshot has "Pause on loop closure, ... , Mapping, Localization" buttons. In recent versions, these actions are in the Edit menu. I'll update the screenshot soon.

For you red screen, make sure to start the Kinect 1 to 3 meters from objects so the odometry can be computed.

rs1990, 2015-07-09:
Thank you so much for your prompt reply. You have done some really amazing
work with this package.

Cheers

Raghavendra.

closed 2016-02-24 by matlabbe