introlab/rtabmap_ros #24: when installing,  catkin_make error

Guhnoo, 2015-07-24:
Hi, when I followed section to install, I meet some problem in last .

[image]

I installed 'RTAB-Map standalone libraries' in ' catkin_ws/devel ' folder 
[image]

## problem
And, finally did  git clone  in catkin_ws .
But, When I typed $ catkine_make    ( I didn't check out from 'catkin_ws')
I got this error 
beta@alpa:~$ cd catkin_ws/
beta@alpa:~/catkin_ws$ git clone https://github.com/introlab/rtabmap_ros.git src/rtabmap_ros
Cloning into 'src/rtabmap_ros'...
remote: Counting objects: 3329, done.
remote: Compressing objects: 100% (45/45), done.
remote: Total 3329 (delta 18), reused 0 (delta 0), pack-reused 3284
Receiving objects: 100% (3329/3329), 2.24 MiB | 566.00 KiB/s, done.
Resolving deltas: 100% (2434/2434), done.
Checking connectivity... done.

beta@alpa:~/catkin_ws$ catkin_make
Base path: /home/beta/catkin_ws
Source space: /home/beta/catkin_ws/src
Build space: /home/beta/catkin_ws/build
Devel space: /home/beta/catkin_ws/devel
Install space: /home/beta/catkin_ws/install
####
#### Running command: "cmake /home/beta/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/beta/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/beta/catkin_ws/install -G Unix Makefiles" in "/home/beta/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/beta/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/beta/catkin_ws/devel;/opt/ros/indigo
-- This workspace overlays: /home/beta/catkin_ws/devel;/opt/ros/indigo
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/beta/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.14
-- BUILD_SHARED_LIBS is on
/opt/ros/indigo/share/catkin/cmake/em/order_packages.cmake.em:23: error: <type 'exceptions.RuntimeError'>: Multiple packages found with the same name "rtabmap":
- rtabmap
- share/rtabmap
Traceback (most recent call last):
  File "/usr/bin/empy", line 3276, in <module>
    if __name__ == '__main__': main()
  File "/usr/bin/empy", line 3274, in main
    invoke(sys.argv[1:])
  File "/usr/bin/empy", line 3257, in invoke
    interpreter.wrap(interpreter.file, (file, name))
  File "/usr/bin/empy", line 2262, in wrap
    self.fail(e)
  File "/usr/bin/empy", line 2253, in wrap
    callable(*args)
  File "/usr/bin/empy", line 2326, in file
    self.safe(scanner, done, locals)
  File "/usr/bin/empy", line 2368, in safe
    self.parse(scanner, locals)
  File "/usr/bin/empy", line 2388, in parse
    token.run(self, locals)
  File "/usr/bin/empy", line 1403, in run
    interpreter.execute(self.code, locals)
  File "/usr/bin/empy", line 2565, in execute
    exec(statements, self.globals)
  File "<string>", line 17, in <module>
  File "/usr/lib/python2.7/dist-packages/catkin_pkg/topological_order.py", line 111, in topological_order
    for path, package in find_packages(space).items():
  File "/usr/lib/python2.7/dist-packages/catkin_pkg/packages.py", line 93, in find_packages
    raise RuntimeError('\n'.join(duplicates))
RuntimeError: Multiple packages found with the same name "rtabmap":
- rtabmap
- share/rtabmap
CMake Error at /opt/ros/indigo/share/catkin/cmake/safe_execute_process.cmake:11 (message):
  execute_process(/home/beta/catkin_ws/build/catkin_generated/env_cached.sh
  "/usr/bin/python" "/usr/bin/empy" "--raw-errors" "-F"
  "/home/beta/catkin_ws/build/catkin_generated/order_packages.py" "-o"
  "/home/beta/catkin_ws/build/catkin_generated/order_packages.cmake"
  "/opt/ros/indigo/share/catkin/cmake/em/order_packages.cmake.em") returned
  error code 1
Call Stack (most recent call first):
  /opt/ros/indigo/share/catkin/cmake/em_expand.cmake:25 (safe_execute_process)
  /opt/ros/indigo/share/catkin/cmake/catkin_workspace.cmake:35 (em_expand)
  CMakeLists.txt:63 (catkin_workspace)

-- Configuring incomplete, errors occurred!
See also "/home/beta/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/beta/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

what is this error???   I looked for this issue:  https://github.com/introlab/rtabmap_ros/issues/11
because I thought It is similar . but , I think it isn't 

Please, give me a answer

matlabbe, 2015-07-24:
rtabmap libraries should not be extracted in catkin/devel space. By "install in catkin/devel" I mean the CMAKE_INSTALL_PREFIX below. From the step one of [rtabmap_ros installation instructions](https://github.com/introlab/rtabmap_ros#build-from-source):
```
$ cd ~
$ git clone https://github.com/introlab/rtabmap.git rtabmap
$ cd rtabmap/build
$ cmake -DCMAKE_INSTALL_PREFIX=~/catkin_ws/devel ..  [<---double dots included]
$ make -j4
$ make install
```
rtabmap is extracted in Home folder, then on "make install", it is installed in catkin_ws/devel.

Guhnoo, 2015-07-25:
Ah, Thanks. I didn't understand what you mean well.

I do 
$ sudo apt-get remove ros-indigo-rtabmap ros-indigo-rtabmap 

and than reinstall .    the install is successful

closed 2016-02-24 by matlabbe