introlab/rtabmap_ros #25: Can't call "get_map" service

atomala, 2015-07-25:
I am using the latest Indigo release of RTABmap on Ubuntu 14.04. I ran RTABmap from a pre-existing database and tried to use the 'Download all Clouds' command in RTABmapviz but I only got an error. I can download graphs perfectly fine.

Here is the launch file I used: 

```
<launch>

   <arg name="frame_id" default="kinect2_base" />

    <node name="rtabmap" pkg="rtabmap_ros" type="rtabmap" output="screen" args="">
	  <param name="frame_id" type="string" value="$(arg frame_id)"/>

      <remap from="proj_map" to="/map"/>

      <param name="tf_prefix" type="string" value="vehicle"/>
      <param name="publish_tf" type="bool" value="true"/>

      <param name="Mem/IncrementalMemory" type="string" value="true"/>

    </node>

    <node pkg="rtabmap_ros" type="rtabmapviz" name="rtabmapviz"  output="screen">
      <param name="frame_id" type="string" value="$(arg frame_id)"/>
    </node>
</launch>
```

And here is the output I get when I pause the service, download the graph data, and then download the map data:

```
alex@alex-laptop:~/ROS/src/launch$ roslaunch local_rtabmap.launch
... logging to /home/alex/.ros/log/cec56ca8-32f1-11e5-8da1-c48e8ff4a00b/roslaunch-alex-laptop-19663.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://alex-laptop:43979/

SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.13
 * /rtabmap/Mem/IncrementalMemory: true
 * /rtabmap/frame_id: kinect2_base
 * /rtabmap/publish_tf: True
 * /rtabmap/tf_prefix: vehicle
 * /rtabmapviz/frame_id: kinect2_base

NODES
  /
    rtabmap (rtabmap_ros/rtabmap)
    rtabmapviz (rtabmap_ros/rtabmapviz)

auto-starting new master
process[master]: started with pid [19675]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to cec56ca8-32f1-11e5-8da1-c48e8ff4a00b
process[rosout-1]: started with pid [19688]
started core service [/rosout]
process[rtabmap-2]: started with pid [19705]
process[rtabmapviz-3]: started with pid [19706]
[ INFO] [1437844985.221548051]: Starting node...
[ INFO] [1437844985.317657646]: rtabmap: frame_id = kinect2_base
[ INFO] [1437844985.317705186]: rtabmap: map_frame_id = map
[ INFO] [1437844985.317724247]: rtabmap: queue_size = 10
[ INFO] [1437844985.317743505]: rtabmap: tf_delay = 0.050000
[ INFO] [1437844985.335569252]: Starting node...
[ INFO] [1437844985.521916064]: rtabmapviz: Using configuration from "/home/alex/.ros/rtabmapGUI.ini"
[ INFO] [1437844985.706925858]: Setting RTAB-Map parameter "Mem/IncrementalMemory"="true"
[ INFO] [1437844986.378141088]: Reading parameters from the ROS server...
[ INFO] [1437844986.668470062]: RTAB-Map rate detection = 1.000000 Hz
[ INFO] [1437844986.668513407]: rtabmap: Using database from "/home/alex/.ros/rtabmap.db".
[ INFO] [1437844986.768564986]: Parameters read = 179
[ INFO] [1437844986.768599848]: Parameters successfully read.
[ INFO] [1437844986.820685509]: 
/rtabmapviz subscribed to:
   
[ INFO] [1437844986.853173899]: rtabmapviz started.
[ INFO] [1437844988.820492928]: Registering Depth callback...
[ INFO] [1437844988.821466688]: 
/rtabmap subscribed to:
   /rgb/image,
   /depth/image,
   /rgb/camera_info,
   /odom
[ INFO] [1437844988.821663102]: rtabmap started...
[ INFO] [1437844999.137004869]: rtabmap: paused!
[ INFO] [1437845009.742197705]: rtabmap: Getting map (global=true optimized=true graphOnly=true)...
[ INFO] [1437845013.259199502]: rtabmap: Getting map (global=true optimized=true graphOnly=false)...
[ WARN] [1437845013.739410023]: Can't call "get_map" service
[ERROR] (2015-07-25 13:23:33.744) MainWindow.cpp:1898::processRtabmapEvent3DMap() Map received with code error 1!
[rtabmap-2] process has died [pid 19705, exit code -11, cmd /opt/ros/indigo/lib/rtabmap_ros/rtabmap proj_map:=/map __name:=rtabmap __log:=/home/alex/.ros/log/cec56ca8-32f1-11e5-8da1-c48e8ff4a00b/rtabmap-2.log].
log file: /home/alex/.ros/log/cec56ca8-32f1-11e5-8da1-c48e8ff4a00b/rtabmap-2*.log

```

matlabbe, 2015-07-26:
I tried with the latest binaries for Indigo and I don't have your problem (the clouds of the previous session are correctly shown). See below my output. In your case, "rtabmap" node seems to crash when loading node data from the database. This explains why rtabmapviz cannot call "get_map" service as the rtabmap node is busy (and finally crashes). 

Is it a database created from the same rtabmap version? and when the database was first created, rtabmap exited normally? If so, try if the standalone can load the clouds from the database:
```
$ rtabmap
```
File -> "Open database..." and select `~/.ros/rtabmap.db`, then Edit->"Download all Clouds".

```
$ roslaunch test.launch 
... logging to /home/mathieu/.ros/log/2a93741a-33da-11e5-9602-58b03561ac71/roslaunch-mathieu-MacBookPro-1451.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://mathieu-MacBookPro:47180/

SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.13
 * /rtabmap/Mem/IncrementalMemory: true
 * /rtabmap/frame_id: base_link
 * /rtabmap/publish_tf: True
 * /rtabmap/tf_prefix: vehicle
 * /rtabmapviz/frame_id: base_link

NODES
  /
    rtabmap (rtabmap_ros/rtabmap)
    rtabmapviz (rtabmap_ros/rtabmapviz)

auto-starting new master
process[master]: started with pid [1463]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 2a93741a-33da-11e5-9602-58b03561ac71
process[rosout-1]: started with pid [1476]
started core service [/rosout]
process[rtabmap-2]: started with pid [1493]
[ INFO] [1437944782.600114047]: Starting node...
process[rtabmapviz-3]: started with pid [1516]
[ INFO] [1437944782.653571565]: rtabmap: frame_id = base_link
[ INFO] [1437944782.653664911]: rtabmap: map_frame_id = map
[ INFO] [1437944782.653706062]: rtabmap: queue_size = 10
[ INFO] [1437944782.653758345]: rtabmap: tf_delay = 0.050000
[ INFO] [1437944782.832241270]: Setting RTAB-Map parameter "Mem/IncrementalMemory"="true"
[ INFO] [1437944782.878467966]: Starting node...
[ INFO] [1437944783.010909452]: rtabmapviz: Using configuration from "/home/mathieu/.ros/rtabmapGUI.ini"
[ INFO] [1437944783.327790918]: RTAB-Map rate detection = 1.000000 Hz
[ INFO] [1437944783.327872240]: rtabmap: Using database from "/home/mathieu/.ros/rtabmap.db".
[ WARN] (2015-07-26 17:06:23.408) Memory.cpp:319::init() _vwd->getUnusedWordsSize() must be empty... size=34
[ INFO] [1437944783.824370890]: Reading parameters from the ROS server...
[ INFO] [1437944783.993093350]: Registering Depth callback...
[ INFO] [1437944783.997333934]: 
/rtabmap subscribed to:
   /rgb/image,
   /depth/image,
   /rgb/camera_info,
   /odom
[ INFO] [1437944783.997742969]: rtabmap started...
[ INFO] [1437944784.038171432]: Parameters read = 179
[ INFO] [1437944784.038255961]: Parameters successfully read.
[ INFO] [1437944784.103117613]: 
/rtabmapviz subscribed to:
   
[ INFO] [1437944784.112952839]: rtabmapviz started.
[ INFO] [1437944786.655224010]: rtabmap: paused!
[ INFO] [1437944792.836083254]: rtabmap: Getting map (global=true optimized=true graphOnly=true)...
[ INFO] [1437944797.162813599]: rtabmap: Getting map (global=true optimized=true graphOnly=false)...
```

atomala, 2015-07-26:
I am using the '5MappingSessions_OnlineConstraint700ms.db.zip' database found on [this site.](https://github.com/introlab/rtabmap/wiki/Multi-session)

Trying it on the standalone RTABmap, I noticed that I can download the cloud but it does not show up on the 3d view for some weird reason. The graph data renders perfectly fine though.

matlabbe, 2015-07-27:
Make sure that "Show 3D clouds" under Map is checked in Preferences->3D Rendering. Well, I tried the database above and it worked on my computer. However, it takes a lot of memory to show full global map with the latest indigo version (6.9 Gig on my computer). On the latest version from source, rtabmap takes 4 Gig for the same database. 

How much memory does your computer have? You can try to download only the "local map" to see if you can at least have the point clouds. When clicking "Download all clouds", select local map instead.

atomala, 2015-07-27:
Using the local map doesn't work either. For some reason  the cloud is not being rendered in the 3d view. When I enable the grid in the 3d view, it still remains black. Changing the background color also doesn't cause anything to happen to the 3d view.

atomala, 2015-07-27:
Just compiled rtabmap_ros (and rtabmap) from source and it gets the data perfectly fine now. Only problem is the black 3d view. It receives the data but doesn't render it in rtabmap_viz

natejgardner, 2016-12-20:
I encounter this error whenever I try to download clouds from a large database. If I spend too much time scanning a room, especially if I run postprocessing, I always get this error. Usually any more than 1,000 loop closures will produce this for me in rtabmap_ros.

matlabbe, 2016-12-22:
Which error exactly do you have? Download error, rosservice call failed or optimization failed error?

"Usually any more than 1,000 loop closures will produce", looks like an optimization error problem (g2o/GTSAM/TORO failing to optimize the graph).