introlab/rtabmap_ros #26: Runnig Mapping_mode  Problem.

Guhnoo, 2015-08-02:
Hi,  I have a problem in running this pkg.
It ran successfully in last week. However, Today It doesn't work after updating  ubuntu software.
(Well, there are Nvidia bug issue after updating ubuntu software)
But, I thought It is not problem because I run it with  OpenCL method of intel GPU.

Following progress is what problem I face.
 
## $ roslaunch kinect2_bridge kinect2_bridge.launch publish_tf:=true

The terminal show me :

beta@alpa:~$ roslaunch kinect2_bridge kinect2_bridge.launch publish_tf:=true
... logging to /home/beta/.ros/log/6e7ea4d2-38db-11e5-8524-fcf8aed93681/roslaunch-alpa-21732.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.0.15:33664/

SUMMARY
========

PARAMETERS
 * /kinect2_bridge/base_name: kinect2
 * /kinect2_bridge/base_name_tf: kinect2
 * /kinect2_bridge/bilateral_filter: True
 * /kinect2_bridge/calib_path: /home/beta/catkin...
 * /kinect2_bridge/depth_device: -1
 * /kinect2_bridge/depth_method: default
 * /kinect2_bridge/edge_aware_filter: True
 * /kinect2_bridge/fps_limit: -1.0
 * /kinect2_bridge/jpeg_quality: 90
 * /kinect2_bridge/max_depth: 12.0
 * /kinect2_bridge/min_depth: 0.1
 * /kinect2_bridge/png_level: 1
 * /kinect2_bridge/publish_tf: True
 * /kinect2_bridge/queue_size: 5
 * /kinect2_bridge/reg_devive: -1
 * /kinect2_bridge/reg_method: default
 * /kinect2_bridge/sensor: 
 * /kinect2_bridge/use_png: False
 * /kinect2_bridge/worker_threads: 4
 * /kinect2_points_xyzrgb_hd/queue_size: 5
 * /kinect2_points_xyzrgb_qhd/queue_size: 5
 * /kinect2_points_xyzrgb_sd/queue_size: 5
 * /rosdistro: indigo
 * /rosversion: 1.11.13

NODES
  /
    kinect2 (nodelet/nodelet)
    kinect2_bridge (nodelet/nodelet)
    kinect2_points_xyzrgb_hd (nodelet/nodelet)
    kinect2_points_xyzrgb_qhd (nodelet/nodelet)
    kinect2_points_xyzrgb_sd (nodelet/nodelet)

ROS_MASTER_URI=http://192.168.0.15:11311

core service [/rosout] found
process[kinect2-1]: started with pid [21750]
[ INFO] [1438495195.108948436]: Initializing nodelet with 4 worker threads.
process[kinect2_bridge-2]: started with pid [21770]
[ INFO] [1438495195.140693280]: Loading nodelet /kinect2_bridge of type kinect2_bridge/kinect2_bridge_nodelet to manager kinect2 with the following remappings:
process[kinect2_points_xyzrgb_sd-3]: started with pid [21786]
process[kinect2_points_xyzrgb_qhd-4]: started with pid [21826]
parameter:
        base_name: kinect2
           sensor: 
        fps_limit: -1
       calib_path: /home/beta/catkin_ws/src/iai_kinect2/kinect2_bridge/data/
          use_png: false
     jpeg_quality: 90
        png_level: 1
     depth_method: default
     depth_device: -1
       reg_method: default
       reg_devive: -1
        max_depth: 12
        min_depth: 0.1
       queue_size: 5
 bilateral_filter: true
edge_aware_filter: true
       publish_tf: true
     base_name_tf: kinect2
   worker_threads: 4

[OpenCLDepthPacketProcessor::listDevice]  devices:
  0: Intel(R) HD Graphics Haswell GT2 Mobile (GPU)[Intel]
[OpenCLDepthPacketProcessor::init]  selected device: Intel(R) HD Graphics Haswell GT2 Mobile (GPU)[Intel]
[OpenCLDepthPacketProcessor::buildProgram] building OpenCL program...
process[kinect2_points_xyzrgb_hd-5]: started with pid [21855]
[OpenCLDepthPacketProcessor::buildProgram] OpenCL program built successfully
[OpenCLDepthPacketProcessor::buildProgram] building OpenCL program...
[OpenCLDepthPacketProcessor::buildProgram] OpenCL program built successfully
[Freenect2Impl] enumerating devices...
[Freenect2Impl] 12 usb devices connected
[Freenect2Impl] found valid Kinect v2 @4:3 with serial 029754145047
[Freenect2Impl] found 1 devices
Kinect2 devices found: 
  0: 029754145047 (selected)
[Freenect2DeviceImpl] opening...
[Freenect2DeviceImpl] opened

starting kinect2

[Freenect2DeviceImpl] starting...
[Freenect2DeviceImpl] ReadData0x14 response
92 bytes of raw data
0x0000:  00 00 12 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 ef 00 17 40 00 08 00 00 10 00 00 00    2e2133552e2e2e402e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 05 14 0d 47 4d 46 39 31 32 2e 31    31332e2e2e2e2e2e474d463931322e31
0x0050:  58 00 00 00 00 00 00 00 07 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  01 26 00 00                                        2e262e2e

[Freenect2DeviceImpl] ReadStatus0x090000 response
4 bytes of raw data
0x0000:  03 26 00 00                                        2e262e2e

[Freenect2DeviceImpl] enabling usb transfer submission...
[Freenect2DeviceImpl] submitting usb transfers...
[Freenect2DeviceImpl] started

device serial: 029754145047
device firmware: 4.3.3913.0.7
[Freenect2DeviceImpl] stopping...
[Freenect2DeviceImpl] disabling usb transfer submission...
[Freenect2DeviceImpl] canceling usb transfers...
[DepthPacketStreamParser::onDataReceived] not all subsequences received 0
[DepthPacketStreamParser::onDataReceived] not all subsequences received 512
[Freenect2DeviceImpl] stopped

default ir camera parameters: 
fx 365.82, fy 365.82, cx 261.954, cy 208.26
k1 0.0911889, k2 -0.271671, p1 0, p2 0, k3 0.0936603

default color camera parameters: 
fx 1081.37, fy 1081.37, cx 959.5, cy 539.5

camera parameters used:
camera matrix color:
[1133.923395373093, 0, 949.3852251462348;
  0, 1137.933434837629, 565.8910492688069;
  0, 0, 1]
distortion coefficients color:
[-0.2164195095026075, 6.731872045007637, -0.008686524459920842, -0.03039441404873483, -58.89171124284388]
camera matrix ir:
[2390.838493523019, 0, 255.7450083441651;
  0, 2229.278843545982, 211.6842679979474;
  0, 0, 1]
distortion coefficients ir:
[16.26194620894292, -2761.688219848951, 0.2450612211659857, -0.004496948418098325, -7.600562335179906]
rotation:
[0.9583645909446891, 0.002431295431940966, -0.2855370372227418;
  0.02268510740073345, 0.9961549079269922, 0.08462142350009744;
  0.2846448607047667, -0.08757561427411331, 0.9546244366549997]
translation:
[1.574080529774825; -0.4623861872280801; -4.385956440506297]
depth shift:
-13.3542

[DepthRegistration::New] Using OpenCL registration method!
[DepthRegistration::New] Using OpenCL registration method!
[DepthRegistrationOpenCL::listDevice]  devices:
  0: Intel(R) HD Graphics Haswell GT2 Mobile (GPU)[Intel]
[DepthRegistrationOpenCL::init]  selected device: Intel(R) HD Graphics Haswell GT2 Mobile (GPU)[Intel]
[DepthRegistrationOpenCL::listDevice]  devices:
  0: Intel(R) HD Graphics Haswell GT2 Mobile (GPU)[Intel]
[DepthRegistrationOpenCL::init]  selected device: Intel(R) HD Graphics Haswell GT2 Mobile (GPU)[Intel]
[kinect2_bridge] waiting for clients to connect

and 
## $ roslaunch rtabmap_ros rgbd_mapping_kinect2.launch
[image]

[ INFO] [1438496189.893945590]: Odom: quality=0, std dev=0.000000m, update time=0.031826s
[ INFO] [1438496189.926260029]: Odom: quality=0, std dev=0.000000m, update time=0.028664s
[ INFO] [1438496189.966557442]: Odom: quality=0, std dev=0.000000m, update time=0.031023s
[ INFO] [1438496190.004746849]: Odom: quality=0, std dev=0.000000m, update time=0.034221s
[ INFO] [1438496190.038229040]: Odom: quality=0, std dev=0.000000m, update time=0.031758s
[ INFO] [1438496190.077097860]: Odom: quality=0, std dev=0.000000m, update time=0.033900s
[ INFO] [1438496190.110425978]: Odom: quality=0, std dev=0.000000m, update time=0.032299s
^C[ INFO] [1438496190.141507145]: Odom: quality=0, std dev=0.000000m, update time=0.030088s
[ WARN] [1438496190.149635015]: Could not get transform from odom to kinect2_base_link after 1 second!
[rtabmap/rtabmapviz-4] killing on exit
[rtabmap/rtabmap-3] killing on exit
[rtabmap/visual_odometry-2] killing on exit
[kinect2_base_link-1] killing on exit
[ERROR] (2015-08-02 15:16:30.293) Rtabmap.cpp:874::process() RGB-D SLAM mode is enabled and no odometry is provided. Image 1793 is ignored!
rtabmap: Saving database/long-term memory... (located at /home/beta/.ros/rtabmap.db)
[ WARN] [1438496191.175695063]: Could not get transform from odom to kinect2_base_link after 1 second!
[ INFO] [1438496191.175920443]: rtabmapviz: All done! Closing...
shutting down processing monitor...
... shutting down processing monitor complete
done

Do I need to reinstall ubuntu? or How can I fix it?

matlabbe, 2015-08-02:
The odometry node seems to receive images but it is lost (quality=0). Can you see a cloud when executing this after launching `kinect2_bridge.launch`?
```
$ roslaunch kinect2_bridge kinect2_bridge.launch publish_tf:=true
$ rosrun kinect2_viewer kinect2_viewer kinect2 sd cloud
```

Guhnoo, 2015-08-07:
Thanks. That 's problem from wrong calibration.  I retried calibration and successfully executed it

closed 2016-02-24 by matlabbe