introlab/rtabmap_ros #27: CMake error on version issue

Megacephalo, 2015-08-04:
Hi! 
Thank you guys for this awesome work. About one month ago, RTABMap can be fully compiled and run like a charm under both ROS Hydro and Indigo installed on both computers that I'm using. However, the compilation starts to run into problem lately with error message such as follows:

    CMake Error at rtabmap_ros/CMakeLists.txt:20 (find_package):
    Could not find a configuration file for package "RTABMap" that is
    compatible with requested version "0.10.4".
    The following configuration files were considered but not accepted:

        /home/beebot/hydroBeebot_ws/devel/lib/rtabmap-0.9/RTABMapConfig.cmake, version: 0.9.0

 Well, basically the entire compilation message looks like this:

	Base path: /home/beebot/hydroBeebot_ws
	Source space: /home/beebot/hydroBeebot_ws/src
	Build space: /home/beebot/hydroBeebot_ws/build
	Devel space: /home/beebot/hydroBeebot_ws/devel
	Install space: /home/beebot/hydroBeebot_ws/install
	####
	#### Running command: "make cmake_check_build_system" in "/home/beebot/hydroBeebot_ws/build"
	####
	-- Using CATKIN_DEVEL_PREFIX: /home/beebot/hydroBeebot_ws/devel
	-- Using CMAKE_PREFIX_PATH: /home/beebot/hydroBeebot_ws/devel;/opt/ros/hydro
	-- This workspace overlays: /home/beebot/hydroBeebot_ws/devel;/opt/ros/hydro
	-- Using PYTHON_EXECUTABLE: /usr/bin/python
	-- Python version: 2.7
	-- Using Debian Python package layout
	-- Using CATKIN_ENABLE_TESTING: ON
	-- Call enable_testing()
	-- Using CATKIN_TEST_RESULTS_DIR: /home/beebot/hydroBeebot_ws/build/test_results
	-- Found gtest sources under '/usr/src/gtest': gtests will be built
	-- catkin 0.5.90
	-- BUILD_SHARED_LIBS is on
	WARNING: Package "libg2o" does not follow the version conventions. It should not contain leading zeros (unless the number is 0).
	-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
	-- ~~  traversing 6 packages in topological order:
	-- ~~  - mobile_safeguard_primitive
	-- ~~  - simulation_tobot
	-- ~~  - teleop_twist_keyboard
	-- ~~  - my_pcl_tutorial
	-- ~~  - tobotdrivers
	-- ~~  - rtabmap_ros
	-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
	-- +++ processing catkin package: 'mobile_safeguard_primitive'
	-- ==> add_subdirectory(NTU_BeeBot/mobile_safeguard_primitive)
	-- +++ processing catkin package: 'simulation_tobot'
	-- ==> add_subdirectory(NTU_BeeBot/simulation_tobot)
	-- Using these message generators: gencpp;genlisp;genpy
	-- simulation_tobot: 0 messages, 1 services
	-- +++ processing catkin package: 'teleop_twist_keyboard'
	-- ==> add_subdirectory(NTU_BeeBot/teleop_tobot)
	-- Using these message generators: gencpp;genlisp;genpy
	-- teleop_twist_keyboard: 0 messages, 3 services
	-- +++ processing catkin package: 'my_pcl_tutorial'
	-- ==> add_subdirectory(my_pcl_tutorial)
	-- Using these message generators: gencpp;genlisp;genpy
	-- +++ processing catkin package: 'tobotdrivers'
	-- ==> add_subdirectory(NTU_BeeBot/Tobotdriver)
	-- Using these message generators: gencpp;genlisp;genpy
	-- tobotdrivers: 0 messages, 1 services
	-- +++ processing catkin package: 'rtabmap_ros'
	-- ==> add_subdirectory(rtabmap_ros)
	-- Using these message generators: gencpp;genlisp;genpy
	CMake Warning at rtabmap_ros/CMakeLists.txt:16 (find_package):
	  Could not find module Findoctomap_ros.cmake or a configuration file for
	  package octomap_ros.

	  Adjust CMAKE_MODULE_PATH to find Findoctomap_ros.cmake or set
	  octomap_ros_DIR to the directory containing a CMake configuration file for
	  octomap_ros.  The file will have one of the following names:

	    octomap_rosConfig.cmake
	    octomap_ros-config.cmake

	CMake Error at rtabmap_ros/CMakeLists.txt:20 (find_package):
	  Could not find a configuration file for package "RTABMap" that is
	  compatible with requested version "0.10.4".

	  The following configuration files were considered but not accepted:

	    /home/beebot/hydroBeebot_ws/devel/lib/rtabmap-0.9/RTABMapConfig.cmake, version: 0.9.0

	-- rtabmap_ros: 7 messages, 6 services
	-- WITH costmap_2d
	-- Configuring incomplete, errors occurred!
	make: *** [cmake_check_build_system] Error 1
	Invoking "make cmake_check_build_system" failed

Although I have updated to the newest version of both the rtabmap library and the rtabmap_ros package, the problem persists. Is it possible to update to the newest version of 0.10? Or the error is due to other causes? Thank you in advance !

matlabbe, 2015-08-04:
Make sure you did the `$ make install` (see [Update to new version](https://github.com/introlab/rtabmap_ros#update-to-new-version)) for the rtabmap library to be correctly  installed in catkin workspace, so `/home/beebot/hydroBeebot_ws/devel/lib/rtabmap-0.10/RTABMapConfig.cmake` can then be found by CMake.

Megacephalo, 2015-08-05:
Hi! 
OK! I fixed the problem. First I noticed that I forgot to change the workspace directory as shown in your instruction ```cmake -DCMAKE_INSTALL_PREFIX=~/catkin_ws/devel .. ``` to the directory of mine. Second, under the path ```/home/beebot/hydroBeebot_ws/devel/lib``` I could only find the folder "rtabmap-0.9", and thus I deleted it and started the installation again. Probably if I were to noticed that I didn't switch to the right directory for version update I wouldn't have to do so much work. And voila! Ready to go! Thank you for your help !

Megacephalo, 2015-08-05:
I tried to fix the same problem on my other computer running on ROS Indigo. I still need to delete the rtabmap file altogether and reinstall to update to the new version. The git version update mechanism does not seem to work to update rtabmap from v.0.9 to 0.10. I wonder why.

closed 2016-02-24 by matlabbe