introlab/rtabmap_ros #28: Debugging "process has died" errors on startup

matlabbe, 2015-08-07:
# A : Build from source
If you are using the binaries (e.g., ros-indigo-rtabmap and ros-indigo-rtabmap-ros):

1. Uninstall them:
  ```bash
$ sudo apt-get remove ros-indigo-rtabmap ros-indigo-rtabmap-ros
```

2. Install `rtabmap` and `rtabmap_ros` from source following this [instructions](https://github.com/introlab/rtabmap_ros#build-from-source). 

# B : Build in Debug mode
If **A** doesn't work, clean up your catkin workspace and build rtabmap in Debug mode:
 
1. Remove the binaries: 
  ```bash
$ sudo apt-get remove ros-indigo-rtabmap ros-indigo-rtabmap-ros
```

2. Clean your catkin workspace (assuming your catkin workspace is `~/catkin_ws`, just keep `src` folder):
 ```bash
$ cd ~/catkin_ws
$ rm -r build devel install
```

3. Build `rtabmap` in **debug mode** (should **not be cloned in catkin workspace**):
 ```bash
$ cd ~
$ git clone https://github.com/introlab/rtabmap.git rtabmap
$ cd rtabmap/build
$ cmake -DCMAKE_INSTALL_PREFIX=~/catkin_ws/devel -DCMAKE_BUILD_TYPE=Debug .. 
$ make -j4
$ make install
```

4. Build `rtabmap_ros` in **debug mode**:
 ```bash
$ cd ~/catkin_ws
$ git clone https://github.com/introlab/rtabmap_ros.git src/rtabmap_ros
$ catkin_make -DCMAKE_BUILD_TYPE=Debug
```

5. Source your `setup.bash`
 ```bash
$ source ~/catkin_ws/devel/setup.bash
```

6. Launch rgbd_mapping.launch
 ```bash
$ roslaunch rtabmap_ros rgbd_mapping.launch
```

7. Now, if rtabmap is still crashing, start the [launch file with gdb](http://wiki.ros.org/roslaunch/Tutorials/Roslaunch%20Nodes%20in%20Valgrind%20or%20GDB) to track down where it is crashing:
 ```bash
$ roslaunch rtabmap_ros rgbd_mapping.launch launch_prefix:="xterm -e gdb -ex run --args"
```

8. Open an issue here with the log of gdb for the nodes crashing

jiminotron, 2015-08-08:
i follow this tutorial and get a log file:
=========================================================================
can not access memory at address 0x0

matlabbe, 2015-08-08:
Do **every** nodes (rtabmap, rgbd_odometry, rtabmapviz) produce this error? Is there a line where it crashes in the code? Can you show the full log for each node?

A fast search on Google for this error and this what I've found so far. It seems related to some modifications of the argc and argv variables (memory stack of the application corrupted). It seems happening on 32 bits architecture and on specific gcc version. Are you 32bits or 64bits? Which gcc version do you have (`gcc --version`)?
* [Bug with Qt](http://www.qtcentre.org/threads/11513-Cannot-access-memory-at-address-0x0)
* [General case](https://forums.opensuse.org/showthread.php/415438-argc-Cannot-access-memory-at-address-0x0): solution may be [here](https://bugzilla.novell.com/show_bug.cgi?id=509527#c1)

matlabbe, 2016-06-02:
Make sure you source the `setup.bash` in your `catkin_ws/devel` folder:
http://answers.ros.org/question/143496/roslaunch-is-neither-a-launch-file-in-package-nor-is-a-launch-file-name/

KamalDSOberoi, 2016-07-28:
What if the node doesn't crash on startup but crashes after some time on ROS Indigo?

willdzeng, 2016-07-28:
@KamalDSOberoi 
What is the crash error?

KamalDSOberoi, 2016-07-29:
@willdzeng 
There is no error. It just says that the process has died. In my case map_optimizer crashes/dies frequently but not on startup. It executes for sometime and then it dies.

gabrielarpino, 2016-08-01:
Fixed it by sourcing the setup file in the catkin_ws. Thanks!

matlabbe, 2016-08-01:
@KamalDSOberoi 
Are you using a modified version of `map_optimizer` node?

KamalDSOberoi, 2016-08-02:
@matlabbe I think no. Did you make significant changes?

matlabbe, 2016-08-02:
I just made a simple test launch file [test_map_optimizer.launch](https://github.com/introlab/rtabmap_ros/blob/master/launch/tests/test_map_optimizer.launch) for [map_optimizer](https://github.com/introlab/rtabmap_ros/blob/master/src/MapOptimizerNode.cpp) node. I cannot reproduce the crash. This error can appear on start though when the camera stands still:
```bash
[ERROR] [1470151850.812122935]: map_optimizer: Poses=2 and edges=0 (poses must not be null if there are edges, and edges must be null if poses <= 1)
```
But disappear when the camera starts moving.

matlabbe, 2022-12-04:
# ROS2 + GDB

```bash
ros2 launch rtabmap_ros rtabmap.launch.py rtabmap_args:="-d --udebug" launch_prefix:="xterm -e gdb -ex run --args"
```

Or if you use a Node confuguration in your launch file:
```py
Node(
  package='rtabmap_ros', executable='rtabmap', output="screen",
  arguments="-d --udebug",
  prefix="xterm -e gdb -ex run --args",
  ...
)