introlab/rtabmap_ros #29: Best way to add custom node

atomala, 2015-08-17:
I am working on a custom version of the grid map assembler node found in rtabmap_ros and I am wondering if there is any preferred method to do so.

I can just modify the current node but that would end up being a fork of the rtabmap_ros project, which I prefer not to do. I was thinking about making the node its own project but I noticed that the line  `#include "rtabmap_ros/MsgConversion.h"` fails because the header file is only found in the rtabmap_ros source files and not in the build directory.

Any help would be appreciated.

matlabbe, 2015-08-17:
You can create the modified node in your project. The current [GridMapAssemblerNode.cpp](https://github.com/introlab/rtabmap_ros/blob/master/src/GridMapAssemblerNode.cpp) is just a ROS wrapper over the rtabmap methods. Here an example using the downloaded file above in a new package named `custom_grid_assembler`:
```bash
$ cd ~/catkin_ws/src
$ catkin_create_pkg custom_grid_assembler rtabmap_ros
$ cd custom_grid_assembler
$ mkdir src
$ cd src
$ cp ~/Downloads/GridMapAssemblerNode.cpp custom_grid_assembler_node.cpp
```
In the `CMakeLists.txt`, uncommented these lines:
```cmake
## Declare a cpp executable
add_executable(custom_grid_assembler_node src/custom_grid_assembler_node.cpp)

## Specify libraries to link a library or executable target against
target_link_libraries(custom_grid_assembler_node
  ${catkin_LIBRARIES}
)
```
Then build:
```bash
$ cd ~/catkin_ws
$ catkin_make
```

If rtabmap and rtabmap_ros are built from source, you should slightly modify the `CMakeLists.txt`:
```cmake

find_package(RTABMap REQUIRED)
find_package(OpenCV REQUIRED)

## Specify additional locations of header files
## Your package locations should be listed before other locations
# include_directories(include)
include_directories(
  ${RTABMap_INCLUDE_DIRS}
  ${OpenCV_INCLUDE_DIRS}
  ${catkin_INCLUDE_DIRS}
)

## Declare a cpp executable
add_executable(custom_grid_assembler_node src/custom_grid_assembler_node.cpp)

## Specify libraries to link a library or executable target against
 target_link_libraries(custom_grid_assembler_node
   ${RTABMap_LIBRARIES}
   ${OpenCV_LIBRARIES}
   ${catkin_LIBRARIES}
 )
```
On my computer (Indigo), RTAB-Map is built against OpenCV built from source (to have access to SIFT and SURF), the `OpenCV_LIBRARIES` variable above should be before `catkin_LIBRARIES` to avoid an error like : `/usr/local/lib/libopencv_nonfree.so.2.4.11: undefined reference to 'cv::ocl::integral(cv::ocl::oclMat const&, cv::ocl::oclMat&)'`

cheers

atomala, 2015-08-17:
Thanks for the help. Following these steps worked perfectly. I am guessing I somehow screwed up the `CMakeList.txt` file when I tried doing it on my own.

closed 2015-08-17 by atomala