introlab/rtabmap_ros #30: error: ‘ros::message_traits::rtabmap_ros’ has not been declared when catkin_make

lligen, 2015-08-27:
**rtabmap** is installed correctly.
 when **catkin_make**
```javascript
 ~~errer 1~~. /home/lligen/catkin_ws/build/rtabmap_ros/src/rviz/moc_InfoDisplay.cxx:39:27: error: ‘ros::message_traits::rtabmap_ros’ has not been declared.  
 ~~errer 2~~./home/lligen/catkin_ws/src/rtabmap_ros/src/OdometryROS.cpp:384:91: error: invalid initialization of reference of type ‘const std::multimap<int, pcl::PointXYZ>&’ from expression of type ‘const std::map<int, pcl::PointXYZ>’  const std::multimap<int, pcl::PointXYZ> & map = ((OdometryBOW*)odometry_)->getLocalMap();
```

matlabbe, 2015-08-27:
What is your system?

lligen, 2015-08-28:
ubuntu 14.04, ros indigo, 
rtabmap 0.10.4-indigo, rtabmap_ros 0.10.4-indigo.

matlabbe, 2015-08-28:
The line 384 of OdometryROS.cpp ([there](https://github.com/introlab/rtabmap_ros/blob/indigo-devel/src/OdometryROS.cpp#L384) on the indigo-devel 0.10.4 branch): 
`const std::multimap<int, pcl::PointXYZ> & map = ((OdometryBOW*)odometry_)->getLocalMap();` 
should be 
`const std::map<int, pcl::PointXYZ> & map = ((OdometryBOW*)odometry_)->getLocalMap();` 
in the latest version. The corresponding [getLocalMap()](https://github.com/introlab/rtabmap/blob/indigo-devel/corelib/include/rtabmap/core/Odometry.h#L112) method returns a `std::map` too.

Well, you can also update to latest version on "master" branch. The indigo-devel branch is synced with the release binaries on ROS.

closed 2016-02-24 by matlabbe