introlab/rtabmap_ros #31: Map assembler taking current point cloud into account

Tobias-Fischer, 2015-09-02:
Hi @matlabbe,
First of all, thanks again for making RTAB-Map open source, it's a great piece of software :)!
I am playing around with the map assembler, and I was wondering whether it is possible to assemble the map data as it is implemented already, but on top of that merge the point cloud acquired by the Kinect.

I am thinking about concatenating the two point clouds, in another ROS node, but I am not sure whether there would be an easier way directly in the map assembler.

Thanks very much,
Tobias

Tobias-Fischer, 2015-09-02:
Hi,
I quickly put some code together which concatenates the point clouds, and that's working fine. However, obviously the clouds are overlapping, leading to the case that some objects are duplicated in the new cloud as they were moved around. So to refine my question, is it somehow possible to use the point cloud by the camera for the area currently perceived by the camera, and the assembled cloud for the other areas?

Thanks!

matlabbe, 2015-09-03:
Hi,

[pcl::CropHull](http://www.pcl-users.org/CropHull-filter-question-td4030345.html) filter could be used to "crop" the camera field of view from the assembled cloud of the map, then add the point cloud of the camera. The shape of the hull used to filter the point cloud would look like a [frustrum](https://en.wikipedia.org/wiki/Frustum) used in computer graphics.

Another way could be to filter cameras looking in the same direction as the current camera. This would be faster than filtering the whole point cloud, but I think the first approach would be more precise.

cheers

Tobias-Fischer, 2015-09-21:
Hi,

sorry for the delay in getting back to you. Thanks very much, I'll use CropHull to do the job :+1: 

Best, Tobias

closed 2015-09-21 by Tobias-Fischer

matlabbe, 2015-09-21:
I forgot to mention it, but last week I needed to do something like this... then I remembered there was already in PCL a filter called `pcl::FrustumCulling` class (`pcl/filters/frustum_culling.h`). I've made a wrapper function for convenience called [util3d::frustumFiltering(...)](https://github.com/introlab/rtabmap/blob/55834644cf64fcc3b638a03f05004872e92931b8/corelib/src/util3d_filtering.cpp#L182-L208). It is [integrated](https://github.com/introlab/rtabmap_ros/blob/96b7ba1f8802f0cb9c5a7b2c5c2f28cab2922a9a/src/MapsManager.cpp#L422-L429) with `/rtabmap/cloud_map` topic when ROS parameter `cloud_frustum_culling` is true for the `rtabmap` node.

cheers

Tobias-Fischer, 2015-09-21:
Cool, nice work! I'll have a look at this soon :).