introlab/rtabmap_ros #32: Undefined reference to EdgeSE2Switchable::EdgeSE2Switchable, etc.

jfstepha, 2015-09-04:
Hi,

I'm trying to build rtabmap from source, and I get the following errors when making:

    [ 87%] Building CXX object examples/BOWMapping/CMakeFiles/bow_mapping.dir/main.cpp.o
    ../../../bin/librtabmap_core.so.0.10.6: undefined reference to `EdgeSE2Switchable::EdgeSE2Switchable()'
    ../../../bin/librtabmap_core.so.0.10.6: undefined reference to `EdgeSwitchPrior::EdgeSwitchPrior()'
    ../../../bin/librtabmap_core.so.0.10.6: undefined reference to `VertexSwitchLinear::VertexSwitchLinear()'
    ../../../bin/librtabmap_core.so.0.10.6: undefined reference to `EdgeSE3Switchable::EdgeSE3Switchable()'
    collect2: error: ld returned 1 exit status 
    make[2]: *** [../bin/rtabmap-console] Error 1 
    make[1]: *** [tools/ConsoleApp/CMakeFiles/consoleApp.dir/all] Error 2
    make[1]: *** Waiting for unfinished jobs....
(It also has the same error when doing the linking on other files, I just included the first).

I got as far as figuring out that these are defined in corelib/src/vertigo/g2o/edge_se2Switchable.cpp, and referenced in Graph.cpp.  Graph.cpp does have `#include "vertigo/g2o/edge_se2Switchable.h"
` in it.

I suspect  edge_se2Switchable.cpp needs to be added to a CMakeLists somewhere, but don't know enough about how those work to know where.  Can anyone help?

matlabbe, 2015-09-04:
Thank you to report, 
this is now fixed in https://github.com/introlab/rtabmap/commit/f731b30cf9a03aa4d181b0ddf1e09df5348b6ec2 !

cheers

closed 2015-09-04 by matlabbe

jfstepha, 2015-09-04:
Excellent, thanks!