introlab/rtabmap_ros #33: rtabmap_ros catkin make fails with DSO missing from command line

jfstepha, 2015-09-04:


jfstepha, 2015-09-04:
When I try to catkin_make rtabmap_ros, it fails with:
`Linking CXX executable /home/jfstepha/catkin_ws/devel/lib/rtabmap_ros/rtabmapviz
    /usr/bin/ld: warning: libopencv_core.so.3.0, needed by /home/jfstepha/catkin_ws/devel/lib/librtabmap_core.so, may conflict with libopencv_core.so.2.4
    /usr/bin/ld: CMakeFiles/rtabmapviz.dir/src/GuiWrapper.cpp.o: undefined reference to symbol '_ZN2cv6String10deallocateEv'
    //usr/local/lib/libopencv_core.so.3.0: error adding symbols: DSO missing from command line
    collect2: error: ld returned 1 exit status
    make[2]: *** [/home/jfstepha/catkin_ws/devel/lib/rtabmap_ros/rtabmapviz] Error 1
    make[1]: *** [rtabmap_ros/CMakeFiles/rtabmapviz.dir/all] Error 2
    make: *** [all] Error 2
    make: INTERNAL: Exiting with 3 jobserver tokens available; should be 2!
`

I have less of a clue of how to fix this one.

matlabbe, 2015-09-05:
Mixing OpenCV 2 and OpenCV 3 can be problematic. ROS packages are using OpenCV 2. Three solutions: 
 1. Only use OpenCV installed by ROS. If you need SIFT or SURF, you may go with solution 2.
 2. Build OpenCV 2 from source and install in /usr/local (default). There should not be any incompatibilities with the OpenCV installed by ROS (the two can live side-by-side).
 3. If you really need OpenCV3, uninstall OpenCV 2 binaries. This will remove all ROS packages depending on it or indirectly through cv_bridge. You should build all these packages from source with your OpenCV 3 installation. I think cv_bridge can be built with OpenCV 3. So after that, everything will link on the same OpenCV libraries.

cheers,
Mathieu

jfstepha, 2015-09-05:
Excellent, thanks for your help.  I removed OpenCV3, and now have rtabmap_ros building.

closed 2016-02-24 by matlabbe