introlab/rtabmap_ros #34: unable to get pointcloud map for stereo only

gitunit, 2015-09-14:
hello @matlabbe . i was able to make the stereo example run but i only get the graph without a pointcloud. did i forget something or what could be the issue here?

this is what im seeing in rviz:

[image]

matlabbe, 2015-09-14:
Can you provide the link of the referred example? 

 1. Verify that your subscribed topic is published:
  ```bash
$ rostopic hz /rtabmap/assembled_cloud
```

 2. Try to subscribe to `/rtabmap/cloud_map` instead of `/rtabmap/assembled_cloud` (the second one is deprecated). 

 3. Depending on the camera calibration parameters (particularly focal length and baseline), the depth of the point cloud may be small or large. By default the point clouds generated have a maximum depth of 4 meters. What kind of camera do you use? In a indoor environment, there should be information under 4 meters. That maximum depth can changed with `cloud_max_depth` parameter (set to 0.0 to disable filtering) of [rtabmap](http://wiki.ros.org/rtabmap_ros#rtabmap) or [map_assembler](http://wiki.ros.org/rtabmap_ros#map_assembler) nodes. You can also check the point cloud generated by [stereo_image_proc](http://wiki.ros.org/stereo_image_proc), to know the size of the point cloud.

 4. You can also show up the [MapCloud](http://wiki.ros.org/rtabmap_ros#Map_Cloud_Display) display in RVIZ on the `/rtabmap/mapData` topic, which has parameters that you can set directly in RVIZ and is far less computationally expensive to update than `/rtabmap/cloud_map`.

gitunit, 2015-09-14:
thank you. i changed cloud_max_depth to 0 in MapCloud display in RVIZ. now there is finally a pointcloud visible:

[image]

still something seems incorrect to me, i expected to see the cloud over the ground floor (as in the video). also subscribing to `/rtabmap/cloud_map` did nothing, the console said: "Cloud map is empty! (clouds=65)" ..

oh, btw, i was referring to [demo_stereo_outdoor](https://github.com/introlab/rtabmap_ros/blob/master/launch/demo/demo_stereo_outdoor.launch) which i have modified (using the viso2 node for odometry) and im using the DUO3D stereo camera. how exactly would i check the size with the stereo_image_proc's pointcloud? with rviz measuring tool?

im on ubuntu 14.04. using ROS indigo. did you ever encounter any problems regarding RVIZ? whenever i use the default RVIZ version and load Info, MapCloud or MapGraph plugin, RVIZ crashes. it only works for me with a self compiled RVIZ version.

matlabbe, 2015-09-14:
The cloud generated on `/rtabmap/cloud_map` has also a maximum depth set to 4 meters. See the referred parameter `cloud_max_depth` on [rtabmap](http://wiki.ros.org/rtabmap_ros#rtabmap) node.

To check the size of the cloud, you can add a PointCloud2 display in RVIZ and subscribe to stereo_image_proc's pointcloud `/stereo_camera/points2`. You may have to set the `fixed_frame` of RVIZ to the camera link. By default, RVIZ shows a grid with cells 1 m x 1 m. Compare the point cloud to this grid and see if the scale is realistic from what the camera is currently seeing. If not, the calibration is wrong ([tutorial](http://wiki.ros.org/camera_calibration/Tutorials/StereoCalibration) for stereo calibration).

You may have to add a TF frame between base_link -> camera_link to rotate the optical axis of the camera (so to make the point cloud in world frame instead of camera frame). For example, see [bumblebee.launch](https://github.com/introlab/rtabmap_ros/blob/master/launch/tests/bumblebee.launch#L13-L16).

Another stereo example here ([stereo_mapping.launch](https://github.com/introlab/rtabmap_ros/blob/master/launch/stereo_mapping.launch)) for hand-held stereo mapping. I used it like the [Kinect](RGB-D Hand-held mapping with a Kinect) tutorial, but with a bumblebee camera:
 ```bash
$ roslaunch rtabmap_ros bumblebee.launch
$ roslaunch rtabmap_ros stereo_mapping.launch rtabmap_args:="--delete_db_on_start" rviz:=true rtabmapviz:=false
```

For the problems with RVIZ, are you using the rtabmap binaries or from source?

cheers

gitunit, 2015-09-14:
thx for the hints. i will try that.

regarding the RVIZ problem: im using rtabmap build from source. i could try installing the binaries alongside the current version and try if this happens there also. all i need is to source the main ROS directory to use the binary rtabmap_ros, right?

matlabbe, 2015-09-14:
Yes, if you only do `$ source /opt/ros/indigo/setup.bash ` and not  `$ source ~/catkin_ws/devel/setup.bash `, you will use the binaries. I will use RVIZ a lot in the next days, I'll see if I have your problem too.

gitunit, 2015-09-14:
ok i've tried it, the crash doesn't happen with the binaries.

matlabbe, 2015-09-23:
While playing with RVIZ over the last week, I didn't have any crash problem. This morning, I've updated ROS binary packages on Ubuntu, then rviz crashed after that. To solve the problem, I rebuilt all my source packages from scratch (removed `~/catkin_ws/build` directory), then the problem is solved:
 ```bash
$ cd ~/catkin_ws
$ rm -r build
$ catkin_make
```
I think this problem happens when the pluginlib package is updated, we should then explicitly rebuild all plugins.

gitunit, 2015-09-23:
that didn't help in my case. are you also on indigo?

matlabbe, 2015-09-23:
Yes (Ubuntu 14.04/ROS Indigo), and using the latest version built from source.

closed 2016-02-24 by matlabbe