introlab/rtabmap_ros #35: getting some tf error

shabhu18, 2015-10-01:
hii 
i am using zed stereo camera for stereo odometery.
The node for this stereo camera is https://github.com/Myzhar/ros_simple_zed_cuda_driver/blob/develop/src/zed_simple_camera_node.cpp

[image]
In this right hand side of image shows the list of topic published from camera.On the left side shows the launch file which i used for running stereo odometery.
After running this i am getting this error 
"Invalid argument passed to lookupTransform argument source_frame in tf2 frame_ids cannot be empty " 
.
[image]

My rqt_graph and tf is like this.
[image]

I am not able to understand the error need help.

matlabbe, 2015-10-01:
Hi,
Is the link ([zed_simple_camera_node.cpp](https://github.com/Myzhar/ros_simple_zed_cuda_driver/blob/develop/src/zed_simple_camera_node.cpp)) of the camera node is the right one? I see only left and depth topics published.

The error indicates that there are messages without header.frame_id defined. Make sure that your camera node publishes also the TF (e.g., /left_frame, /right_frame and/or /depth_frame linked to a /stereo_camera), which doesn't seem to do in your last image. 

cheers

shabhu18, 2015-10-06:
Yes the link is right for zed camera driver in ros . I have modified it to add right image . The frame ids were not added (you correctly pointed out mistake . thanks for it) i added it later .Still there is no data published on /odom topic .If i try to check the message rate it shows no value . Can u help . Here is a screenshot with image , camera info , and rqt_graph . thanks for help 
[image]

matlabbe, 2015-10-06:
If you are based on this file, I recommend to set the same timestamp for all your published messages (header.stamp). [stereo_odometry](http://wiki.ros.org/rtabmap_ros#stereo_odometry) node assumes by default (`approx_sync=false`) that all messages have exactly the same timestamp.

Another thing, is it your right camera_info shown on the right terminal, it seems missing the P(0,3) value (baseline). Is the camera calibrated using this [tutorial](http://wiki.ros.org/camera_calibration/Tutorials/StereoCalibration)?

cheers

Myzhar, 2015-10-06:
Hi all,
I'm the developer of the driver you are speaking about.
I've chosen to not publish right image because with ZED SDK you have depth
image "gratis"...

About tf, the driver does not publish it by default because I prefer to use
"state publisher" node for static frame (add the camera to your robot
descriptor).

However if you think that something must be changed in my code please feel
free change it and to make a pull request so i can update it... I wrote the
driver very quickly and I have not tested it with other ROS nodes, so I'm
sure that there is something wrong 😉

Thank you for your help

Cheers
Walter

matlabbe, 2015-10-07:
Thanks for the information. In this case, @shabhu18, you could use `rgbd_mapping.launch` instead of `stereo_mapping.launch` with left and depth images.

@Myzhar The [stereo_odometry](http://wiki.ros.org/rtabmap_ros#stereo_odometry) and [rgbd_odometry](http://wiki.ros.org/rtabmap_ros#rgbd_odometry) nodes need that the transform from the `frame_id` of the left/right/depth messages is linked to the fixed `frame_id` of the robot (like `base_link`). If the depth image is registered with the left image, you can set the `frame_id` of the depth messages as `left` directly (we will then not need any TF between the two cameras).

Example of a combined launch file:
```xml
<launch>
  <arg name="pi/2" value="1.5707963267948966" />
  <arg name="optical_rotate" value="0 0 0 -$(arg pi/2) 0 -$(arg pi/2)" />
  <node pkg="tf" type="static_transform_publisher" name="camera_base_link"
        args="$(arg optical_rotate) base_link left 100" />  

  <node pkg="ros_simple_zed_cuda_driver" type="zed_simple_camera_node" name="zed_simple_camera_node"/>

  <node pkg="rtabmap_ros" type="rgbd_odometry" name="rgbd_odometry" output="screen">
    <remap from="rgb/image"       to="stereo/left/image_rect"/>
    <remap from="depth/image"     to="stereo/depth/image_rect"/>
    <remap from="rgb/camera_info" to="stereo/left/camera_info"/>
  </node>
</launch>
```

cheers

Myzhar, 2015-10-07:
Hi matlabbe,
that's exactly what I meant.

Since Stereolabs SDK calculates depth map automatically using CUDA and
since RGB images, Disparity Map and Depth Map are automatically rectified
and synchronized its "easier" using ZED Camera similarly to an RGBD sensor,
so you do not spend calculation power to perform Stereo elaboration TWO
TIMES.

Furthermore, from what I verified, depth map calculated by Stereolabs SDK
is really really well done.

Finally, you do not need to calibrate the system by yourself using the
checkboard, Stereolabs provides intrinsic matrix... and also distortion and
extrinsic parameters, that you do not need since images are rectified.

Walt

shabhu18, 2015-10-07:
hii,
finally i edited the launch file for  stereo_odometry node with approx_sync=true.Added baseline in pixels in P matrix.(thanks for ur suggestion )  The node ran with no problem .The value of odometry i am getting i zero for both rotation and translation matrix .It is also not printing the number of inliners .
here is screenshot. Every value is zero . Can u help .
[image]

matlabbe, 2015-10-07:
The odometry cannot be computed for some reasons. Can you record a small rosbag (10 sec)? So I can try it here (I guessed your topics from your previous post):
```bash
$ rosbag record /stereo_camera/left/image_rect/compressed /stereo_camera/right/image_rect/compressed /stereo_camera/left/camera_info /stereo_camera/right/camera_info /tf
```

shabhu18, 2015-10-09:
I have recorded the bag file . Can u please check the error .No tf has been published in bag files. 

https://drive.google.com/file/d/0B5N_gJxysT-FNTE0Qlg1VWlHMGc/view?usp=sharing

Here the the file of the driver used 
https://drive.google.com/folderview?id=0B5N_gJxysT-FdlNRbkdBMXBiVm8&usp=sharing

Thank for all ur help.

matlabbe, 2015-10-09:
Thanks for the rosbag, I have done two examples:

### Stereo mapping
 
 * `test_stereo.launch`:
   ```xml
<launch>
  <arg name="pi/2" value="1.5707963267948966" />
  <arg name="optical_rotate" value="0 0 0 -$(arg pi/2) 0 -$(arg pi/2)" />
  <node pkg="tf" type="static_transform_publisher" name="camera_base_link"
        args="$(arg optical_rotate) base_link stereo_camera 100" /> 

  <param name="use_sim_time"  type="bool" value="true"/> <!-- only with rosbag -->

  <node pkg="rtabmap_ros" type="stereo_odometry" name="stereo_odometry" output="screen">
    <remap from="left/image_rect"        to="stereo_camera/left/image_rect"/>
    <remap from="right/image_rect"       to="stereo_camera/right/image_rect"/>
    <remap from="left/camera_info"       to="stereo_camera/left/camera_info"/>
    <remap from="right/camera_info"      to="stereo_camera/right/camera_info"/>
    <param name="approx_sync"          type="bool"   value="true"/>
    <param name="Odom/EstimationType"  type="string" value="1"/> <!-- 3D->2D PnP reprojection -->
  </node>

  <node pkg="rtabmap_ros" type="rtabmap" name="rtabmap" output="screen" args="--delete_db_on_start">
    <remap from="left/image_rect"        to="stereo_camera/left/image_rect"/>
    <remap from="right/image_rect"       to="stereo_camera/right/image_rect"/>
    <remap from="left/camera_info"  to="stereo_camera/left/camera_info"/>
    <remap from="right/camera_info" to="stereo_camera/right/camera_info"/>
    <param name="subscribe_depth"        type="bool"   value="false"/>
    <param name="subscribe_stereo"       type="bool"   value="true"/>
    <param name="stereo_approx_sync"            type="bool"   value="true"/>
    <param name="LccBow/EstimationType"  type="string" value="1"/> <!-- 3D->2D PnP reprojection -->
    <param name="Mem/SaveDepth16Format"  type="string" value="false"/> 
  </node>

  <node pkg="rtabmap_ros" type="rtabmapviz" name="rtabmapviz">
    <remap from="left/image_rect"        to="stereo_camera/left/image_rect"/>
    <remap from="right/image_rect"       to="stereo_camera/right/image_rect"/>
    <remap from="left/camera_info"  to="stereo_camera/left/camera_info"/>
    <remap from="right/camera_info" to="stereo_camera/right/camera_info"/>
    <param name="subscribe_stereo"       type="bool"   value="true"/>
    <param name="stereo_approx_sync"            type="bool"   value="true"/>
    <param name="subscribe_odom_info"    type="bool" value="true"/>
  </node>
</launch>
```
**The right camera_info must be modified with a baseline in meters**. So instead of -84000 for P(0,3), it should be -84 (to have a baseline of 0.12 m). Without that, the scale of the trajectory is huge (like the RGB-D example below).

Run:
 ```bash
$roslaunch test_stereo.launch
$rosbag play --clock 2015-10-08-18-33-17.bag
```

Results:
[image]
 

### RGB-D mapping
* `test_rgbd.launch`:
  ```xml
<launch>
  <arg name="pi/2" value="1.5707963267948966" />
  <arg name="optical_rotate" value="0 0 0 -$(arg pi/2) 0 -$(arg pi/2)" />
  <node pkg="tf" type="static_transform_publisher" name="camera_base_link"
        args="$(arg optical_rotate) base_link stereo_camera 100" /> 

  <param name="use_sim_time"  type="bool" value="true"/> <!-- only with rosbag -->

  <node pkg="rtabmap_ros" type="rgbd_odometry" name="rgbd_odometry" output="screen">
    <remap from="rgb/image"       to="stereo_camera/left/image_rect"/>
    <remap from="depth/image"     to="stereo_camera/depth/image_rect"/>
    <remap from="rgb/camera_info" to="stereo_camera/left/camera_info"/>
    <param name="Odom/EstimationType"  type="string" value="1"/> <!-- 3D->2D PnP reprojection -->
  </node>

  <node pkg="rtabmap_ros" type="rtabmap" name="rtabmap" output="screen" args="--delete_db_on_start">
    <remap from="rgb/image"       to="stereo_camera/left/image_rect"/>
    <remap from="depth/image"     to="stereo_camera/depth/image_rect"/>
    <remap from="rgb/camera_info" to="stereo_camera/left/camera_info"/>
    <param name="LccBow/EstimationType"  type="string" value="1"/> <!-- 3D->2D PnP reprojection -->
    <param name="Mem/SaveDepth16Format"  type="string" value="false"/> 
  </node>

  <node pkg="rtabmap_ros" type="rtabmapviz" name="rtabmapviz">
    <remap from="rgb/image"       to="stereo_camera/left/image_rect"/>
    <remap from="depth/image"     to="stereo_camera/depth/image_rect"/>
    <remap from="rgb/camera_info" to="stereo_camera/left/camera_info"/>
    <param name="subscribe_depth" type="bool" value="true"/>
    <param name="subscribe_odom_info" type="bool" value="true"/>
  </node>
</launch>
```
The depth format seems to be in mm instead of meters, which create large maps. You will even need to go in rtabmapviz->Preferences->3D Rendering, scroll down for Maximum depth **under Map column, set 0 m instead of 4 meters, otherwise you will not see the map's clouds**. The node generating depth images in 32 bits float format should be in meters (this is the default format used by openni, openni2 or freenect nodes).

Run:
  ```bash
$roslaunch test_rgbd.launch
$rosbag play --clock 2015-10-08-18-33-17.bag
```

Results:
[image]

Cheers,
Mathieu

theDrGray, 2015-10-17:
Is there a way to incorporate IMU into the RGBD odometry from the Zed example above?

matlabbe, 2015-10-17:
There is no direct way to do that in RTAB-Map, however, you can look at the [robot_pose_ekf](http://wiki.ros.org/robot_pose_ekf) node that can merge odometry from different sources.

shabhu18, 2015-10-19:
Thanks for all your help and support i am really grateful .

I am also getting same issue as in 
https://github.com/introlab/rtabmap_ros/issues/39 . Slight flicking it shows  red screen. How did you got such a good map  from bag. What parameters you set can you share . 
My map is showing just current point cloud ,Is it due to constant flicking odometery getting lost message.

matlabbe, 2015-10-19:
The bag contains not compatible depth format and camera_info. Some things to do for your node publishing the images (as explained above):
 * The P(0,3) value of the right camera info should be divided by 1000 (e.g., instead of -84000, it should be -84) to have the baseline in meters
 * The depth values should be divided by 1000 to have them in meters instead of mm for a CV32F format. You can also keep the values in mm, but change the depth format to CV16U.

For the parameters, they are those shown in the launch examples above. If you still have flickering problems, post a new bag with the modified camera_info and depth values.

theDrGray, 2015-10-19:
I was having the same issue. It turned out to be a USB issue that I am yet to track down. I believe it was the USB3 repeater I was using, but when I tried it again in the AM I no longer had an issue. What appeared to be happening to me was the Zed wasnt connecting to the computer properly and would lose frame because of under utilization of the GPU. 

If you check your GPU usage with nvidia-smi -l you should see 80+% load when the Zed is running. If thats not the case and youre using an extension of any sort or a USB port that isn't connected to a hub on the mobo (any front port on your computer case or a USB3 card plugged into the mobo via the connecting cable, not PCI) try plugging the Zed cable directly into a USB3 port on the mobo and give it a test.

closed 2015-12-03 by shabhu18