introlab/rtabmap_ros #37: IR+Depth only (no RGB-D) with rtabmap_ros?

toddsampson, 2015-10-14:
I am trying to get rtabmap_ros working with the Structure Sensor.  Unfortunately the Structure Sensor only has IR+Depth output -- no RGB-D.  I saw on the rtabmap release notes (https://github.com/introlab/rtabmap/releases) that it can now use an IR+Depth for the Kinect One.  Is there a way to configure rtabmap_ros to use just IR?

matlabbe, 2015-10-14:
If the Structure Sensor is based on Structured light (like Kinect for Xbox 360), RTAB-Map cannot use the IR image as it contains the IR pattern. Example:
[image]
Kinect for Xbox One uses time of light (TOF) technology, so the IR image is not disturbed by any semi-random dotted pattern and RTAB-Map can do visual odometry with it.

On the iPad, they seem to have calibrated the IR camera with iPad's RGB camera: http://forums.structure.io/t/announcing-calibrator/2037

Under ROS, if you want to calibrate an IR camera with an external RGB camera, you could look at this tutorial: http://wiki.ros.org/openni_launch/Tutorials/ExtrinsicCalibrationExternal
[image]

cheers

toddsampson, 2015-10-18:
Great stuff... Thanks!

closed 2015-10-18 by toddsampson

arogowiec, 2017-03-30:
Hi there! Can anyone tell me how to switch to IR+Depth mode?

matlabbe, 2017-03-31:
It is only available for Kinect v2:

[image]

[image]

nufanDK, 2017-12-09:
@toddsampson Did you get it working with a tethered Structure Sensor with no RGB? I'm trying the same just not on ros.. :-)