introlab/rtabmap_ros #39: Flickering/Losing features often

nanohaddad, 2015-10-17:
I'm currently using the following workflow: 
Zed Camera
Zed Ros Wrapper
Depth Image Proc
RTABMAP_ROS::RGBD_Mapping.launch

While I keep getting a ton of great features, the tracking constantly loses all features for a second and returns to normal. 

I implemented the Odom/ResetOdometery parameter to avoid complete loss of tracking if it loses features for more than two seconds- while that does help in avoiding a complete halt it definitely doesn't solve the flickering issue. 

My initial thought is that its due to the Zed Ros Wrapper being CPU bound, but I can't seem to get the CUDA GPU ZED wrapper working properly. 

Any thoughts or suggestions? 

It'd also be great if someone has some advice on fusing IMU data  with RGB-D Odometry

matlabbe, 2015-10-17:
Hi,

Is the tracking lost even when the camera is still or moved slowly? The section "Odometry Lost" of this [tutorial](https://github.com/introlab/rtabmap/wiki/Kinect-mapping) describes some events that can make odometry losing tracking.

Are you using the binaries? If yes, you could try the [latest version](https://github.com/introlab/rtabmap_ros/blob/master/launch/rgbd_mapping.launch) of rgbd_mapping.launch (which has slightly different default parameters).

Some other info about ZED camera in this [post](https://github.com/introlab/rtabmap_ros/issues/35).

If you can record a rosbag with the RGB images, depth images and RGB camera infos, I could see better if there is a bug that would make these flickering.

cheers

nanohaddad, 2015-10-17:
Hey,

Thanks for a prompt response!

Yes I am getting a flicker even when the camera is still, I followed the Odometry Lost section and that did help a ton, I ended up pointing it towards the ground at an angle which made a massive difference. 

The post that you mentioned was also greatly helpful, unfortunately I was not able to get the GPU version of the ZED working like the OP was able to, but i'm still trying. 

I'll send you that rosbag file ASAP, 
Thank you for your help!!

nanohaddad, 2015-10-17:
https://drive.google.com/file/d/0B7ZEq9mLUuJoU2N3LWgxQ2dFaTA/view?usp=sharing

Here's the ROS Bag, thanks again

matlabbe, 2015-10-17:
There is a problem with the depth. Its format is 16UC1, so rtabmap assumes that values are in mm (0->65535 mm). However the values inside the messages seem to be in meters (and truncated):
```bash
$ rostopic echo /camera/depth_registered/image_rect
header: 
  seq: 0
  stamp: 
    secs: 1445059029
    nsecs: 212834723
  frame_id: zed_optical_frame
height: 480
width: 640
encoding: 16UC1
is_bigendian: 0
step: 1280
data: [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 3, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 0, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 0, 0, 0, 0, 0, 0, 0, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 0, 0, 3, 0, 3, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 3, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 4, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 0, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, ...
```

To be able to see the point clouds in rtabmapviz, I had to scale the depth images X 1000, which gives this:
[image]

The commands I used with the rosbag:
```bash
$ rosbag filter 2015-10-16-22-17-08.bag output.bag "topic == \"/camera/rgb/image_rect\" or topic == \"/camera/rgb/camera_info\" or topic == \"/camera/depth_registered/image_rect\""
$ roscore
$ rosparam set use_sime_time true
$ roslaunch rtabmap_ros rgbd_mapping.launch rgb_topic:=/camera/rgb/image_rect depth_registered_topic:=/camera/depth_registered/image_rect frame_id:=zed_optical_frame
$ rosbag play --clock output.bag
```
Note that the rosbag filter is just because my laptop cannot process all topics of the original bag in real-time. I use also latest version of rgbd_mapping.launch (and modified the code to scale x1000 the depth).

cheers

nanohaddad, 2015-10-17:
Thank you so much for that! 
How'd you scale your settings x1000 properly?
Would it be better to give it mm or change the image type?

Thanks again

matlabbe, 2015-10-17:
There are no settings to scale x1000 (it was just a hard-coded test above). It should be done in your node publishing the depth image. 

Standard depth formats compatible with rtabmap:
 * 16UC1 : mm
 * 32FC1 : m

cheers

Myzhar, 2015-10-17:
To avoid that problem with the ZED you can change the value of the baseline
in the parameters of the camera.

As you can see the baseline is 12... assumed to be in mm.
If you change it in 0.12 you will have the depth map in meters.

I spoke about this issue with the people of StereoLabs and they said that
they will put a parameter in the SDK to choose the depth unit.

nanohaddad, 2015-10-17:
You're absolutely right Myzhar, but if you read above when we switch from mm to m, the image type must change. 
It would be extremely helpful if your package checked the baseline and modified the output accordingly.
Thanks for the response though

theDrGray, 2015-10-19:
rtabmap states that it is converting my images from 32FC to 16UC even though my units are in meters. I am also using the Zed camera and I have my baseline to 0.12. I know this doesnt pertain to the original issue, but wanted to ask while on the subject. Can open a new issue if that is preferred.

matlabbe, 2015-10-19:
Note that in version `0.10.10` (see [here](https://github.com/introlab/rtabmap_ros/blob/674fe3425574f84104bb016f9cb51abe328051a3/src/CoreWrapper.cpp#L807-L820)), the conversion is now optional through `Mem/SaveDepth16Format` parameter for the `rtabmap` node. By default, this parameter is `true` to reduce the size of the depth images saved in the database (16bits instead of 32bits). Set `Mem/SaveDepth16Format=false` to keep 32bits depth images if you have a depth resolution below 1 mm or if there are depth values over 65.545 meters.

The `rgbd_odometry` and `rtabmapviz` nodes use now either 16bits or 32bits without conversion.

theDrGray, 2015-10-19:
Awesome! Thanks so much. Will pull this down and give it a try. Really appreciate all the assistance.

theDrGray, 2015-10-20:
Has this been updated on aptitude? Tried to pull down and was given 10.04.

matlabbe, 2015-10-20:
0.10.10 binaries have just been released on the ROS buildfarm and are not public yet: http://www.ros.org/debbuild/indigo.html?q=rtabmap, the next Indigo release may happens in the next week. They can be installed through [shadow-fixed](http://wiki.ros.org/ShadowRepository) though.

nanohaddad, 2015-10-21:
Hey,

So we've tried to change some settings around, we're attempting to track these markers at high speeds, what are your thoughts on improvements? How can we perfect this high speed tracking? 
Thank you so much for your time! I've included the rosbag file If you have time to look at it. 
https://drive.google.com/file/d/0B7ZEq9mLUuJoNHJ4S3V2M0tJOFU/view?usp=sharing

matlabbe, 2015-10-21:
Hi,

I tried your bag. To be able to track all images, reducing the image rate is required. Example:
```bash
[Filter the bag to remove TF and other not used messages]
$ rosbag filter 2015-10-20-04-37-39.bag output.bag "topic == \"/stereo_camera/left/image_rect\" or topic == \"/stereo_camera/left/camera_info\" or topic == \"/stereo_camera/depth/image_rect\""
$ roscore
$ rosparam set use_sime_time true
$ roslaunch rtabmap_ros rgbd_mapping.launch rgb_topic:=/stereo_camera/left/image_rect depth_registered_topic:=/stereo_camera/depth/image_rect camera_info_topic:=/stereo_camera/left/camera_info frame_id:=stereo_camera rtabmap_args:="--delete_db_on_start" convert_depth_to_mm:=false estimation:=1 strategy:=0
$ rosbag play --clock -r 0.2 output.bag
```
Odometry uses between ~70 and ~130 ms to compute on my laptop.
[image]
Is the road should be straight? Note also the depth value shown in Odometry view, the nearest value is at 12 meters!? I checked in RVIZ and the cloud is huge (up to 66 meters) unless the camera is really high over the ground:
[image]
There are also a lot of depth values at -999 meters:
[image]

Note that if you could put a rotation on the camera to have the point cloud in RVIZ referential (x->forward, y->left, z->up instead of x->right, y->bottom, z->forward), it will be easier to use RVIZ or rtabmapviz:
```xml
<arg name="pi/2" value="1.5707963267948966" />
    <arg name="optical_rotate" value="0 0 0 -$(arg pi/2) 0 -$(arg pi/2)" />
  	<node pkg="tf" type="static_transform_publisher" name="camera_base_link"
        args="$(arg optical_rotate) base_link stereo_camera 100" />  
```

To increase Odometry speed, we can use OpticalFlow strategy (`strategy:=1`):
```bash
$ roslaunch rtabmap_ros rgbd_mapping.launch rgb_topic:=/stereo_camera/left/image_rect depth_registered_topic:=/stereo_camera/depth/image_rect camera_info_topic:=/stereo_camera/left/camera_info frame_id:=stereo_camera rtabmap_args:="--delete_db_on_start" convert_depth_to_mm:=false estimation:=1 strategy:=1
```
This gives around ~60 ms for odometry updates. However, the camera gets lost more often. There are some parameters that can be tuned for OpticalFlow:
```bash
$ rosrun rtabmap_ros rgbd_odometry --params | grep OdomFlow
Param: OdomFlow/Eps = "0.01"          [Used for optical flow approach. See cv::calcOpticalFlowPyrLK().]
Param: OdomFlow/Iterations = "30"     [Used for optical flow approach. See cv::calcOpticalFlowPyrLK().]
Param: OdomFlow/MaxLevel = "3"        [Used for optical flow approach. See cv::calcOpticalFlowPyrLK().]
Param: OdomFlow/WinSize = "16"        [Used for optical flow approach. See cv::calcOpticalFlowPyrLK().]

$ rosrun rtabmap_ros rgbd_odometry --params | grep GFTT
Param: GFTT/BlockSize = "3"                              
Param: GFTT/K = "0.04"                                    
Param: GFTT/MinDistance = "5"                            
Param: GFTT/QualityLevel = "0.001"                       
Param: GFTT/UseHarrisDetector = "false"       

$ rosrun rtabmap_ros rgbd_odometry --params | grep Odom/M
Param: Odom/MaxDepth = "0"                                 [Max depth of the words (0 means no limit).]
Param: Odom/MaxFeatures = "1000"                           [0 no limits.]
Param: Odom/MinInliers = "20"                              [Minimum visual word correspondences to compute geometry transform.]

$ rosrun rtabmap_ros rgbd_odometry --params | grep Odom/PnP
Param: Odom/PnPFlags = "1"                                 [PnP flags: 0=Iterative, 1=EPNP, 2=P3P]
Param: Odom/PnPReprojError = "5.0"                         [PnP reprojection error.]
           
```

cheers

nanohaddad, 2015-10-21:
Thank you so much! your help is greatly appreciated.

We're currently having an issue where the map returns elevation and 3d movement even though its locked to 2d, almost like barrel rolls in space. 

After using your parameters i'm having amazingly better loop closures, but i keep getting non-zero values for my initial position- even at the "grid" map's start. (monitoring odom values using rostopic echo /odom) 

Thanks again!

nanohaddad, 2015-10-21:
Our camera is pointed at negative 24 degrees, is there a proper parameter that allows rtabmap to take that into account? thank you!!

nanohaddad, 2015-10-21:
https://drive.google.com/file/d/0Bw2tbkskappKOTIxc25PRW0xb2M/view?usp=sharing
Here's a rosbag file, my last one i promise! Just really need to pin point this issue, it thinks im going in a corkscrew like motion, Im looking for a flat map. 

Just to clarify im trying to map a massive parking lot and get Odom out of it afterwards

matlabbe, 2015-10-21:
For the camera transform, it is related to TF. For example (using format `x y z yaw pitch roll`), you can set `/base_link` -> `/camera_link` = `2.170 0 -0.360 0 0.41 0` and `/camera_link` -> `/stereo_camera` = `0 0 0 -1.571 0.0 -1.571`. Note the `0.41` which is your 24 degrees looking down.
```bash
$ rosrun tf static_transform_publisher 0 0.00 0 -1.571 0.0 -1.571 camera_link stereo_camera 100
$ rosrun tf static_transform_publisher 2.170 0.00 -0.360 0 0.3 0 base_link camera_link 100
```
Then set rtabmap to use `/base_link` as the fixed `frame_id`:
```bash
$ rosbag filter 2015-10-21-10-10-34.bag output.bag "topic == \"/stereo_camera/left/image_rect\" or topic == \"/stereo_camera/left/camera_info\" or topic == \"/stereo_camera/depth/image_rect\""
$ roscore
$ rosparam set use_sim_time true
$ roslaunch rtabmap_ros rgbd_mapping.launch rgb_topic:=/stereo_camera/left/image_rect depth_registered_topic:=/stereo_camera/depth/image_rect camera_info_topic:=/stereo_camera/left/camera_info frame_id:=base_link rtabmap_args:="--delete_db_on_start" convert_depth_to_mm:=false estimation:=1 rtabmapviz:=false rviz:=true
$ rosbag play --clock output.bag
```
Note that I use RVIZ here instead of rtabmapviz to see better the TF tree.

The left images don't contain a lot of robust features to track (images are dark) and the depth images have some distortions. I assume that if you are mapping a parking, the cloud in RVIZ should look like a plane, not with large waves like this:
[image]
Not to mention there are many **large** negative depth values that could influence the odometry. Maybe why you have an odometry of 9000 meters in your recorded rosbag. However, I am running the latest version of rtabmap, the odometry scale seems ok (in scale of some meters). The depth here is the biggest issue. The [stereo outdoor tutorial](http://wiki.ros.org/rtabmap_ros/Tutorials/StereoOutdoorMapping) shows an example of depth data generated by a calibrated stereo camera:
[image]
The depth quality is better, but it is taken outdoor with a lot of texture.

cheers

closed 2016-02-24 by matlabbe