introlab/rtabmap_ros #40: cannot build rtabmapviz

nanohaddad, 2015-10-27:
On build from source, I get a string of the following errors toward the end of the build:

/home/testing/test/devel/lib/rtabmap-0.10/../../include/rtabmap-0.10/rtabmap/core/RtabmapEvent.h:197:7: note: rtabmap::RtabmapGlobalPathEvent::RtabmapGlobalPathEvent(const rtabmap::RtabmapGlobalPathEvent&)
 class RtabmapGlobalPathEvent : public UEvent
       ^
/home/testing/test/devel/lib/rtabmap-0.10/../../include/rtabmap-0.10/rtabmap/core/RtabmapEvent.h:197:7: note:   candidate expects 1 argument, 3 provided
Linking CXX executable /home/testing/test/devel/lib/rtabmap_ros/stereo_odometry
make[2]: *** [rtabmap_ros/CMakeFiles/rtabmapviz.dir/src/GuiWrapper.cpp.o] Error 1
make[1]: *** [rtabmap_ros/CMakeFiles/rtabmapviz.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[100%] Built target stereo_odometry
Linking CXX executable /home/testing/test/devel/lib/rtabmap_ros/rtabmap
[100%] Built target rtabmap
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

All of rtabmap works properly, but I am unable to call rtabmapviz. When I use a launch file that is calling it, I get the following:

ERROR: cannot launch node of type [rtabmap_ros/rtabmapviz]: can't locate node [rtabmapviz] in package [rtabmap_ros]

nanohaddad, 2015-10-27:
I am also unable to subscribe to the left/image_rect and right/image_rect topics when I launch rtabmap from source. Not sure if these issues are related. The same launch file works fine when I launch the apt-get version of RTAB. 

From the source build, the package launches without error, but does not show any results. The RVIZ template is pointing to image_rect_color. I changed the remap to left/image_rect from left/image_rect_color. Is this hard coded anywhere in RTAB by chance?

matlabbe, 2015-10-27:
For the build error, I've pushed the fix above. Thx to point out!

The stereo topic names are still the same (for rtabmap, rtabmapviz and stereo_odometry nodes). Example to remap the stereo topics:
```xml
<!-- subscribe_stereo only required for rtabmap and rtabmapviz nodes --> 
<param name="subscribe_stereo" type="bool" value="true"/>

<remap from="left/image_rect"   to="/stereo_camera/left/image_rect_color"/>
<remap from="right/image_rect"  to="/stereo_camera/right/image_rect"/>
<remap from="left/camera_info"  to="/stereo_camera/left/camera_info"/>
<remap from="right/camera_info" to="/stereo_camera/right/camera"/>
```

I tried again the [stereo tutorial](http://wiki.ros.org/rtabmap_ros/Tutorials/StereoOutdoorMapping) and it still works. What is your rqt_graph?

nanohaddad, 2015-10-28:
Thanks, now Im getting the following errors on build:

/home/testing/test/src/rtabmap_ros/src/CoreWrapper.cpp: In member function ‘bool CoreWrapper::setGoalCallback(rtabmap_ros::SetGoal::Request&, rtabmap_ros::SetGoal::Response&)’:
/home/testing/test/src/rtabmap_ros/src/CoreWrapper.cpp:1923:6: error: ‘rtabmap_ros::SetGoal::Response’ has no member named ‘planning_time’
  res.planning_time = planningTime;
      ^
[100%] Built target rgbd_odometry
/home/testing/test/src/rtabmap_ros/src/GuiWrapper.cpp: In member function ‘virtual void GuiWrapper::handleEvent(UEvent*)’:
/home/testing/test/src/rtabmap_ros/src/GuiWrapper.cpp:444:129: error: ‘rtabmap_ros::SetGoal::Response’ has no member named ‘planning_time’
     this->post(new RtabmapGlobalPathEvent(setGoalSrv.request.node_id, setGoalSrv.request.node_label, poses, setGoalSrv.response.planning_time));
                                                                                                                                 ^
make[2]: *** [rtabmap_ros/CMakeFiles/rtabmap.dir/src/CoreWrapper.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....

matlabbe, 2015-10-28:
Sorry, the change I changed in SetGoal service was missing in the previous commit.

cheers

closed 2016-02-24 by matlabbe