introlab/rtabmap_ros #41: Can't Install the RTAB-Map standalone libraries

772358053, 2015-11-13:
I install the RTAB-Map according to these steps:
https://github.com/introlab/rtabmap_ros#rtabmap_ros
```
$ cd ~
$ git clone https://github.com/introlab/rtabmap.git rtabmap
$ cd rtabmap/build
$ cmake -DCMAKE_INSTALL_PREFIX=~/catkin_ws/devel ..  [<---double dots included]
$ make -j4
$ make install
```

but when I run $ make -j4 the compilation starts to run into problem with error message such as follows:
```
siat@ThinkPad-siat:~/rtabmap/build$ make -j4
[  7%] Built target rtabmap_utilite
[  7%] Built target uresourcegenerator
[  8%] Built target imagesJoiner
[  9%] Built target extractObject
[ 10%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/CameraThread.cpp.o
[ 10%] [ 11%] [ 12%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/CameraRGBD.cpp.o
Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/VisualWord.cpp.o
Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/EpipolarGeometry.cpp.o
In file included from /usr/include/ni/XnOS.h:27:0,
                 from /usr/include/pcl-1.7/pcl/io/openni_camera/openni.h:47,
                 from /usr/include/pcl-1.7/pcl/io/openni_camera/openni_depth_image.h:44,
                 from /home/siat/rtabmap/corelib/src/../include/rtabmap/core/CameraRGBD.h:37,
                 from /home/siat/rtabmap/corelib/src/CameraRGBD.cpp:28:
/usr/include/ni/XnPlatform.h:73:3: error: #error OpenNI Platform Abstraction Layer - Unsupported Platform!
  #error OpenNI Platform Abstraction Layer - Unsupported Platform!
   ^
.
.
.
In file included from /usr/include/pcl-1.7/pcl/io/openni_camera/openni.h:50:0,
                 from /usr/include/pcl-1.7/pcl/io/openni_camera/openni_depth_image.h:44,
                 from /home/siat/rtabmap/corelib/src/../include/rtabmap/core/CameraRGBD.h:37,
                 from /home/siat/rtabmap/corelib/src/CameraThread.cpp:31:
/usr/include/ni/XnCppWrapper.h:10045:76: error: macro "XN_VALIDATE_NEW" passed 4 arguments, but takes just 2
    XN_VALIDATE_NEW(pTrans, StateChangedCallbackTranslator, handler, pCookie);
                                                                            ^
/usr/include/ni/XnCppWrapper.h:10104:75: error: macro "XN_VALIDATE_NEW" passed 4 arguments, but takes just 2
   XN_VALIDATE_NEW(pTrans, StateChangedCallbackTranslator, handler, pCookie);
                                                                           ^
[ 13%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/VWDictionary.cpp.o
In file included from /usr/include/ni/XnStatusCodes.h:27:0,
                 from /usr/include/ni/XnMacros.h:27,
                 from /usr/include/ni/XnOS.h:28,
                 from /usr/include/pcl-1.7/pcl/io/openni_camera/openni.h:47,
                 from /usr/include/pcl-1.7/pcl/io/openni_camera/openni_depth_image.h:44,
                 from /home/siat/rtabmap/corelib/src/../include/rtabmap/core/CameraRGBD.h:37,
                 from /home/siat/rtabmap/corelib/src/CameraRGBD.cpp:28:
/usr/include/ni/XnStatus.h:33:9: error: ‘XnUInt32’ does not name a type
 typedef XnUInt32 XnStatus;
         ^
```
Could anyone help me? I have no idea what's wrong with it.

772358053, 2015-11-18:
I solved this problem after reinstalling my system. Maybe the ni package in /usr/include was wrong.

closed 2016-02-24 by matlabbe

half-potato, 2016-07-06:
Do
`sudo dpkg -r --force-depends "libopenni-dev" && sudo apt-get install libopenni-dev`

Instead. More convenient.

Pine-sha, 2017-02-20:
http://official-rtab-map-forum.67519.x6.nabble.com/Compilitaion-td955.html
from the link ：
While trying compiling on Fedora, I got the openni Xn*** errors too. When building with G20 enabled, "-std=c++11" is added to compilation instructions. If I do "cmake -DWITH_G2O=OFF", "-std=c++11" is not used and it builds without errors.

so，you can test ：
cmake -DWITH_G2O=OFF ..

JinglinZhang, 2017-02-21:
@Pine-sha But g2o is needed in the optimization process.