introlab/rtabmap_ros #42: rtabmap_ros install problem and cant not run it with kinectV2

peterlin300, 2015-11-24:
First,i run sudo apt-get install ros-indigo-rtabmap-ros. but it fail,
it said can not install two part :ros-indigo-costmap-2d and  ros-indigo-pcl-ros
[image]
so, i try sudo apt-get install ros-indigo-rtabmap, its seem work successful！
i run  rtabmap inTerminal and try to select kinect2, but it cant be  selected.like this
[image]
i already install the libfreenect2 and run it successful before!!
[image]

what is this error????

matlabbe, 2015-11-24:
Hi,

This is strange that you cannot install `ros-indigo-costmap-2d` and `ros-indigo-pcl-ros`. Here what I have:
```bash
$ sudo apt-get install ros-indigo-rtabmap-ros
The following extra packages will be installed:
  ros-indigo-costmap-2d ros-indigo-pcl-ros ros-indigo-rtabmap
The following NEW packages will be installed:
  ros-indigo-costmap-2d ros-indigo-pcl-ros ros-indigo-rtabmap
  ros-indigo-rtabmap-ros
0 upgraded, 4 newly installed, 0 to remove and 582 not upgraded.
Need to get 0 B/5,882 kB of archives.
After this operation, 26.5 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Selecting previously unselected package ros-indigo-pcl-ros.
(Reading database ... 269343 files and directories currently installed.)
Preparing to unpack .../ros-indigo-pcl-ros_1.2.7-0trusty-20151111-0815-+0000_amd64.deb ...
Unpacking ros-indigo-pcl-ros (1.2.7-0trusty-20151111-0815-+0000) ...
Selecting previously unselected package ros-indigo-costmap-2d.
Preparing to unpack .../ros-indigo-costmap-2d_1.12.5-0trusty-20151111-0831-+0000_amd64.deb ...
Unpacking ros-indigo-costmap-2d (1.12.5-0trusty-20151111-0831-+0000) ...
Selecting previously unselected package ros-indigo-rtabmap.
Preparing to unpack .../ros-indigo-rtabmap_0.10.10-0trusty-20151111-1102-+0000_amd64.deb ...
Unpacking ros-indigo-rtabmap (0.10.10-0trusty-20151111-1102-+0000) ...
Selecting previously unselected package ros-indigo-rtabmap-ros.
Preparing to unpack .../ros-indigo-rtabmap-ros_0.10.10-0trusty-20151111-1118-+0000_amd64.deb ...
Unpacking ros-indigo-rtabmap-ros (0.10.10-0trusty-20151111-1118-+0000) ...
Setting up ros-indigo-pcl-ros (1.2.7-0trusty-20151111-0815-+0000) ...
Setting up ros-indigo-costmap-2d (1.12.5-0trusty-20151111-0831-+0000) ...
Setting up ros-indigo-rtabmap (0.10.10-0trusty-20151111-1102-+0000) ...
Setting up ros-indigo-rtabmap-ros (0.10.10-0trusty-20151111-1118-+0000) ...
Processing triggers for libc-bin (2.19-0ubuntu6.6) ...
```
Try `$ sudo apt-get update` before trying to install the package.

For the Kinect v2, it is because the binaries are not built with libfreenect2 dependency. But you should be able to use the Kinect v2 with `ros-indigo-rtabmap-ros` and the [iai_kinect2](https://github.com/code-iai/iai_kinect2) package like in this following [tutorial](http://wiki.ros.org/rtabmap_ros/Tutorials/HandHeldMapping).

If you want to use Kinect v2 driver in the standalone version of rtabmap, you should build rtabmap from source. CMake should detect your libfreenect2 installation and build with it.

cheers

peterlin300, 2015-11-26:
thanks for your advices!!!!!! i will try it again!

closed 2016-02-24 by matlabbe