introlab/rtabmap_ros #43: genmsg problem

Yvaine, 2016-01-16:
I have meet a problem caused by genmsg module when i run catkin_make in indigo version, the result is as follows:

Using these message generators: gencpp;genlisp;genpy
CMake Error: Could not open file for write in copy operation /home/ubuntu/catkin_ws/devel/share/rtabmap_ros/cmake/rtabmap_ros-msg-paths.cmake.tmp
CMake Error: : System Error: No such file or directory
CMake Error at /opt/ros/indigo/share/genmsg/cmake/genmsg-extras.cmake:215 (configure_file):
  configure_file Problem configuring file
Call Stack (most recent call first):
  rtabmap_ros/CMakeLists.txt:70 (generate_messages)

CMake Error: Could not open file for write in copy operation /home/ubuntu/catkin_ws/devel/share/rtabmap_ros/cmake/rtabmap_rosConfig.cmake.tmp
CMake Error: : System Error: No such file or directory
CMake Error at /opt/ros/indigo/share/catkin/cmake/catkin_package.cmake:375 (configure_file):
  configure_file Problem configuring file
Call Stack (most recent call first):
  /opt/ros/indigo/share/catkin/cmake/catkin_package.cmake:98 (_catkin_package)
  rtabmap_ros/CMakeLists.txt:98 (catkin_package)

CMake Error: Could not open file for write in copy operation /home/ubuntu/catkin_ws/devel/share/rtabmap_ros/cmake/rtabmap_rosConfig-version.cmake.tmp
CMake Error: : System Error: No such file or directory
CMake Error at /opt/ros/indigo/share/catkin/cmake/catkin_package.cmake:380 (configure_file):
  configure_file Problem configuring file
Call Stack (most recent call first):
  /opt/ros/indigo/share/catkin/cmake/catkin_package.cmake:98 (_catkin_package)
  rtabmap_ros/CMakeLists.txt:98 (catkin_package)

-- WITH costmap_2d
-- WITH octomap
-- Configuring incomplete, errors occurred!
See also "/home/ubuntu/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/ubuntu/catkin_ws/build/CMakeFiles/CMakeError.log".
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed

The problem is i can not find rtabmap_ros in my catkin_ws/devel/share/ folder, what's the matter?

matlabbe, 2016-01-16:
Hi,

The catkin build seems broken. Try cleaning your catkin workspace and build again. 
```bash
$ cd ~/catkin_ws
$ rm -r build devel

// To build rtabmap_ros, reinstall rtabmap standalone library in ~/catkin/devel
$ cd ~/rtabmap/build
$ make install

$ cd ~/catkin_ws
$ catkin_make
```

If it doesn't work, post the whole catkin_make output.

cheers,
Mathieu

Yvaine, 2016-01-17:
Thank you for your answer! I have try to clean my workspace, but also put out the same issue.The whole output as follows

~/catkin_ws$ catkin_make
Base path: /home/ubuntu/catkin_ws
Source space: /home/ubuntu/catkin_ws/src
Build space: /home/ubuntu/catkin_ws/build
Devel space: /home/ubuntu/catkin_ws/devel
Install space: /home/ubuntu/catkin_ws/install
####
#### Running command: "cmake /home/ubuntu/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/ubuntu/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/ubuntu/catkin_ws/install -G Unix Makefiles" in "/home/ubuntu/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/ubuntu/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/ubuntu/catkin_ws/devel;/opt/ros/indigo
-- This workspace overlays: /home/ubuntu/catkin_ws/devel;/opt/ros/indigo
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/ubuntu/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.16
-- BUILD_SHARED_LIBS is on
WARNING: Package "libg2o" does not follow the version conventions. It should not contain leading zeros (unless the number is 0).
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 1 packages in topological order:
-- ~~  - rtabmap_ros
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'rtabmap_ros'
-- ==> add_subdirectory(rtabmap_ros)
-- Using these message generators: gencpp;genlisp;genpy
-- Looking for Q_WS_X11
-- Looking for Q_WS_X11 - found
-- Looking for Q_WS_WIN
-- Looking for Q_WS_WIN - not found
-- Looking for Q_WS_QWS
-- Looking for Q_WS_QWS - not found
-- Looking for Q_WS_MAC
-- Looking for Q_WS_MAC - not found
-- Found Qt4: /usr/bin/qmake (found version "4.8.6") 
-- checking for module 'OGRE'
--   found OGRE, version 1.8.1
CMake Error: Could not open file for write in copy operation /home/ubuntu/catkin_ws/devel/share/rtabmap_ros/cmake/rtabmap_ros-msg-paths.cmake.tmp
CMake Error: : System Error: No such file or directory
CMake Error at /opt/ros/indigo/share/genmsg/cmake/genmsg-extras.cmake:215 (configure_file):
  configure_file Problem configuring file
Call Stack (most recent call first):
  rtabmap_ros/CMakeLists.txt:70 (generate_messages)

CMake Error: Could not open file for write in copy operation /home/ubuntu/catkin_ws/devel/share/rtabmap_ros/cmake/rtabmap_rosConfig.cmake.tmp
CMake Error: : System Error: No such file or directory
CMake Error at /opt/ros/indigo/share/catkin/cmake/catkin_package.cmake:375 (configure_file):
  configure_file Problem configuring file
Call Stack (most recent call first):
  /opt/ros/indigo/share/catkin/cmake/catkin_package.cmake:98 (_catkin_package)
  rtabmap_ros/CMakeLists.txt:98 (catkin_package)

CMake Error: Could not open file for write in copy operation /home/ubuntu/catkin_ws/devel/share/rtabmap_ros/cmake/rtabmap_rosConfig-version.cmake.tmp
CMake Error: : System Error: No such file or directory
CMake Error at /opt/ros/indigo/share/catkin/cmake/catkin_package.cmake:380 (configure_file):
  configure_file Problem configuring file
Call Stack (most recent call first):
  /opt/ros/indigo/share/catkin/cmake/catkin_package.cmake:98 (_catkin_package)
  rtabmap_ros/CMakeLists.txt:98 (catkin_package)

-- WITH costmap_2d
-- WITH octomap
-- Configuring incomplete, errors occurred!
See also "/home/ubuntu/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/ubuntu/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

Yvaine, 2016-01-17:
Hello,

I just have reboot my ubuntu14.04 system, and rebuild rtabmap-ros, the same problem of genmsg occur, this is the output:
$ catkin_make
Base path: /home/ubuntu/catkin_ws
Source space: /home/ubuntu/catkin_ws/src
Build space: /home/ubuntu/catkin_ws/build
Devel space: /home/ubuntu/catkin_ws/devel
Install space: /home/ubuntu/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/ubuntu/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/ubuntu/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/ubuntu/catkin_ws/devel;/opt/ros/indigo
-- This workspace overlays: /home/ubuntu/catkin_ws/devel;/opt/ros/indigo
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/ubuntu/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.16
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 4 packages in topological order:
-- ~~  - g2o
-- ~~  - beginner
-- ~~  - orb_slam
-- ~~  - rtabmap_ros
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'g2o'
-- ==> add_subdirectory(ORB_SLAM_Raulmur_Indigo/g2o)
-- BUILD TYPE:Release
-- Compiling on Unix
-- Found CHOLMOD and its dependencies
-- +++ processing catkin package: 'beginner'
-- ==> add_subdirectory(beginner)
-- +++ processing catkin package: 'orb_slam'
-- ==> add_subdirectory(ORB_SLAM_Raulmur_Indigo/orb_slam)
-- Using these message generators: gencpp;genlisp;genpy
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
--   date_time
--   iostreams
--   serialization
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- looking for PCL_COMMON
-- looking for PCL_OCTREE
-- looking for PCL_IO
-- looking for PCL_KDTREE
-- looking for PCL_SEARCH
-- looking for PCL_SAMPLE_CONSENSUS
-- looking for PCL_FILTERS
-- looking for PCL_FEATURES
-- looking for PCL_KEYPOINTS
-- looking for PCL_GEOMETRY
-- looking for PCL_SEGMENTATION
-- looking for PCL_VISUALIZATION
-- looking for PCL_OUTOFCORE
-- looking for PCL_REGISTRATION
[... 131 lines cut ...]
CMake Error at /opt/ros/indigo/share/gencpp/cmake/gencpp-extras.cmake:8 (file):
  file problem creating directory:
  /home/ubuntu/catkin_ws/devel/include/rtabmap_ros
Call Stack (most recent call first):
  /opt/ros/indigo/share/gencpp/cmake/gencpp-extras.cmake:34 (_generate_msg_cpp)
  /home/ubuntu/catkin_ws/build/rtabmap_ros/cmake/rtabmap_ros-genmsg.cmake:179 (_generate_srv_cpp)
  /opt/ros/indigo/share/genmsg/cmake/genmsg-extras.cmake:304 (include)
  rtabmap_ros/CMakeLists.txt:70 (generate_messages)

CMake Error at /opt/ros/indigo/share/gencpp/cmake/gencpp-extras.cmake:8 (file):
  file problem creating directory:
  /home/ubuntu/catkin_ws/devel/include/rtabmap_ros
Call Stack (most recent call first):
  /opt/ros/indigo/share/gencpp/cmake/gencpp-extras.cmake:34 (_generate_msg_cpp)
  /home/ubuntu/catkin_ws/build/rtabmap_ros/cmake/rtabmap_ros-genmsg.cmake:185 (_generate_srv_cpp)
  /opt/ros/indigo/share/genmsg/cmake/genmsg-extras.cmake:304 (include)
  rtabmap_ros/CMakeLists.txt:70 (generate_messages)

-- WITH costmap_2d
-- WITH octomap
-- Configuring incomplete, errors occurred!
See also "/home/ubuntu/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/ubuntu/catkin_ws/build/CMakeFiles/CMakeError.log".
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed

matlabbe, 2016-01-17:
Are you using the code from `master` git branch? Can you try `$ catkin_make -j1`?

matlabbe, 2016-01-17:
The line `CMake Error at /opt/ros/indigo/share/gencpp/cmake/gencpp-extras.cmake:8 (file):` is:
```cmake
file(MAKE_DIRECTORY ${ARG_GEN_OUTPUT_DIR})
```
or this if we replace the argument:
```cmake
file(MAKE_DIRECTORY /home/ubuntu/catkin_ws/devel/include/rtabmap_ros)
```

It seems that CMake does not have the permissions to create a directory in your home directory. Check on google `cmake MAKE_DIRECTORY "file problem creating directory"`.

If you create manually the `/home/ubuntu/catkin_ws/devel/include/rtabmap_ros` directory, will CMake fail farther?

Here is the normal output (Indigo Ubuntu 14.04, rtabmap_ros master branch):
```bash
~/catkin_ws: $ catkin_make -j1 --pkg rtabmap_ros
Base path: /home/mathieu/catkin_ws
Source space: /home/mathieu/catkin_ws/src
Build space: /home/mathieu/catkin_ws/build
Devel space: /home/mathieu/catkin_ws/devel
Install space: /home/mathieu/catkin_ws/install
####
#### Running command: "cmake /home/mathieu/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/mathieu/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/mathieu/catkin_ws/install -G Unix Makefiles" in "/home/mathieu/catkin_ws/build"
####
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Using CATKIN_DEVEL_PREFIX: /home/mathieu/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/mathieu/catkin_ws/devel;/opt/ros/indigo
-- This workspace overlays: /home/mathieu/catkin_ws/devel;/opt/ros/indigo
-- Found PythonInterp: /usr/bin/python (found version "2.7.6") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/mathieu/catkin_ws/build/test_results
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.16
-- BUILD_SHARED_LIBS is on
WARNING: Package "libg2o" does not follow the version conventions. It should not contain leading zeros (unless the number is 0).
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 2 packages in topological order:
-- ~~  - find_object_2d
-- ~~  - rtabmap_ros
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'find_object_2d'
-- ==> add_subdirectory(find_object_2d)
-- Looking for Q_WS_X11
-- Looking for Q_WS_X11 - found
-- Looking for Q_WS_WIN
-- Looking for Q_WS_WIN - not found
-- Looking for Q_WS_QWS
[... 171 lines cut ...]
[ 90%] Building CXX object rtabmap_ros/CMakeFiles/odom_msg_to_tf.dir/src/OdomMsgToTFNode.cpp.o
Linking CXX executable /home/mathieu/catkin_ws/devel/lib/rtabmap_ros/odom_msg_to_tf
[ 90%] Built target odom_msg_to_tf
Scanning dependencies of target rgbd_odometry
[ 92%] Building CXX object rtabmap_ros/CMakeFiles/rgbd_odometry.dir/src/RGBDOdometryNode.cpp.o
Linking CXX executable /home/mathieu/catkin_ws/devel/lib/rtabmap_ros/rgbd_odometry
[ 92%] Built target rgbd_odometry
Scanning dependencies of target rtabmap
[ 94%] Building CXX object rtabmap_ros/CMakeFiles/rtabmap.dir/src/CoreNode.cpp.o
[ 94%] Building CXX object rtabmap_ros/CMakeFiles/rtabmap.dir/src/CoreWrapper.cpp.o
Linking CXX executable /home/mathieu/catkin_ws/devel/lib/rtabmap_ros/rtabmap
[ 94%] Built target rtabmap
Scanning dependencies of target rtabmap_ros_generate_messages
[ 94%] Built target rtabmap_ros_generate_messages
Scanning dependencies of target rtabmapviz
[ 96%] Building CXX object rtabmap_ros/CMakeFiles/rtabmapviz.dir/src/GuiNode.cpp.o
[ 98%] Building CXX object rtabmap_ros/CMakeFiles/rtabmapviz.dir/src/GuiWrapper.cpp.o
[ 98%] Building CXX object rtabmap_ros/CMakeFiles/rtabmapviz.dir/src/PreferencesDialogROS.cpp.o
Linking CXX executable /home/mathieu/catkin_ws/devel/lib/rtabmap_ros/rtabmapviz
[ 98%] Built target rtabmapviz
Scanning dependencies of target stereo_odometry
[100%] Building CXX object rtabmap_ros/CMakeFiles/stereo_odometry.dir/src/StereoOdometryNode.cpp.o
Linking CXX executable /home/mathieu/catkin_ws/devel/lib/rtabmap_ros/stereo_odometry
[100%] Built target stereo_odometry
```

Yvaine, 2016-01-18:
Thank you very much! I find why the issue occur. When i install rtab_map, i use sudo install. Now i delete all, and reinstall not use sudo, the genmsg problem disappear. Now another problem occur.

Scanning dependencies of target _rtabmap_ros_generate_messages_check_deps_ResetPose
[ 47%] Built target _rtabmap_ros_generate_messages_check_deps_ResetPose
Scanning dependencies of target std_msgs_generate_messages_py
[ 47%] Built target std_msgs_generate_messages_py
Scanning dependencies of target geometry_msgs_generate_messages_py
[ 47%] Built target geometry_msgs_generate_messages_py
Scanning dependencies of target sensor_msgs_generate_messages_py
[ 47%] Built target sensor_msgs_generate_messages_py
Scanning dependencies of target _rtabmap_ros_generate_messages_check_deps_ListLabels
[ 47%] Built target _rtabmap_ros_generate_messages_check_deps_ListLabels
Scanning dependencies of target _rtabmap_ros_generate_messages_check_deps_Link
[ 47%] Built target _rtabmap_ros_generate_messages_check_deps_Link
Scanning dependencies of target _rtabmap_ros_generate_messages_check_deps_Point2f
[ 47%] Built target _rtabmap_ros_generate_messages_check_deps_Point2f
Scanning dependencies of target _rtabmap_ros_generate_messages_check_deps_NodeData
[ 47%] Built target _rtabmap_ros_generate_messages_check_deps_NodeData
Scanning dependencies of target _rtabmap_ros_generate_messages_check_deps_GetMap
[ 47%] Built target _rtabmap_ros_generate_messages_check_deps_GetMap
Scanning dependencies of target _rtabmap_ros_generate_messages_check_deps_Info
[ 47%] Built target _rtabmap_ros_generate_messages_check_deps_Info
Scanning dependencies of target _rtabmap_ros_generate_messages_check_deps_SetLabel
[ 47%] Built target _rtabmap_ros_generate_messages_check_deps_SetLabel
Scanning dependencies of target _rtabmap_ros_generate_messages_check_deps_KeyPoint
[ 47%] Built target _rtabmap_ros_generate_messages_check_deps_KeyPoint
Scanning dependencies of target _rtabmap_ros_generate_messages_check_deps_Goal
[ 47%] Built target _rtabmap_ros_generate_messages_check_deps_Goal
Scanning dependencies of target _rtabmap_ros_generate_messages_check_deps_PublishMap
[ 47%] Built target _rtabmap_ros_generate_messages_check_deps_PublishMap
Scanning dependencies of target _rtabmap_ros_generate_messages_check_deps_OdomInfo
[ 47%] Built target _rtabmap_ros_generate_messages_check_deps_OdomInfo
Scanning dependencies of target _rtabmap_ros_generate_messages_check_deps_SetGoal
[ 47%] Built target _rtabmap_ros_generate_messages_check_deps_SetGoal
Scanning dependencies of target _rtabmap_ros_generate_messages_check_deps_MapData
[ 47%] Built target _rtabmap_ros_generate_messages_check_deps_MapData
Scanning dependencies of target _rtabmap_ros_generate_messages_check_deps_MapGraph
[ 47%] Built target _rtabmap_ros_generate_messages_check_deps_MapGraph
Scanning dependencies of target rtabmap_ros_generate_messages_py
[ 47%] Generating Python from MSG rtabmap_ros/Goal
[ 48%] Generating Python from MSG rtabmap_ros/MapGraph
[ 50%] Generating Python from MSG rtabmap_ros/Info
[ 50%] Generating Python from MSG rtabmap_ros/OdomInfo
[ 51%] Generating Python from MSG rtabmap_ros/Link
[ 51%] Generating Python from MSG rtabmap_ros/KeyPoint
[ 52%] Generating Python from MSG rtabmap_ros/Point2f
[ 53%] Generating Python from MSG rtabmap_ros/NodeData
[ 53%] Generating Python from MSG rtabmap_ros/MapData
[ 54%] Generating Python code from SRV rtabmap_ros/GetMap
[ 54%] Generating Python code from SRV rtabmap_ros/PublishMap
[ 55%] Generating Python code from SRV rtabmap_ros/ResetPose
[ 56%] Generating Python code from SRV rtabmap_ros/SetLabel
[ 56%] Generating Python code from SRV rtabmap_ros/ListLabels
[ 57%] Generating Python code from SRV rtabmap_ros/SetGoal
[ 57%] Generating Python msg __init__.py for rtabmap_ros
[ 58%] Generating Python srv __init__.py for rtabmap_ros
[ 58%] Built target rtabmap_ros_generate_messages_py
Scanning dependencies of target rtabmap_ros_gencfg
[ 58%] Generating dynamic reconfigure files from cfg/Camera.cfg: /home/ubuntu/catkin_ws/devel/include/rtabmap_ros/CameraConfig.h /home/ubuntu/catkin_ws/devel/lib/python2.7/dist-packages/rtabmap_ros/cfg/CameraConfig.py
../catkin_generated/env_cached.sh: 16: exec: /home/ubuntu/catkin_ws/src/rtabmap_ros/cfg/Camera.cfg: Permission denied
make[2]: *** [/home/ubuntu/catkin_ws/devel/include/rtabmap_ros/CameraConfig.h] Error 126
make[1]: *** [rtabmap_ros/CMakeFiles/rtabmap_ros_gencfg.dir/all] Error 2
make: *** [all] Error 2

matlabbe, 2016-01-18:
Hi,
Yes, if you build/install `rtabmap` and `rtabmap_ros` from source, you **should not use** "sudo" anywhere (unless you want to install `rtabmap` library in default `/usr/local` instead of the recommended `~/catkin_ws/devel`). 

There are still files in your workspace from previous build trials that were created in sudo. You must remove them manually in sudo. I recommend to clean your `/home/ubuntu/catkin_ws/devel` directory so there are no more files installed in sudo. Make also sure that `rtabmap_ros` source code can be edited without sudoing too.

cheers,
Mathieu

Yvaine, 2016-01-18:
hello,
Thanks a lot. I just git clone once again. Compile success.

closed 2016-01-18 by matlabbe

yogiji-coder, 2020-04-29:
####
#### Running command: "cmake /home/yogeshwar/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/yogeshwar/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/yogeshwar/catkin_ws/install -G Unix Makefiles" in "/home/yogeshwar/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/yogeshwar/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/yogeshwar/catkin_ws/devel;/opt/ros/kinetic
-- This workspace overlays: /home/yogeshwar/catkin_ws/devel;/opt/ros/kinetic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.12", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/yogeshwar/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.12") 
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.20
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 1 packages in topological order:
-- ~~  - rtabmap_ros
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'rtabmap_ros'
-- ==> add_subdirectory(rtabmap_ros)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
CMake Warning at rtabmap_ros/CMakeLists.txt:28 (find_package):
  By not providing "Findapriltag_ros.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "apriltag_ros", but CMake did not find one.

  Could not find a package configuration file provided by "apriltag_ros" with
  any of the following names:

    apriltag_rosConfig.cmake
    apriltag_ros-config.cmake

  Add the installation prefix of "apriltag_ros" to CMAKE_PREFIX_PATH or set
  "apriltag_ros_DIR" to a directory containing one of the above files.  If
  "apriltag_ros" provides a separate development package or SDK, be sure it
  has been installed.

-- Found RTABMap: /home/yogeshwar/catkin_ws/devel/lib/librtabmap_core.so;/home/yogeshwar/catkin_ws/devel/lib/librtabmap_utilite.so;/home/yogeshwar/catkin_ws/devel/lib/librtabmap_gui.so;/usr/lib/i386-linux-gnu/libz.so;/opt/ros/kinetic/lib/libg2o_core.so;/opt/ros/kinetic/lib/libg2o_types_slam2d.so;/opt/ros/kinetic/lib/libg2o_types_slam3d.so;/opt/ros/kinetic/lib/libg2o_types_sba.so;/opt/ros/kinetic/lib/libg2o_stuff.so;/opt/ros/kinetic/lib/libg2o_solver_csparse.so;/opt/ros/kinetic/lib/libg2o_csparse_extension.so;/usr/lib/i386-linux-gnu/libcxsparse.so;/opt/ros/kinetic/lib/libg2o_solver_cholmod.so;/usr/lib/i386-linux-gnu/libcholmod.so;/usr/lib/i386-linux-gnu/libfreenect.so;/usr/lib/i386-linux-gnu/libfreenect_sync.so;/usr/lib/libOpenNI2.so;/usr/lib/i386-linux-gnu/libdc1394.so;/opt/ros/kinetic/lib/librealsense.so;/opt/ros/kinetic/lib/liboctomap.so;/opt/ros/kinetic/lib/liboctomath.so (Required is at least version "0.19.5") 
-- Found OpenCV: /opt/ros/kinetic (found version "3.3.1") 
-- Checking for module 'eigen3'
--   Found eigen3, version 3.2.92
-- Found eigen: /usr/include/eigen3  
-- Checking for module 'libopenni'
--   Found libopenni, version 1.5.4.0
-- Found openni: /usr/lib/libOpenNI.so  
-- Checking for module 'libopenni2'
--   Found libopenni2, version 2.2.0.3
-- Found OpenNI2: /usr/lib/libOpenNI2.so  
-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/i386-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
[... 30 lines cut ...]
CMake Error: : System Error: No such file or directory
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:384 (configure_file):
  configure_file Problem configuring file
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  rtabmap_ros/CMakeLists.txt:161 (catkin_package)

CMake Error: Could not open file for write in copy operation /home/yogeshwar/catkin_ws/devel/share/rtabmap_ros/cmake/rtabmap_rosConfig-version.cmake.tmp
CMake Error: : System Error: No such file or directory
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:389 (configure_file):
  configure_file Problem configuring file
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  rtabmap_ros/CMakeLists.txt:161 (catkin_package)

-- WITH octomap_msgs
-- WITH find_object_2d
-- checking for module 'OGRE'
--   found OGRE, version 1.9.0
-- WITH rviz
-- WITH costmap_2d
-- Configuring incomplete, errors occurred!
See also "/home/yogeshwar/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/yogeshwar/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

yogiji-coder, 2020-04-29:
Can someone help me this above error

matlabbe, 2020-05-26:
Cleanup your workspace, there is something wrong with catkin_make.
```
$ rm -r catkin_ws/build catkin_ws/devel
```
then redo a catkin_make.

ShahudullahKhan, 2021-07-11:
> #### Running command: "cmake /home/yogeshwar/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/yogeshwar/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/yogeshwar/catkin_ws/install -G Unix Makefiles" in [...]

has your problem solved by cleaning up your workspace and then redoing catkin_make