introlab/rtabmap_ros #44: data_recorder node missing

cloitre, 2016-01-23:
Hello,

I noticed that the node data_recorder is no longer built into the package rtabmap_ros but is still mentioned in the wiki page. Could you please update the wiki page to reflect the latest version of the package?

I found the launch file that illustrates data recording and was able to use it with no issue.
Thank you very much for sharing rtabmap with the ROS community!

Best Regards,

Audren

matlabbe, 2016-01-23:
Hi,

Yes, there is no more node called `data_recorder`, only a launch file `data_recorder.launch`. If you have trouble with the installed `data_recorder.launch` from apt-get, download [this version from master branch](https://github.com/introlab/rtabmap_ros/blob/master/launch/data_recorder.launch).

$ roslaunch

For the ROS wiki, I cannot update now because of this issue: https://github.com/ros-infrastructure/roswiki/issues/139 (wiki is read-only). When the wiki will come back to normal, I'll make the change!

cheers

cloitre, 2016-01-24:
Thanks a lot Matthieu!

closed 2016-01-27 by matlabbe