introlab/rtabmap_ros #45: Mapping and Navigation with turtlebot issue

kaygudo, 2016-01-28:
Hello,
I wish to do RGBD slam navigation using turtle bot as explained in [Here](http://wiki.ros.org/rtabmap_ros/Tutorials/MappingAndNavigationOnTurtlebot). How ever when i try to execute command 
"roslaunch rtabmap_ros demo_turtlebot_mapping.launch"

I get following message:

[ INFO] [1454008734.671710371]: No matching device found.... waiting for devices. Reason: std::string openni2_wrapper::OpenNI2Driver::resolveDeviceURI(const string&) @ /tmp/buildd/ros-indigo-openni2-camera-0.2.5-0trusty-20151130-2131/src/openni2_driver.cpp @ 631 : Invalid device number 1, there are 0 devices connected.
[ WARN] [1454008735.034947191]: Timed out waiting for transform from base_footprint to map to become available before running costmap, tf error: Could not find a connection between 'map' and 'base_footprint' because they are not part of the same tree.Tf has two or more unconnected trees.. canTransform returned after 0.100916 timeout was 0.1.

and after this when i try to execute 
" roslaunch rtabmap_ros demo_turtlebot_rviz.launch"
i get following results:
[image]
can you plz help i am new to ROS.

matlabbe, 2016-01-28:
Hi,

Your main problem is that OpenNI2 cannot detect your camera:
```
there are 0 devices connected.
```
What kind of camera are you using? If you have a Kinect, see [section 5](http://wiki.ros.org/rtabmap_ros/Tutorials/MappingAndNavigationOnTurtlebot#If_you_don.27t_have_the_robot) of the tutorial.

cheers

kaygudo, 2016-01-29:
Hey,
Thanks a lot that worked for me  for mapping on rivz. But, after mapping when i close rivz and try to execute
"$ roslaunch rtabmap_ros demo_turtlebot_mapping.launch localization:=true" for localization i get following error:

[ WARN] [1454091492.277720619]: rtabmap: Could not get transform from odom to base_footprint after 0.100000 second!

[ INFO] [1454091492.290996770]: Stopping device RGB and Depth stream flush.

[ WARN] [1454091492.387544016]: rtabmap: Could not get transform from odom to base_footprint after 0.100000 second!

and this keeps on repeating...

Can you please help?
With regards.