introlab/rtabmap_ros #50: SIGBUS in rtabmap::uncompressData() on ARM

mwerezak, 2016-02-18:
I got the following backtrace following the instructions in the debugging issue.
```
Program received signal SIGBUS, Bus error.
[Switching to Thread 0xdc2b1e10 (LWP 3952)]
0xf69a3232 in rtabmap::uncompressData(unsigned char const*, unsigned long) ()
---Type <return> to continue, or q <return> to quit---
   from /opt/ros/indigo/lib/arm-linux-gnueabihf/librtabmap_core.so.0.10
(gdb) 
(gdb) bt
#0  0xf69a3232 in rtabmap::uncompressData(unsigned char const*, unsigned long)
    () from /opt/ros/indigo/lib/arm-linux-gnueabihf/librtabmap_core.so.0.10
#1  0xf69a358e in rtabmap::uncompressData(cv::Mat const&) ()
   from /opt/ros/indigo/lib/arm-linux-gnueabihf/librtabmap_core.so.0.10
#2  0xf69a37f4 in rtabmap::CompressionThread::mainLoop() ()
   from /opt/ros/indigo/lib/arm-linux-gnueabihf/librtabmap_core.so.0.10
#3  0xf680b1e4 in UThread::ThreadMain() ()
   from /opt/ros/indigo/lib/arm-linux-gnueabihf/librtabmap_utilite.so.0.10
#4  0xf680b402 in UThreadC<void>::ThreadMainHandler(UThreadC<void>::Instance*)
    () from /opt/ros/indigo/lib/arm-linux-gnueabihf/librtabmap_utilite.so.0.10
#5  0xf6dfefbc in start_thread (arg=0xdc2b1e10) at pthread_create.c:314
#6  0xf667120c in ?? ()
    at ../ports/sysdeps/unix/sysv/linux/arm/nptl/../clone.S:92
   from /lib/arm-linux-gnueabihf/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb)
```
Not sure what other information I could provide.

matlabbe, 2016-02-23:
I don't have an ARM processor to build/test with. What exactly are you doing? Is the [robot mapping demo](http://wiki.ros.org/rtabmap_ros#Robot_mapping) work?

mwerezak, 2016-03-03:
@matlabbe sorry about the delay in getting back to you. Checking if it works with the robot mapping demo seems like it is obvious and I should have thought of that, I had been waiting to reply once I had a chance to do that but I've been very busy. I will check if the robot mapping demo works today.

closed 2016-11-09 by matlabbe

matlabbe, 2021-05-22:
crossref http://official-rtab-map-forum.67519.x6.nabble.com/Bus-Error-td7884.html#a8068