introlab/rtabmap_ros #52: Catkin_make fails when generating gradle project (could not find rtabmap_ros on the ros package path)

OAkyildiz, 2016-02-28:
```
Hello,

I 've installed the stand-alone rtabmap package to  my workspace's devel folder as it says in the instructions. The source is somewhere else (as the instructions say). But when I am buildingthe package, I get this:

[ 45%] Generating Java gradle project from rtabmap_ros
Traceback (most recent call last):
  File "/opt/ros/indigo/share/genjava/cmake/../../../lib/genjava/genjava_gradle_project.py", line 14, in <module>
    genjava.main(sys.argv)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/genjava/genjava_main.py", line 82, in main
    gradle_project.create(args.package, args.output_dir)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/genjava/gradle_project.py", line 152, in create
    raise IOError("could not find %s on the ros package path" % msg_pkg_name)
IOError: could not find rtabmap_ros on the ros package path
make[2]: *** [java/rtabmap_ros/build.gradle] Error 1
make[1]: *** [CMakeFiles/rtabmap_ros_generate_messages_java.dir/all] Error 2
make: *** [all] Error 2
```

Here is the workspace configuration:
```
-------------------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/indigo
Workspace:                   /home/oakyildiz/deps_workspace
Source Space:       [exists] /home/oakyildiz/deps_workspace/src
Build Space:        [exists] /home/oakyildiz/deps_workspace/build
Devel Space:        [exists] /home/oakyildiz/deps_workspace/devel
Install Space:      [exists] /home/oakyildiz/deps_workspace/install
DESTDIR:                     None
-------------------------------------------------------------------
Isolate Develspaces:         False
Install Packages:            True
Isolate Installs:            False
-------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
-------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-------------------------------------------------------------------
Workspace configuration appears valid.
```
And here is the cmake configuration of rtabmap

```
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- checking for module 'openni-dev'
--   package 'openni-dev' not found
-- Found OpenCV: /usr/local/include/opencv;/usr/local/include
-- Found PCL: /usr/include/pcl-1.7;/usr/include/eigen3;/usr/include;/usr/include/ni;/usr/include/vtk-5.8
-- Found ZLIB: /usr/include
-- Found Freenect: /usr/include
-- Found freenect2: /usr/local/include
-- Found OpenNI2: /usr/include/openni2
-- Found DC1394: /usr/include/dc1394
-- Found g2o: /usr/local/include;/usr/include/suitesparse
-- GTSAM include directory:  /usr/local/lib/cmake/GTSAM/../../../include;/usr/include;/usr/local/include/gtsam/3rdparty/Eigen/
-- Found cvsba: /usr/local/include
-- Found Pthreads
-- --------------------------------------------
-- Info :
--   CMAKE_INSTALL_PREFIX = /home/oakyildiz/deps_workspace/devel
--   CMAKE_BUILD_TYPE = Release
--   BUILD_SHARED_LIBS = ON
--   With OpenCV 2 nonfree module (SIFT/SURF) = YES
--   With Freenect             = YES
--   With OpenNI2              = YES
--   With Freenect2            = YES
--   With dc1394               = YES
--   With FlyCapture2/Triclops = NO (Point Grey SDK not found)
--   With g2o                  = YES
--   With GTSAM                = YES
--   With cvsba                = YES
--   With Qt                   = YES (version 5)
-- --------------------------------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /home/oakyildiz/opt/rtabmap_standalone/build
```

Why is the make looking for the package it is installing?

(I also get segfault when I try to run the rtabmap executable in the source folder:
```
==2880== Process terminating with default action of signal 11 (SIGSEGV)
==2880==  Bad permissions for mapped region at address 0x65B6C30
==2880==    at 0x490D1DFC: ??? (in /usr/lib/x86_64-linux-gnu/libQtGui.so.4.8.6)
==2880==    by 0x4010139: call_init.part.0 (dl-init.c:78)
==2880==    by 0x4010222: call_init (dl-init.c:36)
==2880==    by 0x4010222: _dl_init (dl-init.c:126)
==2880==    by 0x4001309: ??? (in /lib/x86_64-linux-gnu/ld-2.19.so)
```

But it's linked against Qt5. Is that expected?

matlabbe, 2016-02-29:
Hi,

CMake has detected Qt5 and rtabmap is built against it. However on runtime, the first Qt library found on the PATH is the Qt4 one (it should be the Qt5), which gives you a segmentation fault. For example, if you do `$ qmake --version`, which Qt is found?

What is your ROS installation? I think Java messages are not generated by default. I didn't test the build with ROS java.

cheers

OAkyildiz, 2016-03-01:
Hello again,

I'm sourcing `/opt/qt55/bin/qt55-env.sh` so   `$ qmake --version`  retuns me 
``` 
QMake version 3.0
Using Qt version 5.5.1 in /opt/qt55/lib
``` 

My ROS distro is Indigo. Should I disable gradle? ( I have no idea why gradle project is created in the  first place)

Thanks,

matlabbe, 2016-03-01:
Try to remove `ros-indigo-genjava`:
```bash
$ sudo apt-get remove ros-indigo-genjava
```

OAkyildiz, 2016-04-03:
Forgot to say thanks, I just removed genjava as you said.

closed 2016-04-03 by OAkyildiz