introlab/rtabmap_ros #54: tf2::ExtrapolationException

masterhui, 2016-03-08:
Hello Mat,
I have been using rtabmap previously withou any issues. However, after the last round of updates, the rtabmap node keeps dying following a 'tf2::ExtrapolationException'. 

**Error message:**
```
terminate called after throwing an instance of 'tf2::ExtrapolationException'
  what():  Lookup would require extrapolation into the future.  Requested time 1457438516.876949095 but the latest data is at time 1457438
516.873540253, when looking up transform from frame [laser_scanner_mid] to frame [base_frame]
[rtabmap/rtabmap_3D_MAP-6] process has died [pid 8563, exit code -6, cmd /home/ros/overlay_ws/devel/lib/rtabmap_ros/rtabmap --delete_db_on
_start scan:=/hokuyo_30lx_mid_scan rgb/image:=/kinect2/qhd/image_color depth/image:=/kinect2/qhd/image_depth_rect rgb/camera_info:=/kinect
2/qhd/camera_info odom:=/odom __name:=rtabmap_3D_MAP __log:=/home/ros/.ros/log/8a3af878-e525-11e5-b6c2-00187d93bd4f/rtabmap-rtabmap_3D_MAP
-6.log].
log file: /home/ros/.ros/log/8a3af878-e525-11e5-b6c2-00187d93bd4f/rtabmap-rtabmap_3D_MAP-6*.log
```

**Error description:**
- This only happens if I set <param name="RGBD/OptimizeSlam2D" value="true"/>
- The node runs for a varying amount of time (0-5s) and then crashes with the error above
- The time difference between the two mentioned frames is tiny:
```
ros@autonomy:~$ rosrun tf tf_monitor base_frame laser_scanner_mid                                                                                                                                                                                                                    
Waiting for transform chain to become available between base_frame and laser_scanner_mid 

RESULTS: for base_frame to laser_scanner_mid
Chain is: base_frame -> laser_scanner_mid
Net delay     avg = 0.00165259: max = 0.0159715

Frames:
Frame: laser_scanner_mid published by unknown_publisher Average Delay: 0.000465277 Max Delay: 0.00128411

All Broadcasters:
Node: unknown_publisher 260.943 Hz, Average Delay: -0.0292105 Max Delay: 0.00131634

```

- I have checked the tf tree (tf_echo, tf_monitor, view_frames), all delays etc. seem to be ok:
```
RESULTS: for all Frames

Frames:
Frame: /base_frame published by unknown_publisher Average Delay: -0.0496703 Max Delay: 0
Frame: /kinect2_frame published by unknown_publisher Average Delay: -0.0496107 Max Delay: 0
Frame: /kinect2_ir_frame published by unknown_publisher Average Delay: -0.0498223 Max Delay: 0
Frame: /kinect2_ir_optical_frame published by unknown_publisher Average Delay: -0.0496312 Max Delay: 0
Frame: /kinect2_link published by unknown_publisher Average Delay: -0.0497283 Max Delay: 0
Frame: /kinect2_rgb_frame published by unknown_publisher Average Delay: -0.0497038 Max Delay: 0
Frame: /kinect2_rgb_optical_frame published by unknown_publisher Average Delay: -0.0497969 Max Delay: 0
Frame: base_footprint published by unknown_publisher Average Delay: 0.000451188 Max Delay: 0.00160235
Frame: base_laser_support published by unknown_publisher Average Delay: 0.000450806 Max Delay: 0.00194681
Frame: base_turret published by unknown_publisher Average Delay: 0.000440607 Max Delay: 0.00193166
Frame: box_battery published by unknown_publisher Average Delay: 0.000435769 Max Delay: 0.00191937
Frame: box_slam published by unknown_publisher Average Delay: 0.000431905 Max Delay: 0.00192083
Frame: flipper_left_front published by unknown_publisher Average Delay: 0.000431024 Max Delay: 0.00192143
Frame: flipper_left_rear published by unknown_publisher Average Delay: 0.000428264 Max Delay: 0.00190327
Frame: flipper_right_front published by unknown_publisher Average Delay: 0.000426828 Max Delay: 0.0019016
Frame: flipper_right_rear published by unknown_publisher Average Delay: 0.00042596 Max Delay: 0.00190066
Frame: laser_scanner_mid published by unknown_publisher Average Delay: 0.00042489 Max Delay: 0.00190085

All Broadcasters:
Node: unknown_publisher 237.992 Hz, Average Delay: -0.0284516 Max Delay: 0.00191627

```

**The fix:**
After enough digging around, I have found out how to fix the issue:

```
diff --git a/src/CoreWrapper.cpp b/src/CoreWrapper.cpp
index a9562e7..cf30813 100644
--- a/src/CoreWrapper.cpp
+++ b/src/CoreWrapper.cpp
@@ -698,7 +698,7 @@ Transform CoreWrapper::getTransform(const std::string & fromFrameId, const std::
                if(waitForTransform_ && !stamp.isZero() && waitForTransformDuration_>0.0)
                {
                        //if(!tfBuffer_.canTransform(fromFrameId, toFrameId, stamp, ros::Duration(1)))
-                       if(!tfListener_.waitForTransform(fromFrameId, toFrameId, stamp, ros::Duration(waitForTransformDuration_)))
+                       if(!tfListener_.waitForTransform(fromFrameId, toFrameId, ros::Time::now(), ros::Duration(waitForTransformDuration_)))
                        {
                                ROS_WARN("rtabmap: Could not get transform from %s to %s after %f second!", fromFrameId.c_str(), toFrameId.c_str(), waitForTransformDuration_);
                                return transform;

```
Essentially, by replacing "stamp" with "ros::Time::now()", it works.

**Setup:**
- rtabmap_ros (from source) commit 36f283851257eb3d65dc3b4b918a2a4efdfe26d6 (Mon Feb 8 11:24:06 2016 -0500) added missing launch_prefix
- rtabmap (from source) commit c769f85523037c7b4cb996d787f0033cedaae829(Tue Dec 8 14:33:50 2015 -0500) Merge branch 'bckim92-master'
- ROS Indigo

Any ideas why I am having problems now, but never used to have?

Thank you and best wishes!

matlabbe, 2016-03-09:
Hi,

Normally, the TF exceptions are caught [here](https://github.com/introlab/rtabmap_ros/blob/622d15ea6ae09ea5b8da2e7055fd447a1c7017cb/src/CoreWrapper.cpp#L712-L715), and shown as warnings (and no crashes). There is maybe another method using TF for which the exceptions are not caught. I've seen recently that [transformLaserScanToPointcloud()](https://github.com/introlab/rtabmap_ros/blob/622d15ea6ae09ea5b8da2e7055fd447a1c7017cb/src/CoreWrapper.cpp#L899) method was not correctly used in rtabmap node. The stamp should [include scan time too](http://wiki.ros.org/laser_geometry#High_fidelity_projection). The exception may be thrown by this function. I've made the change in https://github.com/introlab/rtabmap_ros/commit/26c656776c10c445b2f09b63b351f932c20c0cc8, you can try it.

cheers

masterhui, 2016-03-22:
> exception may be thrown by this function. I've made the change in 26c6567

Hi,
Sorry I'm only getting back to you now. The notification message got
filtered away and I only noticed today...

I will check your fix asap and get back to you.

Regards

masterhui, 2016-03-24:
Hi,
I can confirm this is working now. Thank you for the quick fix.

Cheers

closed 2016-03-24 by masterhui