introlab/rtabmap_ros #57: rtabmap: Could not get transform from base_footprint to camera_depth_frame after 0.100000 second!

somebodyus, 2016-03-16:
[ WARN] [1458096517.558391078]: rtabmap: Could not get transform from base_footprint to camera_depth_frame after 0.100000 second!
[ERROR] [1458096517.558518276]: TF of received laser scan topic at time 1458096517.418607s is not set, aborting rtabmap update.

Note that I have written the static_transform_publisher in my launch file.
<node pkg="tf" type="static_transform_publisher" name="base_footprint_xtion" args="0 0 0 0 0 0 1 /base_footprint /camera_rgb_optical_frame 100"/>

somebodyus, 2016-03-16:
problem solved
please delete this thread

closed 2016-03-16 by matlabbe

stevemartinov, 2021-04-10:
How did you solve this?

matlabbe, 2021-05-03:
You can debug tf tree with:
```
rosrun tf2_tools view_frames.py
```
to find which frames are missing or not connected to others.