introlab/rtabmap_ros #59: Could you tell me the tf between Kinect and base_link in hand_held case?

somebodyus, 2016-03-23:
Could you tell me the tf between Kinect and base_link in hand_held case?

I am using:
<node pkg="tf" type="static_transform_publisher" name="kinect_base_link" args="0 0 0 0 0 0 /base_link /kinect10"/>

The built map is not vertical to the grid in RVIZ.

matlabbe, 2016-03-24:
Which package do you use for the Kinect? normally, the TFs are already published (including optical rotation) with [openni_launch](http://wiki.ros.org/openni_launch) or [freenect_launch](http://wiki.ros.org/freenect_launch) with `/camera_link` as the kinect link. You could then set 

```
<node pkg="tf" type="static_transform_publisher" name="kinect_base_link" args="0 0 0 0 0 0 /base_link /camera_link"/>
```

somebodyus, 2016-03-24:
Yes, I set the TF exactly as you mentioned, but the built map is not parallel with the horizontal.

matlabbe, 2016-03-24:
The referential of the map is set as the first camera position. This means that if the camera is tilted, for example, according to the horizon, the whole map will be tilted.

somebodyus, 2016-03-24:
Noted with thanks.
Is there any possibility to rectify the map pose? This is a critical problem because we need the projected 2-D occupancy map. If the whole map is tilted, we will get a useless projected map.

matlabbe, 2016-03-24:
If it is on a robot and the camera is fixed with an angle, you can set TF between /base_link and /camera_link to include this angle. 

For hand-held mapping (where the camera can be arbitrary started with any angle), you could create a node detecting the ground plane, compute camera orientation/height then reset odometry at this pose (even reset rtabmap too to clear the map built so far).

somebodyus, 2016-03-24:
OK. Thanks.
It seems that giving the camera on a ground plane at start is easier.

closed 2016-11-09 by matlabbe