introlab/rtabmap_ros #60: How can I reproduce your stereo demo?

somebodyus, 2016-03-23:
How can I reproduce your stereo demo?
Did you use the wheel odometry information? Could you publish a launch file?

matlabbe, 2016-03-24:
Which stereo demo are you referring? There are some stereo demos (launch included):
 - http://wiki.ros.org/rtabmap_ros/Tutorials/StereoOutdoorMapping
 - http://wiki.ros.org/rtabmap_ros/Tutorials/StereoHandHeldMapping
 - https://github.com/introlab/rtabmap/wiki/Stereo-mapping#process-a-directory-of-stereo-images-in-ros

wheel odometry is not used in all these demos. Generally for stereo, outdoor mapping is better as there are more visual textures. As you can see in the outdoor demo with the robot, the camera is tilted to the ground to see closer features. Stereo cameras with standard baseline around 10/12 cm can triangulate precisely up to 10/15 meters. So if you are looking at a far background, it won't work. Last thing, a poor calibration of the stereo camera will result in poor performance.

somebodyus, 2016-03-24:
Thank you so much. I mean the StereoOutdoorMapping. 
The ZED ROS package comes with a calibration process. It can output rectified left and right images.
Therefore, I delete the following part and feed rtabmap directly with the rectified images give by ZED node. I don't think it is a calibration reason. 
I tested it in an office. Rtabmap always publish warnings about no enough correspondences. 
In your outdoor demo, the camera was looking at the grassland, where there exist repetitive features. It proved that your loop closing is awesome.

`   
   <node name="republish_left" type="republish" pkg="image_transport" args="compressed in:=/stereo_camera/left/image_raw_throttle raw out:=/stereo_camera/left/image_raw_throttle_relay" />
   <node name="republish_right" type="republish" pkg="image_transport" args="compressed in:=/stereo_camera/right/image_raw_throttle raw out:=/stereo_camera/right/image_raw_throttle_relay" />

   
   <group ns="/stereo_camera" >
      <node pkg="nodelet" type="nodelet" name="stereo_nodelet"  args="manager"/>
   
      <node pkg="stereo_image_proc" type="stereo_image_proc" name="stereo_image_proc">
         <remap from="left/image_raw"    to="left/image_raw_throttle_relay"/>
         <remap from="left/camera_info"  to="left/camera_info_throttle"/>
         <remap from="right/image_raw"   to="right/image_raw_throttle_relay"/>
         <remap from="right/camera_info" to="right/camera_info_throttle"/>
         <param name="disparity_range" value="128"/>
      </node>
   </group>`

closed 2016-11-09 by matlabbe