introlab/rtabmap_ros #63: How can I do SLAM autonomously?

somebodyus, 2016-04-05:
How can I do SLAM autonomously?
Autonomously explore the world without remotely controlled by a joystick.

TSC21, 2016-04-05:
That's not a SLAM related issue but more like a navigation one. SLAM will only allow the robot/vehicle to position itself in the environment and at the same time, map that same environment. You need some other tool/algorithm that allows the robot to explore that same environment, which implies something that looks for open gaps on the ocupation map and efficiently follows the shortest path to a auto set goals.
So, as I stated, this is an issue which on my opinion is not directly related to rtabmap, more to some kind of navigation framework.

somebodyus, 2016-04-05:
Thank you so much.
I am wondering if there is a related package in rtabmap_ros.

matlabbe, 2016-04-05:
Navigation examples on rtabmap_ros wiki are done by integrating [move_base](http://wiki.ros.org/move_base) package for navigation. This package will help you to move autonomously from Point A to Point B (by setting goals) while avoiding obstacles. For exploration, you may need to implement your own approach.

cheers

closed 2016-11-09 by matlabbe