introlab/rtabmap_ros #64: No map generating in rviz

pamico, 2016-04-05:
I have my fixed and reference frame set to odom and PointCloud2 topic set to /voxel_cloud and I do not see any output in rviz even though I am seeing a map being created in the rtabmap application. Any idea what I'm doing wrong? The only way I see any output from the kinect2 I'm using is when I switch my PointCloud2 topic to either kinect2/qhd/points or one of the other quality point clouds however when I do this it does not create a saved map.

matlabbe, 2016-04-06:
Can you show your launch file or which ones are used? rtabmap binaries or built from source? In examples, `voxel_cloud` is an output and it is used only for visualization. 

You said "requires a very long time to register", can you show the terminal output of rtabmap to see processing time used by rtabmap node?

matlabbe, 2016-11-09:
closed for inactivity

closed 2016-11-09 by matlabbe