introlab/rtabmap_ros #67: StereoCameraModel.h missing while doing catkin_make

Aswath93, 2016-04-12:
Hey guys . I am trying to use rtabmap_ros with ros indigo. When I do catkin_make, I get the following error:

In file included from /home/aswath/ros/src/rtabmap_ros/src/nodelets/point_cloud_xyz.cpp:32:0:
/home/aswath/ros/src/rtabmap_ros/include/rtabmap_ros/MsgConversion.h:44:44: fatal error: rtabmap/core/StereoCameraModel.h: No such file or directory
 #include <rtabmap/core/StereoCameraModel.h>
                                            ^
compilation terminated.
In file included from /home/aswath/ros/src/rtabmap_ros/src/nodelets/point_cloud_xyzrgb.cpp:36:0:
/home/aswath/ros/src/rtabmap_ros/include/rtabmap_ros/MsgConversion.h:44:44: fatal error: rtabmap/core/StereoCameraModel.h: No such file or directory
 #include <rtabmap/core/StereoCameraModel.h>
                                            ^
compilation terminated.
In file included from /home/aswath/ros/src/rtabmap_ros/src/nodelets/obstacles_detection.cpp:55:0:
/home/aswath/ros/src/rtabmap_ros/include/rtabmap_ros/MsgConversion.h:44:44: fatal error: rtabmap/core/StereoCameraModel.h: No such file or directory
 #include <rtabmap/core/StereoCameraModel.h>
                                            ^
compilation terminated.
In file included from /home/aswath/ros/src/rtabmap_ros/src/nodelets/point_cloud_aggregator.cpp:21:0:
/home/aswath/ros/src/rtabmap_ros/include/rtabmap_ros/MsgConversion.h:44:44: fatal error: rtabmap/core/StereoCameraModel.h: No such file or directory
 #include <rtabmap/core/StereoCameraModel.h>
                                            ^
compilation terminated.
make[2]: *** [rtabmap_ros/CMakeFiles/rtabmap_ros.dir/src/nodelets/point_cloud_xyz.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [rtabmap_ros/CMakeFiles/rtabmap_ros.dir/src/nodelets/point_cloud_xyzrgb.cpp.o] Error 1
make[2]: *** [rtabmap_ros/CMakeFiles/rtabmap_ros.dir/src/nodelets/obstacles_detection.cpp.o] Error 1
make[2]: *** [rtabmap_ros/CMakeFiles/rtabmap_ros.dir/src/nodelets/point_cloud_aggregator.cpp.o] Error 1
make[1]: *** [rtabmap_ros/CMakeFiles/rtabmap_ros.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

Could someone help me fix this?.

Thanks,
Aswath

matlabbe, 2016-04-12:
Did you update recently [rtabmap](https://github.com/introlab/rtabmap) library from 0.10 to 0.11 (master)? If so, cleanup the build directory of rtabmap and install again to make sure that headers are installed in `~/catkin_ws/devel/include/rtabmap-0.11` and not `~/catkin_ws/devel/include/rtabmap-0.10`. You can also make remove `~/catkin_ws/devel/lib/rtabmap-0.10`.

Similar post: http://official-rtab-map-forum.67519.x6.nabble.com/Master-branch-is-now-on-0-11-td1128.html
Migration guide 0.10 to 0.11: https://github.com/introlab/rtabmap_ros/issues/58

cheers

Aswath93, 2016-04-14:
thanks for replying. evry thing is rtabmap_ros. I am not sure what to change. I am not migrating. I just tried installing rtabmap a few days ago

Aswath93, 2016-04-14:
~/catkin_ws/devel/include/rtabmap_ros
this is what is shown in my computer

matlabbe, 2016-04-14:
rtabmap binaries may be installed, so rtabmap_ros tries to link with them instead of rtabmap installed from source. If you are on Indigo, make sure to do:
```
$ sudo apt-get remove ros-indigo-rtabmap
```

If rtabmap 0.11 is installed in `~/catkin_ws/devel` (as suggested in the [build instructions](https://github.com/introlab/rtabmap_ros#build-from-source)), do you have this file?
```
~/catkin_ws/devel/include/rtabmap-0.11/rtabmap/core/StereoCameraModel.h
```
If yes, there is a problem with rtabmap_ros setting wrong include directories for rtabmap (`RTABMap_INCLUDE_DIRS` [here](https://github.com/introlab/rtabmap_ros/blob/63de9018dfa4e582fca2407683676f10d3ed1032/CMakeLists.txt#L118)), set by cmake using this file:
```
~/catkin_ws/devel/lib/rtabmap-0.11/RTABMapConfig.cmake
```

You can check the value of this variable by adding this in the CMakeLists.txt of rtabmap_ros:
```bash
MESSAGE(STATUS "RTABMap_INCLUDE_DIRS=${RTABMap_INCLUDE_DIRS}")
```

closed 2016-11-09 by matlabbe