introlab/rtabmap_ros #71: Compile error after update

masterhui, 2016-04-28:
Hi,
I just updated:

- ROS Indigo (latest package updates)
- rtabmap (commit a78587e505e9b8b870400f4de2e32d84f99c026d "Hidden FPS shown in CloudViewer") 
- rtabmap_ros (commit 424fcd8c4d111597f09f2d6f669a146b3c092234 "Making rviz dependency optional")

rtabmap compiled OK, I installed it with `sudo make install`. However, rtabmap_ros throws this error:

`/home/ros/overlay_ws/devel/lib/librtabmap_ros.so: undefined reference to `rtabmap::util3d::normalFiltering(boost::shared_ptr<pcl::PointClo
ud<pcl::PointXYZRGB> > const&, boost::shared_ptr<std::vector<int, std::allocator<int> > > const&, float, Eigen::Matrix<float, 4, 1, 0, 4, 
1> const&, int, Eigen::Matrix<float, 4, 1, 0, 4, 1> const&)'
/home/ros/overlay_ws/devel/lib/librtabmap_ros.so: undefined reference to `rtabmap::util3d::cloudRGBFromSensorData(rtabmap::SensorData cons
t&, int, float, float, std::vector<int, std::allocator<int> >*)'
/home/ros/overlay_ws/devel/lib/librtabmap_ros.so: undefined reference to `rtabmap::util3d::cloudFromSensorData(rtabmap::SensorData const&,
 int, float, float, std::vector<int, std::allocator<int> >*)'
/home/ros/overlay_ws/devel/lib/librtabmap_ros.so: undefined reference to `rtabmap::util3d::cloudFromStereoImages(cv::Mat const&, cv::Mat c
onst&, rtabmap::StereoCameraModel const&, int, float, float, std::vector<int, std::allocator<int> >*)'
/home/ros/overlay_ws/devel/lib/librtabmap_ros.so: undefined reference to `rtabmap::util3d::normalFiltering(boost::shared_ptr<pcl::PointClo
ud<pcl::PointXYZ> > const&, boost::shared_ptr<std::vector<int, std::allocator<int> > > const&, float, Eigen::Matrix<float, 4, 1, 0, 4, 1> 
const&, int, Eigen::Matrix<float, 4, 1, 0, 4, 1> const&)'
/home/ros/overlay_ws/devel/lib/librtabmap_ros.so: undefined reference to `rtabmap::util3d::cloudFromDepthRGB(cv::Mat const&, cv::Mat const
&, float, float, float, float, int, float, float, std::vector<int, std::allocator<int> >*)'
/home/ros/overlay_ws/devel/lib/librtabmap_ros.so: undefined reference to `rtabmap::util3d::cloudFromDepth(cv::Mat const&, float, float, fl
oat, float, int, float, float, std::vector<int, std::allocator<int> >*)'
/home/ros/overlay_ws/devel/lib/librtabmap_ros.so: undefined reference to `rtabmap::util3d::cloudFromDisparity(cv::Mat const&, rtabmap::Ste
reoCameraModel const&, int, float, float, std::vector<int, std::allocator<int> >*)'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/ros/overlay_ws/devel/lib/rtabmap_ros/map_assembler] Error 1
`

Regards

matlabbe, 2016-04-28:
Hi,
make sure you don't have `ros-indigo-rtabmap` installed. Did you install rtabmap in `/usr/local`? If so, verify that you have `/usr/local/lib/rtabmap-0.11` and not `/usr/local/lib/rtabmap-0.10`. Latest version of `rtabmap_ros` should link on 0.11 libraries.

cheers

masterhui, 2016-04-29:
Hi,
Yes, I did install in `/usr/local`. However, I now noticed that a whole collection of previous rtabmap installations were still kicking around there. So I ran `/usr/local$ sudo find f . -name *rtabmap* -exec rm -rf {} \;` to get rid of everything, and then re-ran `make install.` To get things to work I also had to delete the `build` and `devel` directories of the catkin workspace and rebuild all. 

Thank you for your help!

Cheers

closed 2016-04-29 by masterhui