introlab/rtabmap_ros #72: OptimizerG2O: Condition (v2 != 0) not met!

KamalDSOberoi, 2016-05-13:
Hello guys,

I am working on a project to localize landmarks in a map and use them to localize the position of the robot in that map (Graph Based SLAM). 

When I launch this package, I get an exception as follows:

**OptimizerG2O.cpp:762::optimizeWithLandmarks() Condition (v2 != 0) not met!**

In my code I havent changed the status of the v2 anywhere. Although I have made some changes but I am using the pointer 
Optimizer * optimizer_ and calling optimizeWithLandmarks() function with the arguments fromId_, posesOut and linksOut (default arguments). 

I am not able to understand where the problem is arising. The MapOptimizerNode executes only once and this exception is thrown. 

If someone has faced this problem, it would be very helpful if you could guide me out of it. 

Thanks a lot in advance!!

Cheers,
Kamal

matlabbe, 2016-05-15:
In rtabmap repository, there is no `optimizeWithLandmarks()` method. I assume this is a modified version of `optimize()`. Well, if you search for `v2 != 0` in [OptimizerG2O.cpp](https://github.com/introlab/rtabmap/blob/master/corelib/src/OptimizerG2O.cpp), you will find that when an edge is added to the graph, the vertices should have been already added to the graph or this assert happens.

Note that `getConnectedGraph()` method ensures that the graph is complete before optimizing it (so there are no links referring a node not in poses). As you can see [here in map_optimizer](https://github.com/introlab/rtabmap_ros/blob/b8307449bd2e3eb3dbefc7755763dc6381117099/src/MapOptimizerNode.cpp#L241-L247), it is called right before doing optimization. In the code you sent me, you are modifying `posesOut` and `linksOut` variables before optimization. You should make sure that all poses referred in the links are in `posesOut`, otherwise you may remove them from `linksOut`.

cheers

KamalDSOberoi, 2016-05-15:
Thanks a lot for your help. I will take a look at this. 

Best Regards,
Kamal

closed 2016-05-15 by KamalDSOberoi