introlab/rtabmap_ros #75: NodeData.msg: userData

KamalDSOberoi, 2016-05-18:
Hello,

I am working with the package rtabmap_ros. In the message folder, there is a message file NodeData.msg, which contains an argument userData. This message file is used in MapData.msg and the mapData is sent to MapOptimizerNode.cpp.

My questions is:
Which data is received in the argument userData. Is it relevant to optimize the map? 

Thanks a lot for your guidance.

Best Regards,
Kamal

Thanks for any guidance.

matlabbe, 2016-05-18:
The `userData` field can contain custom data that were added to a node when it is created. However, the default ROS interface doesn't have the option to set this `userData`. Using the C++ API, we can modify it: example [here](https://github.com/introlab/rtabmap/blob/ccc4b4a6c2ee556db518d620de82b0c9e07fc00a/examples/NoEventsExample/main.cpp#L115), add `userData` (which is a `cv::Mat`) [here](https://github.com/introlab/rtabmap_ros/blob/76cb4365dd3b7e189325e991c36241c8d6522736/src/CoreWrapper.cpp#L973-L980) to `SensorData` before it is processed by `rtabmap`. 

Indeed, it would be nice to have for the ROS wrapper an input to `rtabmap` node including `userData` for convenience. For example, if you want to modify the code, you may add a fifth topic to [depthCallback()](https://github.com/introlab/rtabmap_ros/blob/76cb4365dd3b7e189325e991c36241c8d6522736/src/CoreWrapper.cpp#L1146-L1150) and add the user data to `SensorData` used to update the map. The [last argument](https://github.com/introlab/rtabmap/blob/eb21280f0486260b0ac94b4672200f85979f7c99/corelib/include/rtabmap/core/SensorData.h#L106) of `SensorData` constructor is the `userData`. The `userData` will then be after published on the in `nodeData` field of [mapData](http://docs.ros.org/jade/api/rtabmap_ros/html/msg/MapData.html) topic.

Unfortunately, there is no official documentation for these topics. [MapData](http://docs.ros.org/jade/api/rtabmap_ros/html/msg/MapData.html) contains the graph (which contains poses and links) and a vector of node data. The [node data ](http://docs.ros.org/jade/api/rtabmap_ros/html/msg/NodeData.html) contains RGB and depth images, raw odometry pose, laser scan, visual words used by loop closure detection, user data and some identification values.

Actually, [NodeData](http://docs.ros.org/jade/api/rtabmap_ros/html/msg/NodeData.html) message contains the same data as [Signature](https://github.com/introlab/rtabmap/blob/master/corelib/include/rtabmap/core/Signature.h) class of RTAB-Map's library. You can see how conversion is done in [nodeDataFromRos()](https://github.com/introlab/rtabmap_ros/blob/e67770cc319f3eeddea00bf4f039c48d0f4a6997/src/MsgConversion.cpp#L482-L583) method.

cheers

KamalDSOberoi, 2016-05-19:
Thanks a lot for this information. You are awesome! :+1: :100: 

Best Regards,
Kamal

closed 2016-05-19 by KamalDSOberoi