introlab/rtabmap_ros #77: Cannot find transform from base_link to duo_fram

nznobody, 2016-05-25:
Hello all,
Pretty sure this is a novice mistake of mine but I cannot see why i am getting the following error:

`[ WARN] [1464157264.645514361]: odometry: Could not get transform from base_link to duo_frame (stamp=9.398200) after 0.200000 seconds ("wait_for_transform_duration"=0.200000)!
`
In the duo_ros package I am writing (see https://github.com/nznobody/duo_ros ) I publish my images and do the following:
`
std::string 	_cameraFrame = "duo_frame";
...
leftImage.header.frame_id 	= _cameraFrame;
rightImage.header.frame_id 	= _cameraFrame;
`

In my launch file i have:
`<node pkg="tf" type="static_transform_publisher" name="camera_base_link" 
	args="$(arg optical_rotate) /base_link /duo_frame 100" />`

running the following reports:
`
$rosrun tf tf_echo /base_link /duo_frame
At time 1464157728.633
- Translation: [0.000, 0.000, 0.000]
- Rotation: in Quaternion [-0.500, 0.500, -0.500, 0.500]
            in RPY (radian) [-1.571, -0.000, -1.571]
            in RPY (degree) [-90.000, -0.000, -90.000] 
`

And yet still an error. I have made the pdf too and it also shows the transform. Comparing to the zed_wrapper my output is basically identical:
`
$rosrun tf tf_echo /base_link /zed_optical_frame
At time 1464156882.419
- Translation: [0.000, 0.000, 0.000]
- Rotation: in Quaternion [-0.500, 0.500, -0.500, 0.500]
            in RPY (radian) [-1.571, -0.000, -1.571]
            in RPY (degree) [-90.000, -0.000, -90.000]
`

and theirs works... what am I missing?

matlabbe, 2016-05-25:
The output `stamp=9.398200` seems wrong as normal stamps are more something like `1464157728.633` (Wed, 25 May 2016 06:28:48 GMT).
You should set stamps in your published images:
```cpp
ros::Time now = ros::Time::now();
rightImage.header.stamp = now;
leftImage.header.stamp = now;
```

cheers

nznobody, 2016-05-25:
@matlabbe Thanks! I was using the timestamp from the camera, not ros::Time::now()! Changing this fixed the transform issue.

closed 2016-05-25 by nznobody