introlab/rtabmap_ros #78: MapOptimizerNode: linksOut

KamalDSOberoi, 2016-05-25:
Hello,

I have a doubt regarding the value of linksOut defined in MapOptimizerNode. I am working on a project in which I need to create a graph which has robot poses as well as landmark observations. I am able to verify that when the robot moves, a new node is added for the robot pose. 

My doubt is: Should linksOut contain the link created between robot pose and corresponding landmark observation before optimization? If yes, should this link be added before or after computing the connected graph? As of now it contains just the links between two consecutive robot poses and nothing for the landmarks.    

Thanks in advance for your help!!

KamalDSOberoi, 2016-05-25:
Additional Information: When the connected graph is computed, the link created between robot pose and landmark observation in linksOut is removed and I am only left with the links between the consecutive robot poses. 

If I compute the connected graph before and then add the landmark link, the process dies. 

What could be the best way to solve this problem? 

Thanks!!

matlabbe, 2016-05-26:
All landmarks (and poses) should have at least one link to another pose/landmark in the graph. I assume that landmarks are treated like poses. For example, if "poses" have positive ids and "landmarks" have negative ids:
```
poses (with landmarks): [1, 2, 3, 4, 5, -1, -2, -3]
links: [1->2, 2->3, 3->4, 4->5, 2->-1, 3->-2, 4->-2]
```

After cleaning the graph, you should have something like this (note how landmark `-3` just disappeared because it is not linked to graph):
```
posesOut (with landmarks): [1, 2, 3, 4, 5, -1, -2]
linksOut: [1->2, 2->3, 3->4, 4->5, 2->-1, 3->-2, 4->-2]
```

Another example (note that poses `4` is not here):
```
poses (with landmarks): [1, 2, 3, 5, -1, -2, -3]
links: [1->2, 2->3, 3->4, 4->5, 2->-1, 3->-2, 4->-2]
```
After we do `getConnectedGraph()` with `fromId=1` (pose `5` is gone as well as landmark `-3`):
```
posesOut (with landmarks): [1, 2, 3, -1, -2]
linksOut: [1->2, 2->3, 2->-1, 3->-2]
```

You can try to show before/after the poses/links to debug like the examples above.

cheers

KamalDSOberoi, 2016-05-26:
Thanks a lot for this confirmation. :+1: 

Best Regards,
Kamal

closed 2016-05-26 by KamalDSOberoi