introlab/rtabmap_ros #79: MapOptimizerNode: if( poses.size() || constraints.size() )

KamalDSOberoi, 2016-05-26:
Hello,

I have doubt regarding the if condition: **else if(poses.size() || constraints.size())**

When I run my code, sometimes this condition becomes true and there is an error and the same code sometimes gives correct results. 

This means that sometimes there is a false userData and a false node is created without creating the constraint and sometimes everything works well. 

Is this due to synchronization problem? Is there any work around for this error? 

Thanks in advance for your help!! 
PS: Your guidance solves my problem every time :+1: 

Best Regards,
Kamal

matlabbe, 2016-05-26:
It would be easier to see the problem if you can log what is in poses and constraints:
```cpp
for(std::map<int,Transform>::const_iterator iter=poses.begin(); iter!=poses.end(); ++iter)
{
   UWARN("Pose: %d", iter->first);
}
for(std::multimap<int,Link>::const_iterator iter=constraints.begin(); iter!=constraints.end(); ++iter)
{
   UWARN("Link: %d -> %d", iter->second.from(), iter->second.to());
}
```

cheers

KamalDSOberoi, 2016-05-27:
Thanks for your message. I have logged in a file. The thing is that when the code starts executing, I receive userData (which contains pose of the landmarks) in mapData and a new node for the robot pose is created. That means **mapData->nodes.size() != 0**.  Please check [here](https://github.com/KamalDSOberoi/INSA/blob/master/MapOptimizerNode.cpp#L383-L402). Then till the time robot doesn't move, no userdata is received and no new node for the robot pose is created. 

Sometimes this works fine and I am able to visualize the landmarks and robot poses in the graph. But sometimes a new (second) userdata is received even though the robot hasn't moved at all. This creates a new node for the robot pose, without making any constraint, even though the robot is stationary. Then  **else if(poses.size() || constraints.size())** is executed.  

Now, I think that this is a problem. But on the other hand, I am also confused as to why no new userData is received even though in CoreWrapper.cpp it is being sent to sensorData in commonDepthCallback(). 

PS: I don't know if you have time to check it but [here] (https://github.com/KamalDSOberoi/INSA/blob/master/optimization.txt) is the log file when everything is working correctly. The robot has moved when the node 7 has been added and mapData->nodes.size() = 2 and mapData->graph.links.size() = 1. In the log file, modifiedMapData is mapData and posesOutNoLandmarks are the robot poses and posesOutLandmarks are landmark poses.

matlabbe, 2016-05-27:
Do you have an example that fails? A drawing of the poses, links and landmarks could also help to understand what you want. Example:
 
[image]
3 poses, 2 landmarks, 5 constraints

When publishing you may want to modify the ifs with something like this (with `poses` a map containing the landmarks' poses and node's poses):
```cpp

std::multimap<int, Link> constraints = cachedConstraints_;
std::map<int, Transform> poses;
for(std::map<int, Signature>::iterator iter=nodeInfos.begin(); iter!=nodeInfos.end(); ++iter)
{
    poses.insert(std::make_pair(iter->first, iter->second.getPose()));

    if(!iter->second.sensorData().userDataCompressed.empty())
    {
        Transform landmarkTransform;
        int landmarkId;
        // uncompress userData to get landmark's ID and Pose
        createLandmark(iter->second.sensorData().userDataCompressed, landmarkId, landmarkTransform);

        if(poses.find(landmarkId) == landmarks.end())
        {
              // add new landmark (initialize its pose from the first pose it is observed)
              poses.insert(std::make_pair(landmarkId, iter->second.getPose() * landmarkTransform));
              constraints.insert(Link(iter->first, landmarkId, landmarkTransform));
        }
        else
        {
             // just add a constraint
             constraints.insert(Link(iter->first, landmarkId, landmarkTransform));
        }
    }
}

{...}

if(poses.size() > 1 && constraints.size() > 0)
{
    // Optimization required!
    std::map<int, rtabmap::Transform> posesOut;
    std::multimap<int, rtabmap::Link> linksOut;
    optimizer_->getConnectedGraph(fromId, poses, constraints, posesOut, linksOut);
    optimizedPosesAndLandmarks = optimizer_->optimize(fromId, posesOut, linksOut);

    optimizedPoses = // optimizedPosesAndLandmarks without landmarks
    optimizedLandmarks = // optimizedPosesAndLandmarks without poses
}
else if(poses.size() == 1 && constraints.size() == 0)
{
    // No optimization required, we assume it is not a landmark in poses
    optimizedPoses = poses;
}
else if(poses.size() || landmarks.size() || constraints.size())
{
    UERROR("Not supposed to happen!");
}
```

KamalDSOberoi, 2016-05-30:
Hello,

Thanks for your message. I tried to do something similar but the error still exists. 

1. The condition **if(!iter->second.sensorData().userDataCompressed().empty())** never gets true. I even tried to log the uncompressed userData but it didnt work. 

2. The error occurs when the mapData which is received in the MapOptimizer, in that the graph.links.size() = 0 but nodes.size() != 0. I understand that the nodes.size() will contain 1 node for the robot pose (0,0,0), but when a second node has to be added, a link should be added before that in the mapData itself. Is there any way to check this? I think that problem might be in CoreWrapper but I am not sure. 

Thanks a lot.

Best Regards,
Kamal

matlabbe, 2016-06-01:
Hi,

1. From the code:
  ```cpp
std::map<int, Signature> newNodeInfos;
// add new odometry poses
for(unsigned int i=0; i<msg->nodes.size(); ++i)
{
	int id = msg->nodes[i].id;
	Transform pose = rtabmap_ros::transformFromPoseMsg(msg->nodes[i].pose);
	Signature s = rtabmap_ros::nodeInfoFromROS(msg->nodes[i]);
	newNodeInfos.insert(std::make_pair(id, s));
}
```
there is indeed a problem with the conversion `nodeInfoFromROS()`, the `SensorData` inside is not copied! Use `nodeDataFromROS()` instead and the `userData` inside the `SensorData`, if there is one, [should be copied](https://github.com/introlab/rtabmap_ros/blob/e67770cc319f3eeddea00bf4f039c48d0f4a6997/src/MsgConversion.cpp#L579):

  ```cpp
std::map<int, Signature> newNodeInfos;
// add new odometry poses
for(unsigned int i=0; i<msg->nodes.size(); ++i)
{
	int id = msg->nodes[i].id;
	Transform pose = rtabmap_ros::transformFromPoseMsg(msg->nodes[i].pose);
	Signature s = rtabmap_ros::nodeDataFromROS(msg->nodes[i]); // <----------- FIX
	newNodeInfos.insert(std::make_pair(id, s));
}
```

2. You can show up the graph contained in [rtabmap_ros/MapGraph](http://docs.ros.org/api/rtabmap_ros/html/msg/MapGraph.html) msgs:
  ```bash
$ rostopic echo /rtabmap/mapGraph
```
Normally, the message itself should be consistent, i.e., (1 pose 0 links) or (2 poses 1 link). However, if you created new sessions, the maps may not be linked together, i.e., 2 poses from 2 maps and 0 link. It is why `optimizer_->getConnectedGraph(...)` is used, to filter maps not connected from the graph. You may want to delete the database (`~/.ros/rtabmap.db`) each time you start a new session so that poses and links are consistent (all in from the same session).

cheers

closed 2016-11-09 by matlabbe