introlab/rtabmap_ros #83: error: (-215) trainDescCollection[iIdx].rows < IMGIDX_ONE

kevinlyu1, 2016-06-09:
has anyone ever encountered a problem that is "OpenCV Error: Assertion failed (trainDescCollection[iIdx].rows < IMGIDX_ONE) in KnnMatchImpl, file ../matchers.cpp, line 365 terminate called after throwing an instance of 'cv::Exception'"? I am getting it when I do long mapping sessions. For some reason I also get it even if I save one session and start another one. If anyone can help that would be great. I think I've attached the .png below.

[image]

matlabbe, 2016-06-09:
Hi,

you may want to activate the debug log of rtabmap node to see from where the opencv function is called:
```bash
$ rosservice call /rtabmap/rtabmap/log_debug
```

The referring assert in OpenCV is [here](https://github.com/Itseez/opencv/blob/2.4.8/modules/features2d/src/matchers.cpp#L365):
```cpp
    for( iIdx = 0; iIdx < imgCount; iIdx++ )
    {
        CV_Assert( trainDescCollection[iIdx].rows < IMGIDX_ONE );
        batchDistance(queryDescriptors, trainDescCollection[iIdx], dist, dtype, nidx,
                      normType, knn, masks.empty() ? Mat() : masks[iIdx], update, crossCheck);
        update += IMGIDX_ONE;
    }
```

The `trainDescCollection[0]` would be the `newWords` matrix from [there](https://github.com/introlab/rtabmap/blob/7c8583f0253897a8a5acfb22e2c3fb23f66026b5/corelib/src/VWDictionary.cpp#L1086) in rtabmap code (the only place it is called if all NN related parameters are set to KdTree which is the default):
```cpp
std::vector<std::vector<cv::DMatch> > matchesNewWords;
cv::BFMatcher matcher(descriptors.type()==CV_8U?cv::NORM_HAMMING:useDistanceL1_?cv::NORM_L1:cv::NORM_L2SQR);
matcher.knnMatch(descriptors.row(i), newWords, matchesNewWords, newWords.rows>1?2:1);
```

Well, `newWords` would have a maximum of 400 words (by default). The OpenCV assert happens if there are `IMGIDS_ONE` words in that matrix. IMGIDS_ONE is defined [here](https://github.com/Itseez/opencv/blob/2.4.8/modules/features2d/src/matchers.cpp#L343-L344):
```cpp
const int IMGIDX_SHIFT = 18;
const int IMGIDX_ONE = (1 << IMGIDX_SHIFT);
```

If you can have a debug log, it would help to see if the `newWords` matrix is somewhat corrupted.

cheers

kevinlyu1, 2016-06-13:
Thanks for the help. For some reason the error hasn't come up in the past few days of testing. So I am not sure if it's fixed or not.

Due to that should I close this issue or leave it open?

matlabbe, 2016-06-13:
I will close it as "not reproducible". If it happens again, just re-comment on this thread. 

cheers

closed 2016-06-13 by matlabbe

siddharth-gitrepo, 2024-03-08:
Have been facing this issue.

matlabbe, 2024-03-09:
Which opencv version are you using? Your example is purely opencv, I guess the issue is coming from opencv.