introlab/rtabmap_ros #84: Tf related issues

ksgaur12, 2016-06-11:
Hi,
I am able to run the stereo outdoor mapping demo using my own stereo camera. I am having some doubt related to the frames.
1. How is the odom frame initialize? Is it the same as the basefoot print when we start the launch file?
2. When does the odom frame relative to the map changes and why does it change?

I want to get the pose of my stereo camera base wrt to a world frame. The world frame must be set to where I start the lauch file. Please help.

matlabbe, 2016-06-11:
Hi,

`/map`, `/odom` and `/base_footprint` should lie on each other on the origin `(0,0,0)` when starting. When the robot moves, `/odom -> /base_footprint` will change accordingly to pose estimated by `stereo_odometry` node. `/odom` and `/map` will be equal until a loop closure is found by `rtabmap` node. If so, odometry drift correction will be in `/map` -> `/odom`. If you want the `/map` starting to a specific location, you may add a [static_public_transform](http://wiki.ros.org/tf#static_transform_publisher) `/world -> /map`.

Then when you want to get robot's pose in `/world` frame, you would get TF `/world -> /base_footprint`, which indirectly is `/world -> /map -> /odom -> /base_footprint`.
```bash
$ rosrun tf tf_echo /world /base_footprint
```

To read more about the coordinate frames convention used in ROS, see [REP 105](http://www.ros.org/reps/rep-0105.html).

cheers

ksgaur12, 2016-06-13:
Thankyou for your help.

matlabbe, 2016-06-13:
Normally, the twist should be in /base_footprint frame. The pose is in /odom frame. See http://docs.ros.org/api/nav_msgs/html/msg/Odometry.html

cheers

ksgaur12, 2016-06-25:
One more clarification needed 

Transform localTransform = getTransform(this->frameId(), imageRectLeft->header.frame_id, stamp);

the function getTransform in the StereoOdometryNode.cpp calls for the transformation between the base footprint and the left camera frame id.
But in the TF the left camera frame id and the stereo camera base are at the same point.

matlabbe, 2016-06-25:
If the transform between /base_link (or /base_footprint) and the frame in the image's header is Identity (`$ rostun tf tf_echo /base_link /left_camera`), the resulting odometry would be in image frame (x->right, y->down, z->forward). To have odometry computed in base frame (x->forward, y->left, z->up), you should have a rotation between the base and the camera:
```bash
$ rosrun tf static_transform_publisher 0 0 0 -1.5707963267948966 0 -1.5707963267948966 base_link left_camera 100
```

closed 2016-11-09 by matlabbe